lemmingDev/ESP32-BLE-Gamepad #1: Forked FYI with added example

MagnusThome, 2020-01-09:
https://github.com/MagnusThome/ESP32-BLE-Gamepad

Added an example with a combo of rotary encoders and butttons. 

Contact me if you want to merge it or have anything else on your mind!

:-)

jonathan-ko, 2020-01-14:
Hi, I am trying to connect to an Android or iPhone. It connects but isn't recognized as a gamepad or switch. Serial communication is fine, but I am trying to make an esp32 with buttons work with the accessibility options with the devices. Have either of you tried it? it's under settings and accessibility features...

lemmingDev, 2020-03-28:
@MagnusThome 

Well done and glad you're getting some use out of the library

My intent was to mimic an Xbox 360/One pad, and that's why I chose to only add that amount of buttons/axis etc.

I guess the ultimate library would be to define what features you want when creating an object and have it alter the HID descriptor based on that.

lemmingDev, 2020-04-01:
@jonathan-ko 

Hi. I have created a sketch that uses my ESP32s builtin button and is able to work with Android's accessibility feature perfectly.

Am running Android 10

My button is on pin 0 and LED is on pin 16. You'll need to change them to suit.
The sketch is also using the bounce 2 library at https://github.com/thomasfredericks/Bounce2 to ensure correct button detection without noise

```
#include <Bounce2.h>
#include <BleGamepad.h> 

int previousButton1State = HIGH;

#define BUTTON_PIN 0
#define LED_PIN 16

// Instantiate a Bounce object
Bounce debouncer = Bounce(); 
BleGamepad bleGamepad;

void setup() {
  bleGamepad.begin();
  // Setup the button with an internal pull-up :
  pinMode(BUTTON_PIN,INPUT_PULLUP);

  // After setting up the button, setup the Bounce instance :
  debouncer.attach(BUTTON_PIN);
  debouncer.interval(5); // interval in ms

  //Setup the LED :
  pinMode(LED_PIN,OUTPUT);

}

void loop() {
  if(bleGamepad.isConnected()) {
    // Update the Bounce instance :
    debouncer.update();
  
    // Get the updated value :
    int value = debouncer.read();
  
    // Turn on or off the LED as determined by the state :
    if ( value == LOW ) {
      digitalWrite(LED_PIN, LOW );
      bleGamepad.press(BUTTON_1);
    } 
    else {
      digitalWrite(LED_PIN, HIGH );
      bleGamepad.release(BUTTON_1);
    }
  }
}
```

deBoulangerie, 2020-11-07:
> https://github.com/MagnusThome/ESP32-BLE-Gamepad
> 
> Upped it to 32 buttons and added an example with a combo of rotary encoders and butttons.
> 
> Contact me if you want to merge it or have anything else on your mind!
> 
> :-)

I am new to the esp/arduino coding and wiring, iam lost on the connections for your code. 

My rotary encoder has 6 pins in all the other examples there defined as pin a, pin b, switch, vcc and ground. 

In your code if I am correct you use: uppPin, dwnPin, prsPin, encoderUpp, encoderDwn and encoderPrs

 So how do they corrospond to pin a, pin b, switch?

MagnusThome, 2020-11-07:
> I am new to the esp/arduino coding and wiring, iam lost on the connections for your code.
>
> My rotary encoder has 6 pins in all the other examples there defined as pin a, pin b, switch, vcc and [...]

xxxPin are the pins to connect to

encoderXXX are the keypad id/codes that are sent to the computer

flyscha, 2020-11-20:
@MagnusThome 

What did you end up making exactly?  It sounds like a button box for sim racing.  That's what I'm trying to do.

I tried using your sketch, but I can't get anything to read in my sim, but the same goes with lemmingDev's single button sketch either.  Obviously I'm not doing something correctly.

To get one of the keystrokes to read, are you grounding the GPIO pins out, or doing something different?  I know the ESP32 is connected via Bluetooth, since I can upload a BLEkeyboard skecth and get normal keyboard characters to show up in the sim.  I've explained my complete situation in a new "issue" thread within the BLEGamepad comments as well.

Thanks for any help!

deBoulangerie, 2020-11-21:
> @MagnusThome
>
> What did you end up making exactly? It sounds like a button box for sim racing. That's what I'm trying to do.
>
> I tried using your sketch, but I can't get anything to read in my sim, [...]

Try to understand how button matrix work if you want more then 4 buttons.

If you use this sketch simply connect a button between pin 27 and 17 for test. 

For rotary enconder use pin: 36 -> clk, 26 -> dt, 18 -> sw. 
Connect ground and + to the board for power to the encoder. 

Upload the sketch and connect the esp32 via bluetooth to your pc, open up game devices and test inputs.

flyscha, 2020-11-21:
@deBoulangerie 

Thank you for the reply!  I'm glad I have the same thinking as you mentioned regarding connecting pins 17 & 27 together to make a button press test from the keypad portion of the sketch.  Unfortunately, I'm just not getting any readings to show in my sim when I do that.

I show full connection, and other sketches make keyboard characters when grounding out specified pins, so I know at least my board is talking to the sim.  That's where I'm lost.

deBoulangerie, 2020-11-21:
@flyscha do you see the device in device manager or "set up game device" on your pc?

Make your using the right pinout esp32 boards can have very different pin numbers.

flyscha, 2020-11-21:
@deBoulangerie 

Thanks for your help!  You had the same thought as others; and I wasn't seeing the device listed in my game controllers.  After uninstalling everything and re-pairing it all, I was finally able to see the device listed where I should.  I'm actually getting readings now that are device buttons!  Again, thanks for the replies..  Much appreciated!

lemmingDev, 2020-11-28:
https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/v0.1.4

New release supports 16 bit on left and right thumb sticks plus has 32 button support

closed 2020-11-29 by lemmingDev