lemmingDev/ESP32-BLE-Gamepad #4: I coundn't read a analogic port

Gigioss, 2020-03-27:
How do I read an analog port? Because when I define it it just ignores it. I want to put a potentiometer to make a steering wheel

lemmingDev, 2020-03-28:
Send your code and I'll have a look

Might pay to read this too...
https://randomnerdtutorials.com/esp32-adc-analog-read-arduino-ide/

And, according to that, analog reads are 12 bit (0~4095), whereas this gamepad HID is expecting values from -127 to 127
  LOGICAL_MINIMUM(1),  0x81, //     LOGICAL_MINIMUM (-127)
  LOGICAL_MAXIMUM(1),  0x7f, //     LOGICAL_MAXIMUM (127)
So perhaps you need to map the values from the analog read to the values the gamepad axis needs...

Gigioss, 2020-03-30:
#include <BleGamepad.h> 

BleGamepad bleGamepad;

int valor;

void setup() {
  Serial.begin(115200);
  pinMode(15,INPUT);
  bleGamepad.begin();
}

void loop(){

  if(bleGamepad.isConnected()) {
    valor=(-(analogRead(15)*254)/4095+127);
    Serial.println(valor);
    delay(500);
    }

}

//It reads only the maximum value on port 4095.

lemmingDev, 2020-04-01:
/*
 * Reads a potentiometer on pin 34 and maps the reading to the X axis
 * 
 * Potentiometers can be noisy, so the sketch can take multiple samples to average out the readings
 */

#include <BleGamepad.h> 

BleGamepad bleGamepad;

const int potPin = 34;                // Potentiometer is connected to GPIO 34 (Analog ADC1_CH6) 
const int numberOfPotSamples = 5;     // Number of pot samples to take (to smooth the values)
const int delayBetweenSamples = 4;    // Delay in milliseconds between pot samples
const int delayBetweenHIDReports = 5; // Additional delay in milliseconds between HID reports

void setup() 
{
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  bleGamepad.begin();
}

void loop() 
{
  if(bleGamepad.isConnected()) 
  {
    
    int potValues[numberOfPotSamples];  // Array to store pot readings

    // Populate readings
    for(int i = 0 ; i < numberOfPotSamples ; i++)
    {
      potValues[i] = analogRead(potPin);
      delay(delayBetweenSamples);
    }

    int potValue = 0;   // Variable to store calculated pot reading average

    // Iteraate through the readings to sum the values
    for(int i = 0 ; i < numberOfPotSamples ; i++)
    {
      potValue += potValues[i];
    }

    // Calculate the average
    potValue = potValue / numberOfPotSamples;

    // Map analog reading from 0 ~ 4095 to 127 ~ -127 for use as an axis reading
    int adjustedValue = map(potValue, 0, 4096, 127, -127);

    // Print readings to serial port 
    Serial.print("Sent: ");
    Serial.print(adjustedValue);
    Serial.print("\tRaw Avg: ");
    Serial.print(potValue);
    Serial.print("\tRaw: {");

    // Iterate through raw pot values, printing them to the serial port
    for(int i = 0 ; i < numberOfPotSamples ; i++)
    {
      Serial.print(potValues[i]);

      // Format the values into a comma seperated list
      if(i == numberOfPotSamples-1)
      {
        Serial.println("}");
      }
      else
      {
        Serial.print(", ");  
      }
    }
       
    bleGamepad.setAxes(adjustedValue, 0, 0, 0, 0, 0, DPAD_CENTERED);
    delay(delayBetweenHIDReports);
  }
}

lemmingDev, 2020-04-01:
And here is your code fixed up

```
#include <BleGamepad.h>

BleGamepad bleGamepad;

int valor;

void setup() 
{
  Serial.begin(115200);
  pinMode(15,INPUT);
  bleGamepad.begin();
}

void loop()
{
  if(bleGamepad.isConnected()) 
  {
    valor = analogRead(15);
    valor = map(valor, 0, 4095, 127, -127);
    Serial.println(valor);
    delay(500);
  }
}

//Now working
```

Gigioss, 2020-04-04:
I used the first code you sent, because the other code does not worked. And I increased more two poternciometer (accelerator and brake), two Button for switch the gears.

jmtstorres, 2020-07-11:
I'm having a similar issue. When the begin function is called, all analog values read freeze at max (in my case 4095). 
If commented, alnalog values read normally. I'm using GPIOs 2 and 4 for the X and y axes of a 2d potentiometer.

jmtstorres, 2020-07-11:
That was fast, but seems that using any ADC1 pin (in this case GPIO34 as the pot example) works. ADC2 is dead when begin is called.

closed 2020-11-29 by lemmingDev