lemmingDev/ESP32-BLE-Gamepad #5: Don't work on Android

akulij, 2020-03-29:
Hello, my phone don't react to my esp32. My phone mi a2 on Android 10.

lemmingDev, 2020-04-01:
Hi

Not sure what to say. It's working for me and others...

Have you tried the ButtonsAndAxes example?

I've got a TCL Plex Android phone on Android 10 and its working fine. When i connect to it my phone it starts selecting things and moving up and down in response to the controller.

If you're writing your own code, then make sure it works on Windows first...

akulij, 2020-04-01:
Yes, it working on windows, I  used an example "Gamepad" .
Where I can get a ButtonsAndAxes example?
Thanks

lemmingDev, 2020-04-01:
Hi - Yeah - Thant was the old name for the 'Gamepad' example

Hmmn - It might be because this example holds down button 1 and never releases it

You could try commenting out 
```bleGamepad.press(BUTTON_1);```
from the example

akulij, 2020-04-02:
However, I used this: https://github.com/wolfeidau/esp32-hid-joystick, and it works fine on my smartphone

akulij, 2020-04-02:
I commented first button press, but nothing changed.
(I'm sorry for tormenting, I'm stupid))

lemmingDev, 2020-04-02:
Hi - So the ESP32 gamepad pairs with your phone, but nothing responds?

I'm not sure what's happening as every device I've tried it on seems to work fine.

May need someone smarter than me to hep you with this one I'm afraid

closed 2020-11-29 by lemmingDev