lemmingDev/ESP32-BLE-Gamepad #7: Doesn't appear to work on windows 7

AlessandroAU, 2020-04-06:
I've uploaded one of the examples, I can pair using my windows 10 laptop and my android phone. However, I tested on two other windows 7 machines and they cannot see the Bluetooth device at all.

lemmingDev, 2020-04-06:
Thanks for testing

I got the Bluetooth code from this BT mouse library here.
https://github.com/T-vK/ESP32-BLE-Mouse
Could you upload one of the test sketched from that library and see if your Windows 7 machines see the BT mouse

AlessandroAU, 2020-04-07:
Interesting! I get the same result, must be something in the original code.

lemmingDev, 2020-04-07:
Hmmn - Do those Windows 7 machines support at least Bluetooth 4.0?

Me thinks that may be the issue,,,

AlessandroAU, 2020-04-08:
Yeah I think this might be the problem

closed 2020-04-20 by lemmingDev