lemmingDev/ESP32-BLE-Gamepad #8: One operation for dpad and joy stick ?

wolfshow, 2020-04-10:
Do I always need to set the state of dpad and joy stick at the same time?

lemmingDev, 2020-04-20:
Yeah, 
```void BleGamepad::setAxes(signed char x, signed char y, signed char z, signed char rZ, char rX, char rY, signed char hat)

The setAxes function needs to be passed all values.

Library could be modified though

jacobparra, 2020-05-17:
@wolfshow I was just working on something like that:

```
void BleGamepad::setAxes(signed char x, signed char y)
{
  xAxis = x;
  yAxis = y;
}

void BleGamepad::setHat(signed char h)
{
  hat = h;
}
```

and then i could execute `void BleGamepad::notify()` that sets the data array and notify:

```
void BleGamepad::notify()
{
  if (!this->isConnected())
    return;

  uint8_t m[5];
  m[0] = buttons;
  m[1] = (buttons >> 8);
  m[2] = xAxis;
  m[3] = yAxis;
  m[4] = hat;
  if (data[0] != m[0] || data[1] != m[1] || data[2] != m[2] || data[3] != m[3] || data[4] != m[4])
  {
    data[0] = m[0];
    data[1] = m[1];
    data[2] = m[2];
    data[3] = m[3];
    data[4] = m[4];
    this->inputGamepad->setValue(data, sizeof(data));
    this->inputGamepad->notify();
  }
}
```

lemmingDev, 2020-11-29:
Thinking I may rewrite the library when I get a chance to function this way

lemmingDev, 2021-01-09:
Hi - The library has been updated with the ability to set the hat, left thumbstick, right thumbstick, left trigger and right trigger separately

closed 2021-01-09 by lemmingDev