lemmingDev/ESP32-BLE-Gamepad #9: Is there a way to add more than 6 axes?

1mpact21, 2020-04-20:
I'm looking for a way to control 9 encoders, 14 buttons and hopefully a pot using HID, and I'm looking at ESP32 because the Bluetooth is convent and pins are plenty. I was curious in this libraries similarities to the Arduino joystick library which allows an additional 5 axis through braking,steering,rudder etc... Is this library a good substitute or is my 9 encoder setup too much. thanks in advance

lemmingDev, 2020-04-20:
Hi

With some extra programming, the library can be edited to work with any number of gamepad/joystick elements.

Look here: https://github.com/MagnusThome/ESP32-BLE-Gamepad for example.
They modified the library already and it might suit your needs better than my original one.

lemmingDev, 2020-11-29:
Am thinking I may add support for some additional axes.

Have done a little testing and only 7 max are viewable on the Windows game controller panel, but additional ones can still be used by applications.

closed 2020-11-29 by lemmingDev