lemmingDev/ESP32-BLE-Gamepad #12: Force feedback support

joegaffey, 2020-05-05:
Would it be possible to add support for force feedback for a wireless steering wheel?

lemmingDev, 2020-05-05:
Way out of my league I'm afraid

Perhaps someone smarter than me will take up the challenge

joegaffey, 2020-05-07:
I may be just about uninformed enough to give this a shot if I can find some spare time.

I found this snip which shows how to receive outputReports but I don't know how or if its even possible to advertise the ESP32 as a HID Physical Interface Device over BLE. 
https://gist.github.com/sabas1080/93115fb66e09c9b40e5857a19f3e7787    

I feel like if I managed to get some FFB data back then perhaps I could try to decode enough of it to implement some basic effects.

closed 2020-11-29 by lemmingDev