lemmingDev/ESP32-BLE-Gamepad #15: BLE Advertising

toman9, 2020-05-28:
Hi,

On my system (Windows 10), the ESP32 BLE device seems to advertise itself for 10 seconds after reboot, and stops afterwards (Windows no longer sees the BLE device) 
Does anyone know how to prolong advertising time to 30 seconds or more?

adotagarwal, 2020-06-03:
On mine it is advertising much longer... did you try the demo sketch?

closed 2020-11-29 by lemmingDev

redsil, 2022-11-02:
I know this is closed, but just for posterity: I see the same issue and can't figure out what is going on.  If I connect while it is visible, it stays connected, but if I reboot the esp32, it will not reconnect. I have to remove the device in windows and reconnect within the 10 second window.  It's strange, because I have another esp32 device using esp32-ble-gamepad and it doesn't have this issue.

redsil, 2022-11-02:
I ran the example code and what is happening is that it is connecting to "something" pretty quickly and this seems to cause advertising to stop.  However, none of the 3 machines I am working with are connected to it.  Is there a way to list what device the esp32 is connected to?  Is there a way to filter what to connect to?