lemmingDev/ESP32-BLE-Gamepad #17: Support Multiple Gamepads

alijani1, 2020-07-11:
Add ability to support multiple HIDs on the same device

alijani1, 2020-07-11:
I forked and was able to get 2 gamepads to work on windows but on Linux only first gamepad is detected as a hid.  see https://github.com/alijani1/ESP32-BLE-Gamepad   

Any advice?

lemmingDev, 2020-07-25:
Hi

First of all, thanks for taking an interest and forking it to share your improvements with everyone. Great to see.
Sorry - no idea about Linux though. You could try asking the two people listed in the credits

lemmingDev, 2020-11-29:
https://stackoverflow.com/questions/29358179/usb-possible-to-define-multiple-distinct-hid-joysticks-on-one-interface

This seems to explain how to get it working

alijani1, 2020-11-30:
thank you for the suggestion. I tried it but unfortunately no luck.  

I ran the command:  modprobe usbhid quirks=0x8282:0x0132:0x40  suggested using the VID/PID I used in BLECamepad.cpp  (i.e.  BleGamepadInstance->hid->pnp(0x02, 0x8282, 0x0132, 0x0106);   I also tried 0x3201 for id incase they were revered in modprobe command.

Still only js0 on linux.  Works fine in windows.

lemmingDev, 2021-01-09:
Let me know if you ever get this working...

closed 2021-01-09 by lemmingDev

alijani1, 2021-01-10:
Yes. I was able to get it working.  
It turns Linux only supports multiple HIDs on a single usb via a quirk flag HID_QUIRK_MULTI_INPUT set in its hid-ids list. By looking at github.com/.../blob/master/drivers/hid/hid-quirks.c I was able to identify all the VID/PID combinations that have the flag set.  there is also a way to set the quirk manually but its easier to just use a HID combination that already has the quirck set.  Thanks!