lemmingDev/ESP32-BLE-Gamepad #18: ask for help

zuozhehao, 2020-08-27:
I see the absolute coordinates attachment from here, but it doesn't work properly on my iPhone.

This absolute coordinate calculation has been bothering me for a week. I want to ask for help.
[https://github.com/T-vK/ESP32-BLE-Keyboard/issues/14](https://github.com/T-vK/ESP32-BLE-Keyboard/issues/14#issuecomment-605957835)

lemmingDev, 2020-09-09:
I've only tried it on a Windows PC (with good success)

Am unable to test on iOS as I moved to Android after the iPhone 4

As a suggestion, setting the resolution is needed. Did you add your iPhone resolution? 
int res_x = 1920;               // Put your screen resolution width here
int res_y = 1080;               // Put your screen resolution height here

I'm not sure how screen rotation would factor into it either. 

Does the regular BLE Mouse library work? I'd check that before trying my modified version

zuozhehao, 2020-09-10:
> I've only tried it on a Windows PC (with good success)
>
> Am unable to test on iOS as I moved to Android after the iPhone 4
>
> As a suggestion, setting the resolution is needed. Did you add your iPhone [...]

Yes, regular BLE Mouse library work.

lemmingDev, 2020-11-29:
There is some discussion here

https://github.com/T-vK/ESP32-BLE-Mouse/issues/19

closed 2020-11-29 by lemmingDev