lemmingDev/ESP32-BLE-Gamepad #19: How to connect connect joystick and buttons?

1gniculus, 2020-09-08:
Hello, can you tell me, which pins to connect joystick and buttons?

lemmingDev, 2020-09-09:
Any available pins really

https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

Be aware that GPIOs 34 to 39 are GPIs – input only pins. These pins don’t have internal pull-ups or pull-down resistors

1gniculus, 2020-09-09:
> Any available pins really
> 
> https://randomnerdtutorials.com/esp32-pinout-reference-gpios/
> 
> Be aware that GPIOs 34 to 39 are GPIs – input only pins. These pins don’t have internal pull-ups or pull-down resistors

i dont understand
esp32 itself determines which buttons or joystick are where or i need to register that in code which pin use(where)?

lemmingDev, 2020-09-10:
Hi

Any available pins on your particular board can be used.

You need to define those pins in your code, and then check the state/readings of those pins, and then send button presses/axis movements etc based on the state/readings of those pins

The SingleButton.ino sketch shows how pin 2 is registered as an input pin with the internal pullup resistor enabled, and is then checked in the loop to determine if the current state of the pin is different than the previous state of the pin.
If the state is different (user has either pressed or released the button), then a button of the bluetooth controller is pressed or released.

You just need to do that for all pins you want to use as buttons

The PotAsAxis.ino sketch shows how a pin can be used to detect values from a potentiometer to set axis values

If you are really struggling, send me a link to the actual board you have, and what hardware/buttons/axis you want to use and maybe I can help further

1gniculus, 2020-09-11:
No, thanks
And THANKS A LOT for your explanation)

closed 2020-09-11 by 1gniculus