lemmingDev/ESP32-BLE-Gamepad #26: Individual control of D-pad buttons

mikepurvis, 2020-11-17:
Hey, love the library— the examples work great on my LOLIN32 board. However, my intended project is a wireless dance pad for Stepmania and similar games, which requires me to be able to be able to independently command the four cardinal directions. The current API limits the dpad (or "hat") to only the nine cardinal positions:

https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/3706e489ab1ebf836897005b4a1be7cc9044d746/BleGamepad.h#L25-L33

And the X and Y axes must each be a linear scale from -128 to 127.

Neither of these will permit an input where the up and down arrow are simultaneously pressed. From looking at the implementation, it appears this is a limitation of the underlying BLE gamepad protocol, and not just of this particular library. Am I correct in that, or is there something here I've missed? Is there a different approach which permits the four discrete digital inputs (up/down/right/left) to be independently pressed and released?

I know I could just have the pad emulate a BLE keyboard instead, but it would be nice to go the pad route as then it's a cleaner path to supporting multiple of them— just pair both and away you go, whereas with a keyboard device approach, you'd need a way to switch one of them to use WASD or something instead of the arrow keys.

mikepurvis, 2020-11-17:
Looks like when it comes to HID devices, the Bluetooth spec just passes through to what USB HID defines, via the "HID over GATT Profile", per: https://www.bluetooth.com/specifications/gatt/

The discussion in https://github.com/NicoHood/HID/issues/90 led me to [this doc](http://www.freebsddiary.org/APC/usb_hid_usages.php), and this in particular:

```
	0x88	System Menu Exit
	0x89	System Menu Select
	0x8A	System Menu Right
	0x8B	System Menu Left
	0x8C	System Menu Up
	0x8D	System Menu Down
	0x90	D-pad Up
	0x91	D-pad Down
	0x92	D-pad Right
	0x93	D-pad Left
```

Now, when I open the Windows 10 game controller properties for a DualShock 4 (whether connected by USB or bluetooth; I tried both and it makes no difference), it definitely represents the D-pad as the "hat" per this library. Unfortunately I don't have an Xbox 360 or One controller on hand to try, but if the linked discussion is correct, I suspect that those controllers might represent the D-pad as four discrete digital inputs instead (or in supplement to) the "hat" input? Either way, it seems likely that 0x90 - 0x93 would be usable for this purpose, and in any case, if it's just a matter of doing over Bluetooth what other USB pads do over a wire, it should be possible to figure that out without too much trouble.

mikepurvis, 2020-11-17:
Looks like this is where those usages are set up in this library:

https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/ce8cf9e98137e93be4efaa5ee9720a284aea2bc5/BleGamepad.cpp#L43-L52

So it shouldn't be too hard to try out the d-pad ones and see if it does the right thing for me. This use case is probably niche enough that it may not make sense as something for the upstream library to support, but at least this ticket can exist as something for a future traveler to find if they end up with similar questions!

lemmingDev, 2020-11-21:
Hi

It's actually lines 63 to 70
```
  USAGE_PAGE(1),       0x01, //     USAGE_PAGE (Generic Desktop)
  USAGE(1),            0x39, //     USAGE (Hat switch)
  USAGE(1),            0x39, //     USAGE (Hat switch)
  LOGICAL_MINIMUM(1),  0x01, //     LOGICAL_MINIMUM (1)
  LOGICAL_MAXIMUM(1),  0x08, //     LOGICAL_MAXIMUM (8)
  REPORT_SIZE(1),      0x04, //     REPORT_SIZE (4)
  REPORT_COUNT(1),     0x02, //     REPORT_COUNT (2)
  HIDINPUT(1),         0x02, //     INPUT (Data, Variable, Absolute) ;1 byte Hat1, Hat2
```

https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/ce8cf9e98137e93be4efaa5ee9720a284aea2bc5/BleGamepad.cpp#L63-L70

The reason you can't do what you're trying is because I've set it up as a hat switch, instead of a d-pad

I can see the merit in having it set as a d-pad, so I may adjust the library when i get some time, unless you'd like to make a pull request ;)

lemmingDev, 2020-11-21:
Hi

I went and made those changes to my library to see if I could get it working.

I could indeed replace the hat functionality with the dpad functionality, however it seemed pretty pointless as it just added 4 more buttons and removed the hat/direction thing in the lower right of the gamepad dialog in windows. 

[image]

That thread you mentioned basically said the same thing.

Can't you just map any of the 14 buttons to the 4 directions you want in whatever emulator/program you are using to play your rhythm game?

If you really need 14 buttons plus the individual dpad buttons, then I have attached a version of the library for you that does that. I have updated the readme plus the single button example so that you can see the usage.

[ESP32-BLE-Gamepad.zip](https://github.com/lemmingDev/ESP32-BLE-Gamepad/files/5578530/ESP32-BLE-Gamepad.zip)

I guess I could add the dpad code in addition to the hat code, but my intent of my library was to have the same amount of buttons/axes/functionality etc as what shows in the test window as when the xbox360 controller is connected.

lemmingDev, 2020-11-22:
Ok - so as an update - and for the interest of people reading this topic

I connected my official Xbox360 Konami Dance Pad to Windows 10

[image]

It maps the direction arrows on the mat to a hat switch.

They work individually fine, and in the four corners as you would expect, however when up and down are pressed together, it registers as no input at all. When the left and right are pressed together, it incorrectly registers as down and left.

So, I guess the driver on the actual Xbox allows individual dpad presses, but not over on the PC where Windows maps the xinput data to stuff it understands

Note: interestingly, when plugged in, the dance mat does have 4 other buttons show up than it needs:

Button 1 - A
Button 2 - B
Button 3 - X
Button 4 - Y
Button 5 - Unmapped
Button 6 - Unmapped
Button 7 - Back
Button 8 - Start
Button 9 - Unmapped
Button 10 - Unmapped

However I think it does that just so it recognises it as a standard digital gamepad, that would ordinarily map the left/right bumpers/triggers to buttons

There is an unsigned driver here: https://www.goshdarngames.com/how-to-use-a-usb-xbox-360-dance-mat-with-windows-10/ that apparently enables the exact mat I have to function correctly. I have not tested it yet though

lemmingDev, 2020-11-22:
And for completeness, even using a virtual XBOX 360 pad does exactly the same thing if you try to press the left and right dpad directions t the same time

[image]

mikepurvis, 2020-11-23:
Wow, thanks for digging in here— interesting especially to see what the Konami pad does. Were you able to tell whether its extra mystery buttons were 0x8A-0x8D vs 0x90-0x93? I guess the Windows gamepad tester probably isn't enough to tell that, but some kind of lower level USB HID sniffer could tell you?

For sure the easiest thing is to remap in the software, and possibly even just bypass gamepad mode altogether; as I said in the initial post I could do it as one pad sending keyboard arrows and the other sending keyboard WASD.

That said, I'll experiment a bit to see what Necrodancer and Stepmania expect as input. If they want to see the hat input for menu navigation and so on, it would be easy to just have it always set the D-pad button, and set the hat only if it's a single button pressed at once. As I say, though, no pressure to explicit support this in your library; there are enough of the bones here that it might make more sense for me to just copy the bits that I need directly into my own program.

lemmingDev, 2020-11-28:
I've added a new version that has 32 buttons, so there should be plenty of buttons to map to whatever you need...

https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/v0.1.5

I'm thinking it's not worth adding the dpad stuff, but if a few people think they really need it, then I may think about adding it

closed 2020-11-29 by lemmingDev

Palio-120, 2024-08-30:
Hello, Good Afternoon, I downloaded your Release 5.4 version, I want to have 11 buttons and 1 D-PAD, but I can't get Start and Menu or PS to detect them, I saw that they must be registered as SpecialButtons, but I can't get them to be detected. recognize, could you help me please.