lemmingDev/ESP32-BLE-Gamepad #27: In need of a little direction please (not an issue)

flyscha, 2020-11-20:
Sorry, I can't seem to find another way of contacting you other than claiming an issue.

First of all, thank you for taking the time to make the libraries and example sketches.

I'm pretty new to Arduino in general and I've been led to the ESP32 board for my project.  I'm looking to make a button box for sim racing using the Bluetooth connection of the dev board.

With the help of a sketch I found through a different tutorial, I am currently able to treat the ESP32 board as a BT keyboard, and send keystrokes by connecting assigned pins to the ground terminals.  This allows keystrokes to show up in my sim, and I can map those keyboard keys to different buttons.  The issue is I still want to type during races, so having normal letters/numbers/symbols mapped to different functions won't allow me to type/chat.

So, I was hoping that the character created by your single button sketch would read as a device button rather than a standard keyboard key.  Unfortunately, I'm not understanding how to send any characters/keystrokes with the method explained in your example.  I've even tried the https://keycode.info/ page without anything registering.

In the end I'm just hoping for some direction for how to use your sketch to register a character of some kind.  I'm able to wrap my head around the sketch I've used successfully that has pins that read normally high, and it creates the keystroke when touched to ground.  I'm just lost when it comes to understanding the method in your example.

Any help is greatly appreciated!  I'm sorry for my lack of knowledge.

lemmingDev, 2020-11-21:
Hi

Are you talking about my Gamepad library (button presses) or a different Keyboard library (key strokes)?

If you're talking about my library Gamepad library, the valid button presses are:

```bleGamepad.press(BUTTON_1);
bleGamepad.press(BUTTON_2);
bleGamepad.press(BUTTON_3);
...
...
bleGamepad.press(BUTTON_14);
```

and then when you want to release buttons:

```bleGamepad.release(BUTTON_1);
bleGamepad.release(BUTTON_2);
bleGamepad.release(BUTTON_3);
...
...
bleGamepad.release(BUTTON_14);
```

The keyboard keycodes you linked to have nothing to do with this gamepad library as it's not a keyboard.

Let me know if you're still confused.

lemmingDev, 2020-11-21:
Are you aware that for my library to work as intended, you need to wire up physical buttons etc to the ESP32, and when those buttons are pressed, it presses the buttons on the BLE Gamepad device in Windows?

flyscha, 2020-11-21:
Thank you for the reply!

I guess that's where I'm confused.  I know what you mean about your library and sketches compared to what I mentioned about the keyboard setup, I was just mentioning the keyboard because I'm able to get that to work, making me realize I'm getting Bluetooth communication between my board and the sim.

One thing I'm not clear on is what you mean about "when those buttons are pressed, it presses the buttons on the BLE Gamepad device in Windows".  Is there a specific situation this only works on/with?  I was thinking your sketches and libraries could be utilized for any PC games that require buttons/switches.  Maybe that's where I'm mistaken?

Taking your single button sketch for example; am I wrong in assuming that doing something with pin 2 will act as a button press within a game?

My mindset (which sounds incorrect) is that I should be able to touch two pins together to press a button in-game, and then release the button when the pins aren't connected anymore.

Thanks for any additional help!

lemmingDev, 2020-11-21:
Yeah, so in my single button example, you have a wire connected to pin 2, and another wire connected to ground.

When the 2 wires are touched together, your game will detect that gamepad button 1 has been pressed

flyscha, 2020-11-21:
Great, that makes me feel better that I'm understanding it that way as well.

I just uploaded the sketch once again, and I'm not getting anything to happen between pins GPIO 2 and ground.  That's what has been confusing me.

Do you have any suggestions for me to see if the button press can be read in another windows-related situation instead?  Something like Notepad or https://keycode.info/ type of things that you know will show what the ESP32 is providing for the button press?  Maybe it's just my game not reading it.

lemmingDev, 2020-11-21:
Try editing the sketch to use a different pin

Does your board have a built in button. Perhaps you could try that pin

Windows has "Set up USB game controllers"

Select the gamepad and then click properties

You should then see all button presses...

flyscha, 2020-11-21:
Again, thank you so much.  You're definitely helping me get to the root of the problem.

So I see the gamepad in my devices within the Windows control panel, but the device is not showing up on my game controllers list.  Only my steering wheelbase and pedals are showing up in there.  I'm clicking on everything and not seeing how to add the gamepad to that list, but I'll keep at it.

lemmingDev, 2020-11-21:
I would try removing the Bluetooth gamepad, and then searching for it again to re-pair it.

flyscha, 2020-11-21:
You were absolutely correct, that was the issue.  I re-paired it and started over from scratch.  I now see the device listed with my game controllers and I'm actually getting device buttons to read in the simulator!  Now it's all about going through the coding to get everything how I want it regarding pins, etc.

Thank you SO much for your help, this has been frustrating trying to figure it all out.  You rule!

lemmingDev, 2020-11-21:
No problemo

Glad you got it working.

Hey - I just uploaded a new version a few mins ago "Release 1.2" that updates the readme and example code to correctly set the trigger values

You should use that so you don't get confused when trying to set the trigger values.

Let me know how you go...

flyscha, 2020-11-21:
Thanks for the update!

Dumb question.. Can/should I use the new version to update the sketch of the other guy that expanded your gamepad for more buttons and encoders?  That's the sketch I'm starting with & modifying to get what I need.

lemmingDev, 2020-11-21:
Oh

If you mean https://github.com/MagnusThome/ESP32-BLE-Gamepad

It seems he is using the range -127 to 127 for his triggers, while i am using 0 to 255; so you should just use those examples...

flyscha, 2020-11-21:
Yes, that's the one!  When you say use "those examples", do you mean his or yours?

I just want to be sure!  haha

lemmingDev, 2020-11-21:
His

flyscha, 2020-11-22:
Hi,

Question for you if you don't mind please.  This is regarding the https://github.com/MagnusThome/ESP32-BLE-Gamepad sketch.

Regarding the keypad matrix, everything works except for pin 32 with pins 00 and 02.  All other pin connections and button reads for the keypad work fine (14 button presses).  However, when I try to use a different pin other than 32 or 00/02 in the code, EVERYTHING stops working.  Every time I go back to the original code, everything works again minus those 2 options.

I'd like to add a 5th row & 5th column as well, but adding anything also makes the entire matrix stop working.  That's even with changing the rows & cols from 4 to 5, adding pins, and adding button press numbers those pin combinations should produce.

Here is the board I'm using, just in case: https://www.amazon.com/gp/product/B07QCP2451/ref=ppx_yo_dt_b_asin_title_o08_s00?ie=UTF8&psc=1

It's the 38 pin version.  I'm thinking it's more of a code problem, but I've researched the libraries used in the sketch, and can't seem to see why this is happening.  I'm also (think I'm) being careful not to use pins that are already in use for other functions further down the sketch.

Sorry if I'm becoming a pain.

lemmingDev, 2020-11-23:
A great reference for the pins is here:
https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

Pin 32 seems to be fine to use

Perhaps you should swap pin 32 and another one around in the code and in your wiring to see if it starts working

If it works, you know it's a pin issue. If it still doesn't work, you can try something else, like just a minimal sketch with just the keypad code.

lemmingDev, 2020-11-28:
BTW

M latest version now has 32 button support, plus 16 bit axes

closed 2020-11-29 by lemmingDev