lemmingDev/ESP32-BLE-Gamepad #28: Higher resolution

yo90bosses, 2020-11-27:
i noticed that the .setAxes() function uses signed chars as inputs. That leaves us with a resolution of only 256. Is there a way to increase this to at least 16 bit?

lemmingDev, 2020-11-27:
Yes

The resolution of the axes is set with the  LOGICAL_MINIMUM and LOGICAL_MINIMUM in the _hidReportDescriptor in the BleGamepad.cpp file. As you can see, the range is currently set to -127 to 127

```
  USAGE_PAGE(1),       0x01, //     USAGE_PAGE (Generic Desktop)
  USAGE(1),            0x30, //     USAGE (X)
  USAGE(1),            0x31, //     USAGE (Y)
  USAGE(1),            0x32, //     USAGE (Z)
  USAGE(1),            0x35, //     USAGE (rZ)
  LOGICAL_MINIMUM(1),  0x81, //     LOGICAL_MINIMUM (-127)
  LOGICAL_MAXIMUM(1),  0x7f, //     LOGICAL_MAXIMUM (127)
  REPORT_SIZE(1),      0x08, //     REPORT_SIZE (8)
  REPORT_COUNT(1),     0x04, //     REPORT_COUNT (4)
  HIDINPUT(1),         0x02, //     INPUT (Data, Variable, Absolute) ;4 bytes (X,Y,Z,rZ)
```

You would need to make some changes there (the min and maxes and then report size changed from 8 bits to 16 bits ), and also in the data types for the setAxes function from signed char to perhaps a short

I personally think 256 different positions for each axis is fine given the library's intent, but can see how emulating something like a 900 deg steering wheel might need more resolution

lemmingDev, 2020-11-28:
https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/v0.1.4

New release supports 16 bit on left and right thumb sticks

yo90bosses, 2020-11-29:
Great thanks for updating the library with the new features!

closed 2020-11-29 by yo90bosses