lemmingDev/ESP32-BLE-Gamepad #30: Windows does not see ESP32 BLE Gamepad

beastdjw, 2020-11-30:
When I do a Add Bluetooth or other device in Windows 10, it does not see this device. In Linux and Android it sees this device. My bluetooth adapter is a 4.0 usb adapter. Any ideas?

lemmingDev, 2020-12-01:
Hi

Sorry - no idea

Is viewable on every Windows 10 computer (with BT 4.0 or above) I've ever tried

You have to use the top option

[image]

uecken, 2020-12-01:
In Windows 10, I think there are two reasons for not connecting.

1. Bluetooth driver may have bad state.
  Driver may have bad state. Uninstall the driver(ex: intell wireless bluetooth) and reboot Bluetooth Device. I use Bluetooth USB Dongle. 
[image]

2. The Bluetooth module does not support GATT.

beastdjw, 2020-12-02:
Thx for helping! I think my usb bluetooth adapter is not really 4.0 and is not really ble although its mentioned in the description. I've read on the amazon webstore a lot of comments about those usb bt adapters which are not really 4.0. Now I bought a new one (5.0), and will get it in a few days. I will test it and report it here if its functioning.

beastdjw, 2020-12-04:
I have ordered: https://www.amazon.de/-/nl/gp/product/B08K4GZLMH/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
and can confirm the ESP32 BLE Gamepad sofware is now working. So watch out! Buy a good bluetooth ble adapter!

Thanks for helping me!
p.s. I am working at a bluetooth ble buttonbox for a racesteer. I will post this repository soon.

closed 2020-12-04 by beastdjw

lemmingDev, 2020-12-04:
Good to hear

Have fun with the library