lemmingDev/ESP32-BLE-Gamepad #31: [Questions] Support Feedback and Related or Projects using BleGamepad?

LeeNX, 2020-12-07:
Thank you @lemmingDev, as a beginner/tinker, ```BleGamepad``` has been a really nice starting point to build my own joystick/gamepad.  
  
I have a few questions, that I hope this is acceptable channel to post questions and a suggestion or two, if not I apologies.  
  
First question or two, my journey so far, has me testing my project on OSX, using SDL2 and I can confirm it basically works. I was wondering how I can confirm this for others or should I just put in a PR for the ```README.md``` making OSX working?  
  
On my journey, I should be testing ```BleGamepad``` on Linux, thou only Raspberry Pi setup. Not sure that I have any hardware that has BLE, that I could install Linux on, but would R-Pi testing pass as Linux supported?  
  
Suggestion or question, can we post a list of other projects that use ```BleGamepad```. I don't expect ```BleGamepad``` lib to cover all use cases and help, but looking at how or why something did something similar can be very useful. My own project - https://gitlab.com/leet/diy-gamepad - Anybody else have projects they would like listed?  
  
Thanks!

lemmingDev, 2020-12-07:
Hi

Thanks for testing it out and providing feedback

I guess I could mark OSX and Linux as compatible based on your feedback, and some previous feedback I've received.

Perhaps I should add a list a projects to the readme that have used this library as a base.

In the back of my mind, I'm thinking of re-writing the library (or creating another) to have a dynamic/configurable HID descriptor so it can be a lot more flexible.

It's not too difficult for modify the HID and report as it is; if you know what you're doing; which not everyone does however...

LeeNX, 2020-12-08:
Thank you @lemmingDev for the reply!  

Let me post some screenshots with OSX working and do the same for R-Pi a little later, then we could close this issue.  
  
Related or using this project file would be very nice, if anybody else is interested in posting their projects.  
  
As a newibe, I was going to ask if there was a good or simple way to advertise only resources to the host that is in use. Improvements in documentation or examples or features are just plain awesome! ...Thanks!

LeeNX, 2021-01-06:
I have two different projects, which connects to both OSX and Android hosts and works as expected at the current point of development for my projects.  
  
One oddity I did see on OSX, is that I need to use LightBlue to force reconnections to my BLE hardware.  
[image]. 
  
I also ran into problems trying to bind my projects to a R-Pi, but that could just be me and not doing it right. I will report further back on this item.

lemmingDev, 2021-01-07:
Thanks for the feedback

Try the new version and let me know how you go

It send correct axis values on button presses instead of sending zeros. Might help

Also, I have a feeling on Android/Linux/Mac that only 16 buttons can be seen, so make sure you map to those ones
 
https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases

LeeNX, 2021-01-07:
Thanks @lemmingDev for the new release, like christmas in January!  
  
Update my test project, connected to my Mac right away. Will do some more testing, but looks good so far.

LeeNX, 2021-01-07:
Not sure it's useful to have two hats? Watching ```sdl2-jstest```, all buttons and axes are pressed and moved.

lemmingDev, 2021-01-07:
Thanks for testing. It's much appreciated. Good to know OSX can see all 32 buttons.

Yeah. The 2 hats thing was something I got from a HID descriptor I found online when I first created the library. It showed up fine in Windows, so I just went with it.

I may change the implementation if I find a better way/example.

lemmingDev, 2021-01-07:
This is info about hat switches from usb.org :

Hat Switch - A specialized mechanical configuration of switches generating a variable value with a
NULL state. The switches are arranged around a springloaded knob. When the knob
is tilted in the direction of a switch, its contacts are closed. A typical example is four
switches that are capable of generating information about four possible directions
in which the knob can be tilted. Intermediate positions can also be decoded if the
hardware allows two switches to be reported simultaneously.

lemmingDev, 2021-01-13:
I found a way to get the 8 direction working using a single hat.

And then I went crazy and added 4 hats using this method.

The cool thing is that the Windows standard game controller tester shows them all fine (on the same hat pic using different colours)

Anyhoo, the gamepad has now also been upgraded to support 64 buttons, 16 bit on all axes, adds 2 sliders and those 4 hats.

The use http://www.planetpointy.co.uk/joystick-test-application/ this app on Windows if you'd like to test all buttons

closed 2021-01-13 by lemmingDev