lemmingDev/ESP32-BLE-Gamepad #32: Brownout error with NodeMCU 32S

NotQuiteApex, 2020-12-17:
Code example in the README.md with the latest release causes this error on my NodeMCU ESP-32S, (V1.1). Both are brand new and work just fine with other code I write, seems to be caused by something in the begin function. Disabling the brownout functionality simply causes random crashes with random errors.

```
Starting BLE work!

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
```

NotQuiteApex, 2020-12-17:
Turns out the cable I was using was faulty, sorry!

closed 2020-12-17 by NotQuiteApex

lemmingDev, 2020-12-18:
Glad you got it sorted...