lemmingDev/ESP32-BLE-Gamepad #33: How do i go about setting my HID report map?

Terramoto, 2020-12-20:
Hello,

I'm trying to have my flight stick show up in windows game controllers, i've manage to have it connect to windows and display the device but no  configuration is available. I have no idea how/where to setup the HID descriptor report map.

LeeNX, 2020-12-22:
I think you need to update the library to make changes to the HID descriptor, the default is basically what I have posted below.
```
leet-mbp:build leet$ ./sdl2-jstest -l
Found 1 joystick(s)

Joystick Name:     'Manufacturer LeeNX Device Name'
Joystick GUID:     0300000002e50000abcd000001100000
Joystick Number:    0
Number of Axes:     6
Number of Buttons: 32
Number of Hats:     2
Number of Balls:    0
GameControllerConfig:
  missing (see 'gamecontrollerdb.txt' or SDL_GAMECONTROLLERCONFIG)
```
There has been a suggestion update the library to be more dynamic or suggest that it should be easy to override the default, but I will admit I don't know how to do either.

lemmingDev, 2020-12-31:
Are you trying to create a custom controller by modifying the BleGamepad.cpp and BleGamepad.h files?

lemmingDev, 2021-01-03:
Happy to help if you let me know what you're trying to do.

If not, I'll close the issue

lemmingDev, 2021-01-09:
Will close due to no response. 

Feel free to create another issue if you need help

closed 2021-01-09 by lemmingDev