lemmingDev/ESP32-BLE-Gamepad #34: Multiple button presses at once

Noplautikus, 2020-12-31:
I have a 3x4 matrix with 12 buttons and would like to press several buttons at the same time, but this does not work. I remember when I tested the same code about two months ago that it worked, did you change anything about this?

My Code looks like this:

`#include <ESP32Encoder.h>     // https://github.com/madhephaestus/ESP32Encoder/
#include <Keypad.h>           // https://github.com/Chris--A/Keypad
#include <BleGamepad.h>       // https://github.com/MagnusThome/ESP32-BLE-Gamepad

BleGamepad bleGamepad("AMG-Wheel", "SimWheels", 100);

#define ROWS 3
#define COLS 4
uint8_t rowPins[ROWS] = {13, 12, 14};
uint8_t colPins[COLS] = {27, 26, 25, 33};
uint8_t keymap[ROWS][COLS] = {
  {0,1,2,3},
  {4,5,6,7},
  {8,9,10,11}
};
Keypad customKeypad = Keypad( makeKeymap(keymap), rowPins, colPins, ROWS, COLS); 

#define MAXENC 2
uint8_t uppPin[MAXENC] = {19, 17};
uint8_t dwnPin[MAXENC] = {18, 16};
uint8_t encoderUpp[MAXENC] = {12, 14};
uint8_t encoderDwn[MAXENC] = {13, 15};
ESP32Encoder encoder[MAXENC];
unsigned long holdoff[MAXENC] = {0,0};
int32_t prevenccntr[MAXENC] = {0,0};
bool prevprs[MAXENC] = {0,0};
#define HOLDOFFTIME 150   // TO PREVENT MULTIPLE ROTATE "CLICKS" WITH CHEAP ENCODERS WHEN ONLY ONE CLICK IS INTENDED

void keypadEvent(KeypadEvent key){
  uint8_t keystate = customKeypad.getState();
  if (keystate==PRESSED)  { pressKey(key); }
  if (keystate==RELEASED) { releaseKey(key); }
}

void sendKey(uint8_t key) {
    uint32_t gamepadbutton = pow(2,key);      // CONVERT TO THE BINARY MAPPING GAMEPAD KEYS USE
    Serial.print("pulse\t");
    Serial.println(key);
    if(bleGamepad.isConnected()) {
      bleGamepad.press(gamepadbutton);
      delay(150);
      bleGamepad.release(gamepadbutton);
    }
}

void pressKey(uint8_t key) {
    uint32_t gamepadbutton = pow(2,key);      // CONVERT TO THE BINARY MAPPING GAMEPAD KEYS USE
    Serial.print("hold\t");
    Serial.println(key);
    if(bleGamepad.isConnected()) {
      bleGamepad.press(gamepadbutton);
    }
}

void releaseKey(uint8_t key) {
    uint32_t gamepadbutton = pow(2,key);      // CONVERT TO THE BINARY MAPPING GAMEPAD KEYS USE
    Serial.print("release\t");
    Serial.println(key);
    if(bleGamepad.isConnected()) {
      bleGamepad.release(gamepadbutton);
    }
}

void setup() {
  Serial.begin(115200);

  for (uint8_t i=0; i<MAXENC; i++) {
    encoder[i].clearCount();
    encoder[i].attachFullQuad(dwnPin[i], uppPin[i]);
  }
  customKeypad.addEventListener(keypadEvent);
  //customKeypad.setHoldTime(1);
  bleGamepad.begin();
  Serial.println("Booted!");
}

void loop() {

  unsigned long now = millis();

  // -- ROTARY ENCODERS : ROTATION -- //

  for (uint8_t i=0; i<MAXENC; i++) {
    int32_t cntr = encoder[i].getCount();
    if (cntr!=prevenccntr[i]) {
      if (!holdoff[i]) {
        if (cntr>prevenccntr[i]) { sendKey(encoderUpp[i]); }
        if (cntr<prevenccntr[i]) { sendKey(encoderDwn[i]); }
        holdoff[i] = now;
        if (holdoff[i]==0) holdoff[i] = 1;  // SAFEGUARD WRAP AROUND OF millis() (WHICH IS TO 0) SINCE holdoff[i]==0 HAS A SPECIAL MEANING ABOVE
      }
      else if (now-holdoff[i] > HOLDOFFTIME) {
        prevenccntr[i] = encoder[i].getCount();
        holdoff[i] = 0;
      }
    }
  }

  customKeypad.getKey();    // READ BUTTON MATRIX (EVENT CALLBACK SETUP)

  delay(10);
 
}`

And my schematic looks like this:

[image]

lemmingDev, 2020-12-31:
Hi

Are you using my library
https://github.com/lemmingDev/ESP32-BLE-Gamepad

or (according to the URL at the top of your code) a fork someone made of my library?
https://github.com/MagnusThome/ESP32-BLE-Gamepad

If, as your code indicates, you're using the fork, it has not been updated in 12 months. If you've now switched to mine, but it was working on the fork, then perhaps go back to the fork. I think the main functional difference between mine and the fork at the moment is that mine has higher precision on the axes, so if you're just using button presses, they should be interchangable.

What exactly is not working in your code? It seems to compile on my machine.
Are the buttons presses just not being sent as expected / at all?
Is it working for one button, but not multiple buttons at a time?

I see you have some serial degugging code that spits out when keys are held down. Is that working as expected when multiple keys are held down? 

I know the my library works fine for multiple button presses, as seen in the example code where button 32 is held down, and button 1 is toggled every 500ms, so I think the issue may lay in your code somewhere.

lemmingDev, 2020-12-31:
Also shouldn't

```
uint8_t encoderUpp[MAXENC] = {12, 14};
uint8_t encoderDwn[MAXENC] = {13, 15};
```
be changed to
```
uint8_t encoderUpp[MAXENC] = {13, 15};
uint8_t encoderDwn[MAXENC] = {14, 16};
```

as you've programmed your keypad is mapped to buttons 1~12, so I assume you want to have your encoders send buttons 13
~16

lemmingDev, 2020-12-31:
Here is some code to test/prove that my library supports multiple presses. Just upload it and you see every even numbered button (2, 4, 6, .. , 32) activate at the same time for 500ms, and then deactivate for 500ms repeatedly.

[image]

Test it and let me know

Noplautikus, 2020-12-31:
Yeah your code is working

lemmingDev, 2020-12-31:
Ok, well in that case I can only suggest that there is a problem in either your wiring or code you've written.

I have a keypad somewhere out in the garage. If I find it, I might be able to test your code.

Noplautikus, 2020-12-31:
That would be great

Noplautikus, 2020-12-31:
But if the wiring was wrong, I should not see any button presses, right?

Noplautikus, 2020-12-31:
And btw i used your library release 1.5

lemmingDev, 2021-01-01:
I can verify the issue is with your code. I am trying to fix now...

lemmingDev, 2021-01-01:
I think the issue is with trying to use KeyboardEvent

It seems to only pass one char

The getKeys() is also empty when that event fires, which seems to prevent it from being used to iterate through all keys that've had a state change on that event.

I think you might have to forego using the event, and just use getKeys() every loop.
You can write code to only send updates to the BLE Gamepad if getKeys() is populated with key changes, so it will still be pretty efficient. 

I will post some code soon...

lemmingDev, 2021-01-01:
Here. Try this code just to see if it works, and if it does, I can look at getting it to work with your code if you need help

Pin 33 didn't work for me - I changed mine to pin 4

```
#include <Keypad.h>
#include <BleGamepad.h> 

#define ROWS 3
#define COLS 4
uint8_t rowPins[ROWS] = {13, 12, 14};
uint8_t colPins[COLS] = {27, 26, 25, 33};
uint8_t keymap[ROWS][COLS] = {
{0,1,2,3},
{4,5,6,7},
{8,9,10,11}
};

//initialize an instance of class NewKeypad
Keypad kpd = Keypad( makeKeymap(keys), rowPins, colPins, ROWS, COLS); 

BleGamepad bleGamepad;

String msg = "";

void setup() {
    Serial.begin(9600);
    Serial.println("Starting BLE work!");
    bleGamepad.begin();
}

void loop() {
  if(bleGamepad.isConnected()) 
  {
    // Fills kpd.key[ ] array with up-to 10 active keys.
    // Returns true if there are ANY active keys.
    if (kpd.getKeys())
    {
        for (int i=0; i<LIST_MAX; i++)   // Scan the whole key list.
        {
            if ( kpd.key[i].stateChanged )   // Only find keys that have changed state.
            {
                switch (kpd.key[i].kstate) {  // Report active key state : IDLE, PRESSED, HOLD, or RELEASED
                    case PRESSED:
                    msg = " PRESSED.";
                    bleGamepad.press(pow(2,int(kpd.key[i].kchar)));
                break;
                    case RELEASED:
                    msg = " RELEASED.";
                    bleGamepad.release(pow(2,int(kpd.key[i].kchar)));
                }
                Serial.print("Key ");
                Serial.print(int(kpd.key[i].kchar));
                Serial.println(msg);
            }
        }
    }
    delay(20);
  } 
}  // End loop
```

lemmingDev, 2021-01-01:
Oh - and I found an updated version of the library here:

https://github.com/Nullkraft/Keypad/

lemmingDev, 2021-01-01:
Ok - Your complete (hopefully working) code is here:

```
#include <ESP32Encoder.h> // https://github.com/madhephaestus/ESP32Encoder/
#include <Keypad.h>       // https://github.com/Chris--A/Keypad
#include <BleGamepad.h>   // https://github.com/MagnusThome/ESP32-BLE-Gamepad

BleGamepad bleGamepad("AMG-Wheel", "SimWheels", 100);

#define ROWS 3
#define COLS 4
uint8_t rowPins[ROWS] = {13, 12, 14};
uint8_t colPins[COLS] = {27, 26, 25, 33};
uint8_t keymap[ROWS][COLS] = {
{0,1,2,3},
{4,5,6,7},
{8,9,10,11}
};

Keypad customKeypad = Keypad( makeKeymap(keymap), rowPins, colPins, ROWS, COLS);

#define MAXENC 2
uint8_t uppPin[MAXENC] = {19, 17};
uint8_t dwnPin[MAXENC] = {18, 16};
uint8_t encoderUpp[MAXENC] = {12, 14};
uint8_t encoderDwn[MAXENC] = {13, 15};
ESP32Encoder encoder[MAXENC];
unsigned long holdoff[MAXENC] = {0,0};
int32_t prevenccntr[MAXENC] = {0,0};
bool prevprs[MAXENC] = {0,0};
#define HOLDOFFTIME 150 // TO PREVENT MULTIPLE ROTATE "CLICKS" WITH CHEAP ENCODERS WHEN ONLY ONE CLICK IS INTENDED

void KeypadUpdate()
{
  customKeypad.getKeys();
  
  for (int i=0; i<LIST_MAX; i++)   // Scan the whole key list.
  {
      if ( customKeypad.key[i].stateChanged )   // Only find keys that have changed state.
      {
        uint8_t keystate = customKeypad.key[i].kstate;
  
        if (keystate==PRESSED) { pressKey(int(customKeypad.key[i].kchar)); }
        if (keystate==RELEASED) { releaseKey(int(customKeypad.key[i].kchar)); }
      }
  }

}

void sendKey(uint8_t key) 
{
  uint32_t gamepadbutton = pow(2,key); // CONVERT TO THE BINARY MAPPING GAMEPAD KEYS USE
  Serial.print("pulse\t");
  Serial.println(key);
  if(bleGamepad.isConnected()) 
  {
    bleGamepad.press(gamepadbutton);
    delay(150);
    bleGamepad.release(gamepadbutton);
  }
}

void pressKey(uint8_t key) 
{
  uint32_t gamepadbutton = pow(2,key); // CONVERT TO THE BINARY MAPPING GAMEPAD KEYS USE
  Serial.print("hold\t");
  Serial.println(key);
  
  if(bleGamepad.isConnected()) 
  {
    bleGamepad.press(gamepadbutton);
  }
}

void releaseKey(uint8_t key) 
{
  uint32_t gamepadbutton = pow(2,key); // CONVERT TO THE BINARY MAPPING GAMEPAD KEYS USE
  Serial.print("release\t");
  Serial.println(key);
  
  if(bleGamepad.isConnected()) 
  {
    bleGamepad.release(gamepadbutton);
  }
}

void setup() {
  Serial.begin(115200);
  
  for (uint8_t i=0; i<MAXENC; i++) 
  {
    encoder[i].clearCount();
    encoder[i].attachFullQuad(dwnPin[i], uppPin[i]);
  }
  
  //customKeypad.setHoldTime(1);
  bleGamepad.begin();
  Serial.println("Booted!");
}

void loop() 
{
  unsigned long now = millis();
  
  // -- ROTARY ENCODERS : ROTATION -- //
  
  for (uint8_t i=0; i<MAXENC; i++) 
  {
    int32_t cntr = encoder[i].getCount();
    
    if (cntr!=prevenccntr[i]) 
    {
      if (!holdoff[i]) 
      {
        if (cntr>prevenccntr[i]) { sendKey(encoderUpp[i]); }
        if (cntr<prevenccntr[i]) { sendKey(encoderDwn[i]); }
        holdoff[i] = now;
        if (holdoff[i]==0) holdoff[i] = 1; // SAFEGUARD WRAP AROUND OF millis() (WHICH IS TO 0) SINCE holdoff[i]==0 HAS A SPECIAL MEANING ABOVE
      }
      else if (now-holdoff[i] > HOLDOFFTIME) 
      {
        prevenccntr[i] = encoder[i].getCount();
        holdoff[i] = 0;
      }
    }
  }

  KeypadUpdate();
  
  delay(10);
}
```

lemmingDev, 2021-01-01:
BTW - 2 simultaneous buttons work perfectly

Depending on which combination or rows/columns, some combinations of 3 or 4 can work, but others will have ghost keys pressed.

Greater than 4 always has ghost presses.

Noplautikus, 2021-01-01:
Thanks man i testet it and it worked, yeah the ghosting is not a problem but i could use some diodes to prevent that.

lemmingDev, 2021-01-01:
You're welcome

closed 2021-01-01 by lemmingDev