lemmingDev/ESP32-BLE-Gamepad #35: Changing controller name?

chemic, 2021-01-02:
Is it possible to change the name of the controller anyhow?
[image]

Thanks for the great library.. it made it to my simracing custom Ferrari 488 wheel 💯

lemmingDev, 2021-01-02:
Hi

I'm not sure how.
I have asked around...
https://github.com/nkolban/esp32-snippets/issues/230#issuecomment-735285688

closed 2021-01-02 by lemmingDev

lishuai199502, 2022-10-29:
Hi
Maybe these pages can help
https://learn.microsoft.com/en-us/answers/questions/401236/hidd-getproductstring-with-ble-hid-device.html
https://learn.microsoft.com/en-us/answers/questions/406840/hidd-getproductstring-with-classic-bluetooth-devic.html
https://learn.microsoft.com/zh-CN/windows-hardware/drivers/ddi/hidsdi/nf-hidsdi-hidd_getproductstring
It seems like the ble not support，but classic bt can set the Service Name in SDP protocol.