lemmingDev/ESP32-BLE-Gamepad #36: Issues on android

palsayantan, 2021-01-04:
first of thanks to @lemmingDev for the awesome library.

I made a [hardware](https://youtu.be/dODmsoAu0D4) that looks like PS5 controller and supports BLE_GamePad library. although I was having couple of issues while testing it on android.

1.  setAxis works works when it configured like this
(Left Thumb X, Left Thumb Y, Left Trigger, Right Trigger, Right Thumb X, Right Thumb Y, Hat switch position);
2. trigger buttons also not working when it defined under setAxis, instead I use button_7 and button_8 to make it work
3.  Right Thumb X & Right Thumb Y should be mapped to (255, 0)
4.  is there any way to use Hat switches independently from setAxis?

```
    if (currentButton1State != previousButton1State)
    {
      if(currentButton1State == LOW)
      {
        bleGamepad.press(BUTTON_1);
      }
      else
      {
        bleGamepad.release(BUTTON_1);
      }
    }
    previousButton1State = currentButton1State;
  }
```
this doesn't act as a toggle button

lemmingDev, 2021-01-04:
Is it working on Windows, or are the same issues present?

LeeNX, 2021-01-05:
I saw a ```SetHat``` in a fork - https://github.com/Peluko/ESP32-BLE-Gamepad/commit/9f42f703fc9b53011f3f2f7aabb345dd0410d12c. 
  
If that helps, but might be better to get a PR from @Peluko?

palsayantan, 2021-01-05:
with the same setup in windows only D-pad and trigger buttons are working but hat and joysticks aren't. I use this website for testing [Gamepadviewer](https://gamepadviewer.com/). is there any app or something where I can test?

palsayantan, 2021-01-05:
Also one of my viewer suggested an app called BatON on Android, which can shows the battery level

lemmingDev, 2021-01-05:
Hi

Lets get it working in Windows first, and then go from there.

To test the library is working, please upload the following code, and test in Windows using "Set up USB game controllers"

You should see:
Buttons 1 and 32 pressed, all axes move to max and DPAD set to down right
then a 500ms delay, and then:
Buttons 1 released, all axes move to min and DPAD set to centered

```
#include <BleGamepad.h> 

BleGamepad bleGamepad;

void setup() 
{
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  bleGamepad.begin();
}

void loop() 
{
  if(bleGamepad.isConnected()) 
  {
    Serial.println("Press buttons 1 and 32. Move all axes to max. Set DPAD to down right.");
    bleGamepad.press(BUTTON_1);
    bleGamepad.press(BUTTON_32);
    bleGamepad.setAxes(32767, 32767, 32767, 32767, 255, 255, DPAD_DOWN_RIGHT);
    delay(500);

    Serial.println("Release button 1. Move all axes to min. Set DPAD to centred.");
    bleGamepad.release(BUTTON_1);
    bleGamepad.setAxes(-32767, -32767, -32767, -32767, 0, 0, DPAD_CENTERED);
    delay(500);
  }
}
```

lemmingDev, 2021-01-06:
Here is some code I just wrote to test all functions.

I paired it to an Android phone and was able to see everything working (the app I used only saw the first 16 buttons)

```
/*
 * Test all gamepad buttons, axes and dpad 
*/

#include <BleGamepad.h> 

BleGamepad bleGamepad;

void setup() 
{
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  bleGamepad.begin();
}

void loop() 
{
  if(bleGamepad.isConnected()) 
  {    
    Serial.println("Press all buttons one by one");
    for(int i = 0 ; i < 32 ; i += 1)
    {
      bleGamepad.press(pow(2, i));
      delay(200);
      bleGamepad.release(pow(2, i));
      delay(50);
    }

    Serial.println("Move all axis simultaneously from min to max");
    for(int i = -127 ; i < 128 ; i += 1)
    {
      bleGamepad.setAxes(i*256, i*256, i*256, i*256, i+127, i+127, 0);
      delay(50);
    }

    Serial.println("Send all dpad one by one in an anticlockwise rotation");
    for(int i = 8 ; i>= 0 ; i--)
    {
      bleGamepad.setAxes(0,0,0,0,0,0, i);
      delay(500);
    }
  }
}
```

lemmingDev, 2021-01-06:
BTW - that gamepad viewer website doesn't show values for the axes for me when the gamepad is connected. It's just some Javascript library, so I'm not too phased it's not showing axes changes. Button presses display fine

LeeNX, 2021-01-06:
I like using https://gamepad-tester.com/ for testing, has a little more tech details in.  
  
As for BatON - https://play.google.com/store/apps/details?id=com.limi.baton - this is what I have used, see the attached cropped screenshots from my phone, battery closest to the time is the device battery been reported. 
[image]
 
[image]
  
@lemmingDev thanks for the above testing code, will come in handy, might be working adding as a more complete testing example in repo.

palsayantan, 2021-01-06:
Thanks for the test code. I have tested in windows using "Set up USB game controllers" and it is showing buttons as well as the joysticks accurately. But how to test the trigger buttons?

Peluko, 2021-01-06:
> I saw a `SetHat` in a fork - [Peluko@9f42f70](https://github.com/Peluko/ESP32-BLE-Gamepad/commit/9f42f703fc9b53011f3f2f7aabb345dd0410d12c).
> 
> If that helps, but might be better to get a PR from @Peluko?

The project has diverged somewhat since my fork. If you continue interested, I could reintegrate and create the PR.

Regards.

lemmingDev, 2021-01-07:
@Peluko Thanks for the offer.

I think the easiest way to improve the library would be to create 7 variables for the values (X, Y, Z, rZ, rY, rZ and the hat), and then create new functions (such as: setHat, setLeftStick, setRightStick, setLeftTrigger, setRightTrigger) that update those variables.

Then, when any button or axes functions are called, the hid report that is sent can contain the values of those variables.
The buttons already behave in this fashion, however, every time a button press is sent, it currently sends zeros for all of the axes.

I could also make an option to make the library require a specific function call to send the HID report (like in the loop if there are any value changes), instead of having it send a report on every button or axis change. It would be more efficient and cut down un unnecessary HID reports.

I may get around to it tonight...

lemmingDev, 2021-01-07:
New release is out with all of the features I mentioned in the post above.

https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/0.1.6

As for your question "But how to test the trigger buttons?"

There are no trigger buttons defined in the HID. You just set any of the 32 available buttons. 
If you have actual analog triggers, you could write code that made a specific buttons activate when the trigger reached a certain threshold.

lemmingDev, 2021-01-08:
Oh - seems  https://gamepad-tester.com/ shows the axes values fine when using Firefox on Windows, but not on Chrome / Edge / IE etc

lemmingDev, 2021-01-13:
Here is a Windows app that can show the additional buttons (button 33~64) that this controller now has in the latest version.
http://www.planetpointy.co.uk/joystick-test-application/
The standard Windows one was able to show everything else fine

closed 2021-01-13 by lemmingDev