lemmingDev/ESP32-BLE-Gamepad #40: Would it be possible to add more than 32 buttons?

Netmav, 2021-01-12:
Hi.
Thanks for sharing this sweet library! :-)

I hope it's OK to ask this here:
I'm using the BLE Gamepad inside a 3D printed F1 steering wheel for my racing simulator rig.
I could really use 64 buttons instead of 32, to be able to interface all switches and 12 position rotary switches.
I've managed to get all 32 buttons working by using the first 16 buttons with the I/O pins on the ESP32, and the last 16 buttons connected to a couple of 74HC165 shift registers connected to the ESP32. I could easily add more shift registers to be able to read more buttons.

Ideally I would have just the 64 buttons (no need for axis or DPADs), but the important thing for me is to get the 64 buttons.

I'm pretty new to coding, and I have no idea how to alter the library to get the extra 32 buttons.

Thanks.

\Martin

lemmingDev, 2021-01-12:
Why yes - it is possible to add more buttons

Here is a version of the library I made just for you that has 64 buttons!

All of the same features/axes/hat switch are still in there. The only difference is that it has 64 buttons.

I have attached the library and a test app for it

Upload the TestAll sketch and rejoice

[ESP32-BLE-Gamepad-64Buttons.zip](https://github.com/lemmingDev/ESP32-BLE-Gamepad/files/5804949/ESP32-BLE-Gamepad-64Buttons.zip)

[joysticktest_092.zip](https://github.com/lemmingDev/ESP32-BLE-Gamepad/files/5804951/joysticktest_092.zip)

Please let me know if it works ok, especially when setting buttons 33~64

lemmingDev, 2021-01-13:
I have now released an official version for everyone.

Library now additionally supports:
- 2 sliders
- 64 buttons (up from 32)
- 16bit on all axes (left/right triggers and sliders are unsigned, left/right thumbsticks are centered on zero)
- 4 hat switches
- updated examples

LeeNX, 2021-01-13:
Nice! Thanks for the release @lemmingDev ... What is the trade off with so many buttons and bits? Slow response time or less controllers? Is there not a limited about of RF?

Netmav, 2021-01-13:
Thanks @lemmingDev! 
This is very much appreciated.

I'll give it test right away.

\Martin

lemmingDev, 2021-01-13:
I could actually go to 128 buttons, but I don't really see a need.

Performance seems to be fine

I also have a fork that uses a more efficient Bluetooth library if you wanted to check it out

https://github.com/lemmingDev/ESP32-BLE-Gamepad/tree/NimBLE

Netmav, 2021-01-13:
> Why yes - it is possible to add more buttons
>
> Here is a version of the library I made just for you that has 64 buttons!
>
> All of the same features/axes/hat switch are still in there. The only [...]

I just tested this now, and I can't get it to work.
When I update the library and upload my sketch, none of the buttons are registered by Windows or JoystickTest anymore.
(This applies to both the "standard" digital inputs and the inputs from the shift registers).

If I go back to an older version of the library (I tested with version 1.5), everything works again.

Is there anything I should change for the newest version, in order to get it to work?

Thanks.
\Martin

lemmingDev, 2021-01-13:
Does the test code work?

Attach your code...

lemmingDev, 2021-01-13:
Oh, and use the latest version from github, not the one I attached...

Netmav, 2021-01-14:
> Does the test code work?
> 
> Attach your code...

I just tested the examples, but noe of those work (not for me, anyway).
This time I went back to version 1.7 after testing the 1.8 version. This one also works for me (but only 32 buttons of course).

LeeNX, 2021-01-14:
I just tried v1.8.0, downloaded in the Arduino Library Manager and on OSX only seeing 32 buttons.  
  
Is there a way to report the lib version at setup into the serial port? I did have a problem where the download zip for v1.7.0 was installed too.

lemmingDev, 2021-01-14:
Because there has been a change to the number of buttons, you'll have to remove/forget/unpair the bluetooth device, and then re-pair it.

The extra buttons should then show up

At least, that's what you have to do in Windows...

Might pay to also delete all versions you have in your libraries folder and reinstall from the libraries manager.

Let me know if it works for you

lemmingDev, 2021-01-14:
Here is a test of the latest version

As you can see, the Windows gamepad tester on the right only shows the first 32 buttons, but the 3rd party tester on the left shoes them all...

[GamepadTest.zip](https://github.com/lemmingDev/ESP32-BLE-Gamepad/files/5816878/GamepadTest.zip)

lemmingDev, 2021-01-14:
> Is there a way to report the lib version at setup into the serial port? I did have a problem where the download zip for v1.7.0 was installed too.

Found the issue with multiple versions being installed (github folder naming vs library manager naming)
Will be fixed shortly

https://github.com/arduino/Arduino/issues/11162

LeeNX, 2021-01-15:
Thanks @lemmingDev , I will try that later today. Don't remember having to do that for 16 to 32 buttons, but rather try nice and clean. I did clean up the libraries. I only noticed the libraries problem when the TestAll example did not update.

Netmav, 2021-01-15:
Thanks @lemmingDev 

I just tried this, with the 1.9 version, and after I unpaired the BLE Gamepad, and repaired it, everything works like a charm! :-)

Thanks again. I really appreciate it.
Have a good one!

\Martin

lemmingDev, 2021-01-15:
Pleasure

So, when the 2.0 version hits the library manager, it will be under a different name (ESP32-BLE-Gamepad as opposed to ESP32 BLE Gamepad), so you'll need to manually delete the ESP32_BLE_Gamepad folder from your libraries folder, and then search for and install ESP32-BLE-Gamepad , which will create a ESP32-BLE-Gamepad  folder. From then on, it should be smooth sailing...

LeeNX, 2021-01-17:
@lemmingDev you aree correct, deleting all the older BLE Game devices in the Bluetooth manager and I got the below
```
leet-mbp:build leet$ ./sdl2-jstest -l
Found 1 joystick(s)

Joystick Name:     'Espressif Unknown'
Joystick GUID:     0300000002e50000abbb000001100000
Joystick Number:    0
Number of Axes:     8
Number of Buttons: 64
Number of Hats:     4
Number of Balls:    0
GameControllerConfig:
  missing (see 'gamecontrollerdb.txt' or SDL_GAMECONTROLLERCONFIG)
```
Then testing, sorry for the big black screen, but this is the only way to get all axes, buttons and hats on the same screen. All works as expected with the TestAll stretch.
[image]

lemmingDev, 2021-01-17:
Good to see its working for everyone...

witnessmenow, 2021-01-21:
Latest released version on Github version is not working for me, I can't get it to connect to my Windows Machine. I deleted old instances of the connection before trying to connect it, but on the Add a device screen I just "Try connecting your device again"

lemmingDev, 2021-01-22:
@witnessmenow 

Is that with your own code or one of the known working examples?

lemmingDev, 2021-01-22:
If you wanted, you could also try the more efficient NimBLE library version of my library
https://github.com/lemmingDev/ESP32-BLE-Gamepad/tree/NimBLE
Make sure the delete the current version of my library, then install the one linked above from Github, and install NimBLE-Arduino from the library manager

If you give it a go, let me know how you go

witnessmenow, 2021-01-22:
Hey,

It was my one, thinking about it this morning I wonder if having 8
denouncer objects is causing problems ,I was just testing it out, but it
was a little unresponsive

https://github.com/witnessmenow/esp32-gamepad

lemmingDev, 2021-01-22:
Not sure if the debounce objects are absolutely necessary due to the use of pullups

I have added a new example that may be of some interest
https://github.com/lemmingDev/ESP32-BLE-Gamepad/tree/master/examples/MultipleButtons

lemmingDev, 2021-01-22:
In any case, I've now also added a n example that shows how to use multiple buttons with debounce

https://github.com/lemmingDev/ESP32-BLE-Gamepad/tree/master/examples/MultipleButtonsDebounce

closed 2021-01-28 by lemmingDev