lemmingDev/ESP32-BLE-Gamepad #42: Button stayed pressed for several seconds after button is physically released on android

blingingjak, 2021-01-20:
Ok so im a total newbie on Arduino and coding in it, if this is not the right play to ask this could someone suggest a better place to ask?
So I am trying to create my own Bluetooth controller for android and I am currently running in to the issue that when I attempt to map a button in the "MyOldBoy" app to one of the buttons on my  ESP32 it accepts the press (most of the time) but it then keeps the button pressed for anywhere from 3-8 seconds. In the windows arduino IDE when using the serial monitor I can press and hold a button but if I left go it doesnt accept any other button inputs for about 3-8 .

Here is the code I am running

`
#include <BleGamepad.h>   // https://github.com/lemmingDev/ESP32-BLE-Gamepad

BleGamepad bleGamepad;

int previousButton1State = HIGH;

void setup() 
{
  pinMode(4, INPUT_PULLUP);
  pinMode(16, INPUT_PULLUP);
  pinMode(17, INPUT_PULLUP);
  pinMode(18, INPUT_PULLUP);
  pinMode(19, INPUT_PULLUP);
  pinMode(23, INPUT_PULLUP);
  pinMode(25, INPUT_PULLUP);
  pinMode(26, INPUT_PULLUP);
  pinMode(27, INPUT_PULLUP);
  pinMode(32, INPUT_PULLUP);
  bleGamepad.begin();
  Serial.begin(115200);
}

void loop() 
{
  if(bleGamepad.isConnected()) 
  {

    int currentButton1State = digitalRead(4);

    if (currentButton1State != previousButton1State)
    {
      if(currentButton1State == LOW)
      {
        bleGamepad.press(BUTTON_1);
        Serial.println("Button A is pushed.");
      }
      else
      {
       bleGamepad.release(BUTTON_1);
      }
    }
    previousButton1State = currentButton1State;
  }

  delay(20);
  
  if(bleGamepad.isConnected()) 
  {

    int currentButton1State = digitalRead(16);

    if (currentButton1State != previousButton1State)
    {
      if(currentButton1State == LOW)
      {
        bleGamepad.press(BUTTON_2);
        Serial.println("Button B is pushed.");
      }
      else
      {
      bleGamepad.release(BUTTON_2);
      }
    }
    previousButton1State = currentButton1State;
  }
  delay(20);

  if(bleGamepad.isConnected()) 
  {

    int currentButton1State = digitalRead(17);

    if (currentButton1State != previousButton1State)
    {
      if(currentButton1State == LOW)
      {
        bleGamepad.press(BUTTON_3);
        Serial.println("Button START is pushed.");
      }
      else
      {
       bleGamepad.release(BUTTON_3);
      }
    }
    previousButton1State = currentButton1State;
  }
  delay(20);

  if(bleGamepad.isConnected()) 
  {

    int currentButton1State = digitalRead(18);

    if (currentButton1State != previousButton1State)
    {
      if(currentButton1State == LOW)
      {
        bleGamepad.press(BUTTON_4);
        Serial.println("Button SELECT is pushed.");
      }
      else
      {
       bleGamepad.release(BUTTON_4);
      }
    }
    previousButton1State = currentButton1State;
  }
  delay(20);

  if(bleGamepad.isConnected()) 
  {

    int currentButton1State = digitalRead(19);

    if (currentButton1State != previousButton1State)
    {
      if(currentButton1State == LOW)
      {
        bleGamepad.press(BUTTON_5);
        Serial.println("Button UP is pushed.");
      }
      else
      {
      bleGamepad.release(BUTTON_5);
      }
    }
    previousButton1State = currentButton1State;
  }
 delay(20);

   if(bleGamepad.isConnected()) 
  {

    int currentButton1State = digitalRead(23);

    if (currentButton1State != previousButton1State)
    {
      if(currentButton1State == LOW)
      {
        bleGamepad.press(BUTTON_6);
        Serial.println("Button DOWN is pushed.");
      }
      else
      {
       bleGamepad.release(BUTTON_6);
      }
    }
    previousButton1State = currentButton1State;
  }
  delay(20);

   if(bleGamepad.isConnected()) 
  {

    int currentButton1State = digitalRead(25);

    if (currentButton1State != previousButton1State)
    {
      if(currentButton1State == LOW)
      {
        bleGamepad.press(BUTTON_7);
        Serial.println("Button LEFT is pushed.");
      }
      else
      {
       bleGamepad.release(BUTTON_7);
      }
    }
    previousButton1State = currentButton1State;
  }
  delay(20);

   if(bleGamepad.isConnected()) 
  {

    int currentButton1State = digitalRead(26);

    if (currentButton1State != previousButton1State)
    {
      if(currentButton1State == LOW)
      {
        bleGamepad.press(BUTTON_8);
        Serial.println("Button RIGHT is pushed.");
      }
      else
      {
        bleGamepad.release(BUTTON_8);
      }
    }
    previousButton1State = currentButton1State;
  }
  delay(20);

   if(bleGamepad.isConnected()) 
  {

    int currentButton1State = digitalRead(27);

    if (currentButton1State != previousButton1State)
    {
      if(currentButton1State == LOW)
      {
        bleGamepad.press(BUTTON_9);
        Serial.println("Button L1 is pushed.");
      }
      else
      {
       bleGamepad.release(BUTTON_9);
      }
    }
    previousButton1State = currentButton1State;
  }
  delay(20);

   if(bleGamepad.isConnected()) 
  {

    int currentButton1State = digitalRead(32);

    if (currentButton1State != previousButton1State)
    {
      if(currentButton1State == LOW)
      {
        bleGamepad.press(BUTTON_10);
        Serial.println("Button R1 is pushed.");
      }
      else
      {
        bleGamepad.release(BUTTON_10);
      }
    }
    previousButton1State = currentButton1State;
  }
  delay(20);

}
`

lemmingDev, 2021-01-21:
Here you go

```
/*
 * This code programs a number of pins on an ESP32 as buttons on a BLE gamepad
 * 
 * It uses arrays to cut down on code
 *
 * Before using, adjust the numOfButtons, buttonPins and physicalButtons to suit your senario
 *
 */
 
#include <BleGamepad.h> // https://github.com/lemmingDev/ESP32-BLE-Gamepad

BleGamepad bleGamepad;

#define numOfButtons 10

byte previousButtonStates[numOfButtons];
byte currentButtonStates[numOfButtons];
byte buttonPins[numOfButtons] = { 4, 16, 17, 18, 19, 23, 25, 26, 27, 32 };
byte physicalButtons[numOfButtons] = { 1, 2, 3, 4, 5, 6, 7, 8, 9, 10 };

void setup()
{
  for (byte currentPinIndex = 0 ; currentPinIndex < numOfButtons ; currentPinIndex++)
  {
    pinMode(buttonPins[currentPinIndex], INPUT_PULLUP);
    previousButtonStates[currentPinIndex] = HIGH;
    currentButtonStates[currentPinIndex] =  HIGH;
  }

  bleGamepad.begin(false);
  Serial.begin(115200);
}

void loop()
{
  if(bleGamepad.isConnected())
  {
    for (byte currentIndex = 0 ; currentIndex < numOfButtons ; currentIndex++)
    {
      currentButtonStates[currentIndex]  = digitalRead(buttonPins[currentIndex]);

      if (currentButtonStates[currentIndex] != previousButtonStates[currentIndex])
      {
        if(currentButtonStates[currentIndex] == LOW)
        {
          bleGamepad.press(pow(2, physicalButtons[currentIndex] - 1));
          Serial.print("Button ");
          Serial.print(physicalButtons[currentIndex]);
          Serial.println(" pushed.");
        }
        else
        {
          bleGamepad.release(pow(2, physicalButtons[currentIndex] - 1));
          Serial.print("Button ");
          Serial.print(physicalButtons[currentIndex]);
          Serial.println(" released.");
        }
      } 
    }
    
    if (currentButtonStates != previousButtonStates)
    {
      for (byte currentIndex = 0 ; currentIndex < numOfButtons ; currentIndex++)
      {
        previousButtonStates[currentIndex] = currentButtonStates[currentIndex]; 
      }
      
      bleGamepad.sendReport();
    }
    
    delay(20);
  }
}
```

blingingjak, 2021-01-21:
Holy shit, thanks man. I know my copy paste method was messy and needlessly long but I'm learning. I will try to see if I can get my current code to work with the suggestions you made and then also look at your code to see the difference and how to improve, thanks so much. I really appreciate this.

lemmingDev, 2021-01-21:
NP

Well, for now, just load my code up and it should work.

Let me know how it goes.

After that, you can go back any try to edit your code for educational purposes I guess

LeeNX, 2021-01-21:
I have done a similar build for a joystick conversion. Might want to add debounce too. Mine is a work in progress and I can't say I have had much time in the last two weeks ... Need to sort out the button press/release part.  
  
https://gitlab.com/leet/diy-gamepad/-/blob/master/Conversion/ESP-BLE-Joy-Conversion.ino

lemmingDev, 2021-01-21:
@LeeNX 
Thanks for chiming in.

Not sure you really need debounce as it's it using pullup resistors anyway, but I do provide an example with the bounce2 library called SingleButtonDebounce for anyone who needs one.

Might be interesting to test out Street Fighter IV or something with and without it to see if there is any difference. It would add to the latency slightly you would think. Then again, the ESP32 is a beast, so it mightn't.

Haven't really analysed your code, but don't forget that if you're using my updated library, you'll need to change your code to account for the extra 2 sliders
bleGamepad.setAxes(0, 0, 0, 0, 0, 0, BUTTON_BLE[i]);  --> bleGamepad.setAxes(0, 0, 0, 0, 0, 0, 0, 0, BUTTON_BLE[i]);
and
bleGamepad.setAxes(0, 0, 0, 0, 0, 0, DPAD_CENTERED);  --> bleGamepad.setAxes(0, 0, 0, 0, 0, 0, 0, 0, DPAD_CENTERED);

or better yet
bleGamepad.setAxes(0, 0, 0, 0, 0, 0, BUTTON_BLE[i]);  --> bleGamepad.setHat1(BUTTON_BLE[i]);
and
bleGamepad.setAxes(0, 0, 0, 0, 0, 0, DPAD_CENTERED);  --> bleGamepad.setHat1(DPAD_CENTERED);

Let me know if you can't get it working/need a hand with your code

I'm by no means an expert (and have limited experience with C++)

I guess that's evidenced by the fact I used byte vs uint_8 for this example, which are basically exactly the same in C++ (0~255) but I'm used to using the byte syntax from other languages

It's pretty easy to pick up though

blingingjak, 2021-01-21:
Hi Lemming Dev your code works flawlessly. Thank you so much for solving my issue

lemmingDev, 2021-01-22:
@blingingjak 

Glad it's working.

Thanks for the offer. Just pay it forward. Help those you can...

lemmingDev, 2021-01-22:
I have now added an example with multiple buttons and debounce

https://github.com/lemmingDev/ESP32-BLE-Gamepad/tree/master/examples/MultipleButtonsDebounce

closed 2021-01-28 by lemmingDev