lemmingDev/ESP32-BLE-Gamepad #43: [Feature Request] Support Player LED support?

LeeNX, 2021-01-24:
Please can we look at possible support some like Player ID LED Support, something like the PS3 LED ```1 to 4``` or the PS4 Colour bar?  
  
I have looked at this and my very basic C/BLE knowledge has me banging my head against a very hard brick wall ;-\ 
  
Some of my investigation with some suggests, like possible using the keyboard LED ```none-support``` - https://github.com/T-vK/ESP32-BLE-Keyboard/blob/928c09320e3e6e344a63640c7de21d52a20c1d47/KeyboardOutputCallbacks.cpp#L14. 
  
but I would not know how to use this ... Or I mean, I don't see how this hooks back into the code base, so I can connect LEDs to the reply, or turn this into a function to poll the LEDs as needed.  
  
Making a branch for SDL2, with a Generic Controller, using HID over BLE, to setup Player ID - https://github.com/LeeNX/SDL-mirror/tree/generic

lemmingDev, 2021-01-25:
Just tried - and failed

I was able to add the Player 1, Player 2, Player 3 and Player 4 LEDs as outputs to the report HID and then registered to listen for the callbacks, however Windows nor Android didn't seem to send those notifications.

It still functions perfectly, but doesn't receive any events.

Is it even supported in DirectInput?

If someone more knowledgeable would like to troubleshoot it, I have included the modified library

[ESP32_BLE_GamepadPlayerLED.zip](https://github.com/lemmingDev/ESP32-BLE-Gamepad/files/5870272/ESP32_BLE_GamepadPlayerLED.zip)

LeeNX, 2021-01-26:
@lemmingDev I don't think you failed ... I am not sure that the Player Feedback is a standard supported feature, same as haptic or force feedback. I believe that SDL2 add support by using HIDAPI services.  
  
My SDL generic branch, meant to Generic Controller, but that is very poorly chosen name, looks for the VID/PID of the ESP32 device and then ```hooks``` it, but need to work out the details, I only have it compiling and we looking at a large lib and still need an easy tool to call the Player LEDs function.  
  
I am going to look at the following functions in SDL2
  * SDL_JoystickGetSerial()
  * SDL_JoystickRumbleTriggers()
  * SDL_JoystickHasLED()
  * SDL_JoystickSetLED()
 
If you are interested, I think the basic changes would be in https://github.com/LeeNX/SDL-mirror/blob/f58059d5e8847264bd8ee7a285a32ea998fbe4d9/src/joystick/hidapi/SDL_hidapi_generic.c

Thou a simple python HID app that can send and receive ```HID``` packets might be a quick way to test functionality, going to start with https://www.ontrak.net/pythonhidapi.htm as a base.  
  
I have not yet looked at the code yet ... Might be an idea to pop into a git branch and any updates could be tracked. Maybe also need a the correct HID Cap ```PID Page (0x0F)```?  ... Truly just guessing again.  
  
Thanks

lemmingDev, 2021-01-26:
I probably won't spend any more time playing around with it as it's a non-critical feature. 
I did use the correct page 0x0F, but just think it's not supported in the generic driver by Windows.

Good luck with it, and feel free to make pull requests if you get anything working

WarlockSVG, 2021-01-28:
DS4windows is able to control the LED color of an PS4 controller... maybe you can find there something that helps. ( https://github.com/Ryochan7/DS4Windows )

T-vK, 2021-01-30:
I don't know if this could help. But maybe the code for the CapsLock, NumLock, ScrollLock LEDs for keyboards can be adapted for the Gamepads: https://github.com/T-vK/ESP32-BLE-Keyboard/pull/70

LeeNX, 2021-01-30:
@T-vK that had been my train of thought. I think we have some progress, but now we need a way to test the LED support from the PC, but I don't think there is a standard support way to do, unlike keyboard LEDs, which I think is pretty standard.

LeeNX, 2021-02-07:
I think that Windows might support the Player Index number - https://docs.microsoft.com/en-us/windows/win32/api/xinput/nf-xinput-xinputgetstate and https://docs.microsoft.com/en-us/windows/win32/api/xinput/nf-xinput-xinputsetstate. 
  
@lemmingDev would this worth putting some time in? I am still working on adding support in SDL2, but it's pretty slow going form me.  
  
Think this is worth re-opening? Thou I am not sure how the HID would hook XInput ... Looks information still to find and understand.

lemmingDev, 2021-02-07:
@LeeNX 
Seems it would need some type of app running the the background to get that functionality.
Not worth the hassle in my opinion.

There's also an Arduino project that turns some boards with USB HID functionality into xinput controllers with LED and rumble functionality, but I wouldn't know where to start getting it working on the ESP32.

LeeNX, 2021-02-08:
Thanks @lemmingDev ... I was reading some more last night add see that XInput is meant for XBox360 Controllers and all other joysticks/gamepad type devices are normally connect via DirectInput. Basically XInput is Microsoft only or partner interface, not a standard nor open sourced.  
  
I was hoping there would be a supported or open standard to take advantage of. Thou ```libSDL``` is using HIDAPI to add extra functionality, which is then mostly accessibly on other platform, like OSX and Linux or even ARM based systems like Raspberry Pi.  
  
Player Index (LED) just seemed like an easy step toward haptic and forcefeedback and seems to work for BLE Keyboard, but the OS is hooking into the CapLock/NumLock feedback.  
  
I am still trying to get HIDAPI working for me, so that I can build a few tests. Just so many layers and I don't think I remotely understand much of it or GLE GATT stuff, that is why your library is so super useful, thanks for that!

LeeNX, 2022-10-23:
@lemmingDev I was hoping we could possible re-open this feature request.  
  
I have been hacking away at getting some sort of feedback via BlueTooth back into the ESP32 over HID/GATT/HOG. I have not worked out all the details, but was finally able to get my test ESP32 and gamepad working, with me been able to write back.

https://github.com/LeeNX/ESP32-BLE-Gamepad/tree/test-led

I have shoehorned in a Usage PAGE LEDs and can get it reporting via the serial monitoring. I have made some notes, which can be seen at the end of the https://github.com/LeeNX/ESP32-BLE-Gamepad/tree/test-led#troubleshooting section.  
  
Going to see if I can get some feedback from the libSDL community.

lemmingDev, 2025-01-31:
Was there any movement on this?

Should we keep it open, or close it?

LeeNX, 2025-02-01:
I am still working on this, but there is no Android gamepad (what alot of host seen the ESP32-BLE-Gamepad device as) implementation that I can investigate against.

I have a few other ideas to poke at, hopefully I will get some time this weekend.

Please can we keep this open. Thanks.

lemmingDev, 2025-02-01:
FYI - I'm playing around with some other stuff now

Am almost finishing adding gyro and accelerometer, and fixing the right thumbstick issue

Trying to go through the list of outstanding things

With the new force pairing mode, ability to delete bonds, setting of TX level and other things done over the past few days, it should make the library more useful for people...

LeeNX, 2025-02-01:
> FYI - I'm playing around with some other stuff now

 ESP32-BLE-Gamepad related or just in general?  I have a related educational project, hoping to use stuff learned here to make better battle bots.  How is your new place? Gone doing any site seeing?
 
> Am almost finishing adding gyro and accelerometer, 

That would be interesting. As IMU data or extended axis?

> and fixing the right thumbstick issue

I don't remember this.

> Trying to go through the list of outstanding things

I tried to poke opened issues for feedback at the beginning of the year too.  I think closing stale issues is a good idea.

> With the new force pairing mode, ability to delete bonds, setting of TX power level and other things done over the past few days, it should make the library more useful for people...

I think the library has been pretty useful, but making projects that are more feature complete and almost on par with commercialised products is just awesome. I also think we can innovate and move faster than the bigger products.

Thanks for all the new updates! So many new features, so little time to test them all and find new ways to do things.

lemmingDev, 2025-02-01:
Na - with this library

Tries the 0x20 sensor usage page with 3 axis gyro and 3 axis accel and got driver error

For now it's other axes

LeeNX, 2025-02-01:
> Na - with this library

Cool. Awesome seeing all the new progress of late.
 
Posted other notes in https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/141

LeeNX, 2026-08-27:
Testing with PoC in place, which works on Linux - PR #337 

Please test and give feedback.

LeeNX, 2026-09-18:
Since v0.8.0 release, I believe this feature request has been completed.

How we have come so far. Only opened a little more than five and half years ago.  Oh what fun we are going to fun!

closed 2026-09-18 by LeeNX