lemmingDev/ESP32-BLE-Gamepad #44: Possible bug - Stack smashing protect failure!

simonbrommers, 2021-01-26:
Just trying out your library as it sounds exactly like what i want to do in my project - but on firing up the example Gamepad sketch, i get a repeated message:

Stack smashing protect failure!

abort() was called at PC 0x400dba3c on core 0

Backtrace: 0x40091448:0x3ffd6c00 0x40091679:0x3ffd6c20 0x400dba3c:0x3ffd6c40 0x4010dcf9:0x3ffd6c60 0x40133999:0x3ffd6cc0 0x40133fd9:0x3ffd6ce0 0x4013ed16:0x3ffd6d00 0x4011444e:0x3ffd6d20 0x4008e089:0x3ffd6d50

Rebooting...

Being new to the ESP32 platform and environment, I'm not sure what else to provide in terms of a bug report or trace to help reproduce - but it seems to have a crippling issue for me - any chance of a few pointers?

lemmingDev, 2021-01-26:
Try a better USB cable

Are you using one of the working examples?

Let me know how you go...

lemmingDev, 2021-01-28:
Let me know if you're still having issues after trying a new higher quality USB cable...

simonbrommers, 2021-01-28:
Thanks for your reply. I tried again with an older version of the library
and it worked ok. I'll try to remember to let you know what versions I had
and what I moved to....

lemmingDev, 2021-01-28:
Please do...

lemmingDev, 2021-02-14:
No response, so will close...

closed 2021-02-14 by lemmingDev