lemmingDev/ESP32-BLE-Gamepad #45: Wire diagram...

WarlockSVG, 2021-01-28:
Hey, is there a wire diagram / component list available?

lemmingDev, 2021-01-28:
@WarlockSVG 

Not really. Sounds like you need to go through some beginner Arduino tutorials and learn how to hook up buttons and potentiometers...

https://www.arduino.cc/en/Tutorial/DigitalInputPullup  --> button
https://www.arduino.cc/en/tutorial/potentiometer  --> potentiometer

WarlockSVG, 2021-01-28:
I can hook up a few buttons and potentionmeters in a standard way ;-)

I don't know how to connect 64 buttons, 3potentiometers ( 6 axes ), 2 Sliders ( ...more potentiometers ;-) ), 4point of view hats, d-pad and 3 other hat switches... ( more buttons )

I don't have that many IO pins on my ESP32 to do that in a "standard" way.

So can you guide me to a "tutorial" to the correct way to overcome the IO pin limit.

LeeNX, 2021-01-28:
You don't have to hook up every options?  ;-) 
    
 I have over come the poor analog options on the ESP32 by using a ADS1x15 via I2C - https://gitlab.com/leet/diy-gamepad/-/blob/master/ESP-BLE-Joy-01/ESP-BLE-Joy-01.ino. 
  
Has not be updated to the lasted lib version, but you should be able to get the idea.

lemmingDev, 2021-01-28:
As stated above, you only need to hook up the stuff you actually want.

Only need 4 buttons? Only connect 4 buttons...
Don't need any axes? Don't hook any up...

If you need more analog pins, see above.

If you need more digital IO pins, you can use a GPIO expander.

There are several options. Just Google 'Arduino GPIO expander' and you will find a heap of info on different boards and needed libraries such as MCP23017, SX1509 and 

There are some that offer 16 GPIOs on the I2C bus, and you could use 4 of them to add 64 extra GPIO.

So for 64 extra GPIO pins, you'd only need 2 x IO pins from the ESP32 (usually pins 21 and 22 on ESP32, but you can configure) plus 3.3V and GND (some modules need extra pins to address the correct module if using more than 1)

Some of the options top out at 64 (max 4 modules (16 x 4 = 64)), but others go to 128 by using 8 modules

closed 2021-01-28 by lemmingDev

WarlockSVG, 2021-01-28:
I totally get that I only need to connect / setup what I gonna use that is kinda obvious ;-)

Okay... I got something to look into for the insane amount of buttons one can connect... and the "Arduino GPIO expander" search term is also a big help. I had searched for "button matrix" but that did not helped a lot.

I guess it's the same for adding more potentiometers than available ADC pins.