lemmingDev/ESP32-BLE-Gamepad #47: Questions about documentation

lxzcl, 2021-02-01:
Hey, excuse me, I don't seem to understand the documentation.

example:
BUTTON_5   >  Y

The content in keywords.txt seems to be missing, can you give me some hints?
Or other information，thank you very much

LeeNX, 2021-02-01:
Have you looked at one of the many examples like https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/examples/Gamepad/Gamepad.ino?

Great place to start a project off.

lxzcl, 2021-02-01:
Yes, I have already seen it. But the examples are limited.

The reason for this question is that I cannot find buttons A and B on the handle and the up, down, left and right buttons.
If there is no detailed description, it means I need to try one button at a time, which means I will probably try 64 buttons, which is very troublesome.

LeeNX, 2021-02-01:
That depends on your platform, thou I am guessing BUTTON_1 is ```A``` and BUTTON_2 is ```B```, my hardware and code mapped the first four buttons 

DPAD and HAT can be used for directions, if you application maps them.

Hope that helps.

lemmingDev, 2021-02-02:
On Windows, everything is pretty self-explanatory.

If you load up the TestAll sketch and adjust the button delay timings to slow them down from

```
      bleGamepad.press(pow(2, i));
      delay(100);
      bleGamepad.release(pow(2, i));
      delay(25);
```

to 
```
      bleGamepad.press(pow(2, i));
      delay(1000);
      bleGamepad.release(pow(2, i));
      delay(250);
```

then you'll be able to fire up a testing app to see which buttons actuate which features

Off the top of my head, button 1, 2, 3, 4 are A, B, X, Y and then
5, 6 are left/right bumper, 7, 8 are select/start, and then there are 4 for up down left right and two for L3/R3 and two more for triggers, but am not sure of the exact order.

It's really up to the individual operating system / website / app to map them to what they see fit

When I get a chance, I'll see if I can nut out the defaults for Android

lxzcl, 2021-02-02:
Okay, it looks like a good idea.
Looking forward to a better solution.

lemmingDev, 2021-02-14:
Button 1: 
Button 1: 
Button 1: 
Button 1: 
Button 1:

FANHXIN, 2021-03-24:
How to write potentiometer pedal linear axis? thx!

FANHXIN, 2021-03-24:
#include <BleGamepad.h> 

BleGamepad bleGamepad;

const int potPin = 34;     
const int numberOfPotSamples = 5;  
const int delayBetweenSamples = 4;  
const int delayBetweenHIDReports = 5; 

void setup() 
{
  bleGamepad.begin();
}

void loop() 
{
  if(bleGamepad.isConnected()) 
  {
    
    int potValues[numberOfPotSamples]; 
    for(int i = 0 ; i < numberOfPotSamples ; i++)
    {
      potValues[i] = analogRead(potPin);
      delay(delayBetweenSamples);
    }

    int potValue = 0;  
    for(int i = 0 ; i < numberOfPotSamples ; i++)
    {
      potValue += potValues[i];
    }
    potValue = potValue / numberOfPotSamples;

    int adjustedValue = map(potValue, 0, 4095, 32737, -32737);
       
    bleGamepad.setRX(adjustedValue);
    delay(delayBetweenHIDReports);
  }
}

FANHXIN, 2021-03-24:
Abnormal response after potentiometer connection

closed 2021-04-03 by lemmingDev