lemmingDev/ESP32-BLE-Gamepad #48: [Feature Request] Serial and Firmware version field

LeeNX, 2021-02-08:
Please could we get a serial number and firmware version field.  
  
Serial number would be really helpful in udev rules, so that I can match my DIY gamePads for each player on my RetroPi.
Firmware would bee nice to track what release I am using in my projects.  
  
I don't quite yet get the ```HID Device Class Definition```. Thanks

lemmingDev, 2021-02-14:
Hi

My skills are pretty much beginner with HID/Arduino/C++

Am pretty sure you could individualise the controllers by changing the hexadecimal values in BleGamepad.cpp

```
BleGamepadInstance->hid->pnp(0x01,0x02e5,0xabbb,0x0110);
```

From memory, the 3rd parameter is the device id and the last one is the version, but can't remember exactly

LeeNX, 2021-02-14:
@lemmingDev don't be modest! ;-)
  
This had me download many of the specs PDF files, and found ```GATT Specification Supplement 3``` at https://www.bluetooth.com/specifications/specs/gatt-specification-supplement-3/
  
Finding ```3.150 PnP ID```, I don't totally understand, but from the spec:
```Vendor ID Source```, uinit8 with 0x01 - From Bluetooth SIG  
```Vendor ID Source```, uinit16 with 0x02e5
``` Product ID```, uinit16 with 0xabbb
``` Product Version```, uinit16 with 0x0110
  
From my Bluetooth Device info
[image]
Looks like Vendor ID and Product ID is byte swapped? I wonder if exposing and leave a good default for hid-pnp would cool?  
  
Can't see where 0x0110 might be listed, have not searched on my Linux computers. Might be usable in ```udev```.  
  
I have used manufacturer name and product name, but was hoping to try and keep both the same for multiple deploys of device for my RetroPi.  
  
There is a Firmware Revision String, but it does not look to be exposed in the ESP32 BLE library. This might not be possible with the current tools.

LeeNX, 2022-09-22:
Hoping to re-open this feature request, but this time with some more info:

Getting to understand BLE/GATT and bits more, I have tracked down and tested a working sketch to add 
- Model Number String - 0x2A24
- Serial Number String - 0x2A25
- Firmware Revision String - 0x2A26
- Hardware Revision String - 0x2A27
- Software Revision String - 0x2A28

As can be seen in the attached screenshot off my phone using something like BLE Scanner
[image]

The gist I have setup to demo working sketch - https://gist.github.com/LeeNX/f860b4c305216323f3b64c08d9f7e56b. 
  
My reasoning for Serial Number at least, is that things like SDL pulls this info and is used for ordering and possible other feature set, as can be seen in https://github.com/libsdl-org/SDL/blob/2a8d00634ddf4ece71c11ec664996ed8ea05205b/src/joystick/hidapi/SDL_hidapijoystick.c#L617 
  
My current limited C skill is what is holding me back from been able to add to the ESP32-BLE-Gamepad library. I am hoping that @lemmingDev might be able to do this quicker than me or even somebody else might be able to pick it up.  
  
I am going to keep trying to work on this and submit a PR, if nobody else beats me to it.

lemmingDev, 2022-09-22:
Think I closed this one by mistake.

Anyway, I did end up making the Vendor ID and Product ID configurable from the library, which should satisfy most people's requirements.

I just spent a few hours playing around trying to add those strings such as serial number, but I was unsuccessful. 

If you work out how to hardcode one or two of them in to the library, then I'll add them to the configuration class

LeeNX, 2022-09-23:
Thanks @lemmingDev ... Please excuse my poor C coding skills - https://github.com/lemmingDev/ESP32-BLE-Gamepad/pull/134 is a PoC?

Attached is a screenshot of added Serial number -   
[image]

lemmingDev, 2022-09-23:
Great Work!

Am getting onto integrating it now

What testing app is that?

lemmingDev, 2022-09-23:
Think I found it

BLE Scanner

Anyway, the library already sets the device name and manufacturer name, and just to test, I've now hardcoded it to additionally add;

#define CHARACTERISTIC_UUID_MODEL_NUMBER "2A24" // Characteristic - Model Number String - 0x2A24
#define CHARACTERISTIC_UUID_SOFTWARE_REVISION "2A28" // Characteristic - Software Revision String - 0x2A28
#define CHARACTERISTIC_UUID_SERIAL_NUMBER "2A25" // Characteristic - Serial Number String - 0x2A25
#define CHARACTERISTIC_UUID_FIRMWARE_REVISION "2A26" // Characteristic - Firmware Revision String - 0x2A26
#define CHARACTERISTIC_UUID_HARDWARE_REVISION "2A27" // Characteristic - Hardware Revision String - 0x2A27

It seems to work fine, so now I'll look to make it configurable

lemmingDev, 2022-09-24:
All done

[ESP32-BLE-Gamepad.zip](https://github.com/lemmingDev/ESP32-BLE-Gamepad/files/9637619/ESP32-BLE-Gamepad.zip)

I'll push changes and release later, however I have attached the updated library here

The CharacteristicsConfiguration.ino example shows how to use it

lemmingDev, 2022-09-26:
Version 5.2 released which adds these features

closed 2022-09-26 by lemmingDev