lemmingDev/ESP32-BLE-Gamepad #49: constantly getting events, without user input

impatience-cmd, 2021-02-14:
hi,

first of all, i'm kind of new, to how HID devices operate, but i'm willing to read more on those

i was able to get very far with my project, 
and the user input translates very well in various monitoring apps (windows10 gaming controller properties, joysticktest_092.exe application, and gamepadviewer.com)

but as soon as i jump into an emulator program, i get constant input:
sometimes it's pressing "down" (may be Dpad or analog command, or a button?!) constantly/repeatedly
^but this is NOT visible in the monitoring apps i mentioned.

i tried looking around in the library files, but i couldn't find something like "default send"

also it doesn't seem to be a certain button, like button1, or dpad etc.
it's just controller name, and a space; like this "Joypad "

but it gets received repeatedly by the emulator program
(and it seems to me, this background command is somehow blocking my ability to use my buttons, to interact with the emulation)

i haven't tested it on android, but will these days

i'm very thankful this library, it's great, and i got good results for my controller after a few hours, thanks!!

P.S.: is there an ability, to disable certain instances like "slider1, slider2" etc., since i don't need those, and my thought here is, can this probably somehow make the Bluetooth-"Bus" a little bit lighter, and less loaded with unused data?! since from the 64bit value, it seems with the ble-report, i always send all instances, that were created, and i thought they are send serially, so a smaller variable could means faster interaction?!?

thank you, if you can find time to respond

lemmingDev, 2021-02-14:
Hi

Can you please tell me the name of an emulator you're having trouble with so I can test it out?

As for an option with less features,  you could try out version 1.6.
https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/0.1.6 
It doesn't have the sliders and has a reduced number of buttons

impatience-cmd, 2021-02-14:
the emulators are 'retroarch' and 'virtual boy'

in virtual boy the mapping was constantly taken over by this default repeatedly sending, so i wasn't able to map the controller
in retroarch, it was a better, i was able to map the controller, but still in-game the button event was received by the game

i also did a short test with the example sketch 'multiple buttons', but it also appeared there, the character constantly walks down in gb files and the menu keeps rotating, as if i would be holding a down dpad/or control stick in z64 files

should i add the code?

impatience-cmd, 2021-02-14:
it also appeared, with all button press, setHat etc. disabled

might it be in the bluetooth server? (while it's connected)
i would like to try to troubleshoot myself, but my knowledge is not enough
i might try to read some more on it, and keep updating new finds, 

thank you so far! :)

impatience-cmd, 2021-02-21:
another update:
tested on android - successfully.

the gamepad worked without issues on my android device

while scrolling through the settings of an app, that monitors gamepad input, i saw a checkbox: that pointed towards status/ID messages, something like: ~"ignore status ID messages",
the checkbox was true

so my question as of right now is:
- within the library, can i disable constant status commands/messages, the bluetooth device sends (~while connected?)
- if not so, would it be an issue of my windows10 laptop needing to ignore those commands, and NOT pass them on, to the emulator programs, as it does right now?

from my little 2 year experience on arduino so far, my feeling says, it might be able to disable these messages/commands withing the library (i stumbled onto something like "bluetooth server", that the device (ESP32) creates), although i don't know for sure

my knowledge about BLE protocool (HID) is not deep enough to judge upon that - what do you think?

lemmingDev, 2021-02-21:
A couple of suggestions:

If it's not already, set your Core Debug Level to None

[image]

If that doesn't work, are you aware that there is another version of the gamepad library I maintain that uses the new NimBLE server instead of the default one?

It's more optimised, is updated regularly, more efficient and may just solver your problem.

Delete the current gamepad library from your libraries folder
Download and install the alternate gamepad library here: https://github.com/lemmingDev/ESP32-BLE-Gamepad/tree/NimBLE
You'll also need to have this library installed: https://github.com/h2zero/NimBLE-Arduino

Let me know how you go

lemmingDev, 2021-02-21:
Send your code through if you're stiff having issues...

impatience-cmd, 2021-02-25:
i tried your options, the issue still occurs.

how did your testing go? (quoted below):
> Hi
> 
> Can you please tell me the name of an emulator you're having trouble with so I can test it out?
> 
> As for an option with less features, you could try out version 1.6.
> https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/0.1.6
> It doesn't have the sliders and has a reduced number of buttons

(1)
as far as example code goes, i'm using you example library, with digitalRead and != if state has changed from the blink sketch example from arduino ide; i commented out all serial monitor stuff, and also the debug level was none

(2)
version 1.6. did not work at all

(3)
using the NimBLE build, as you suggested didn't change anything.

i'm still getting constant events as stated in my first post.

lemmingDev, 2021-02-25:
I tested with Retroarch and I was unable to reproduce the issue.

Are you using input pullups on your buttons? 

If you'd like further support, can you please provide your full code and the core and game you are trying to use in Retroarch.

Also, when you say Virtual Boy, I'm not sure what you mean. Can you link me to the specific emulator?

Thanks

impatience-cmd, 2021-03-02:
yes the pins were with internal pullups

since the issue occured with the original example sketches, with commented out serial monitor, and also commented out functions: buttonpress, setAxes etc.,
my guess is that the commands are coming from somewhere within the library structure (maybe BT server), since they appear with the device idle (with the example sketch)

regarding the emulator: 
my bad, the emulator is called: "visual boy advance" - when trying to assign buttons (within the menu under 'joypad' (their name for gamepad)), the ghost command takes place repeatedly, and blocks button assignments (quickly overrides assignments). Usually an assignment looks something like "Joy 1 Button 1", but in the case of the idle command it's only "Joy 1 " (repeatedly sent).

link to the emulator:
http://www.emulator-zone.com/doc.php/gba/vboyadvance.html

checking again, in retroarch:
the game "wario blast featuring bomberman" (core: TGB Dual), the character still walks down all the time

another thought i had:
could it be my hardware? i have a esp32 (wroom) clone from ebay

for my project it's ok to shrink it to android, (it works fine there!), the idle command appears under windows only; i was hoping i could find the source of the command, but i'll reduce the bluetooth gamepad to android, and as for windows, i'll use my functioning build with a leonardo under USB

lemmingDev, 2021-03-03:
Hi

Mapping the buttons in virtualboy advance works perfectly fine for me.

Please post your complete code so I can check it for problems

impatience-cmd, 2021-03-04:
#include <BleGamepad.h>   // https://github.com/lemmingDev/ESP32-BLE-Gamepad

//how many buttons and potis?
#define button_amount 19               //13 button (digitalPin in,);
         //^differentiate between: arduino_button_amount = 13, and joystick_button_amount = arduino_button_amount - 4(dpad buttons)
         
#define analogstick_poti_amount 4     //4 analogRead in's for the sticks (left_x(A0); left_y(A1); right_x(A2); right_y(A3));

#define input_pins_amount 23   // actually only 20 -- buttons+analogReads
         //^calculate input_pins from: arduino_button_amount(9)+analogstick_poti_amount(4) = 13

BleGamepad bleGamepad;

int previousButton1State = HIGH;

byte input_pins[button_amount] = {15, 13, 2, 16, 4, 2, 19, 21, 22, 23, 5, 18, 2, 17, 2, 14, 27, 26, 25};
                                       //^D2 will be always low
//free pin; D2(OUTPUT only!)
//D12 free input

//flags for all states of n
bool change_n__flag_array[input_pins_amount] = {};     //21 positions

//button state array
bool button_n__last_state[button_amount] = {};

////STATE if analog readout
//poti ~pseudo-array(manual) for updating/storing the analogReads (faster, if it's only done once every n)
uint16_t state_A0 = 1865;       //4095/2;       //16-bit, for 0..4095(==12-bit; 12-bit ADC in ESP32)            //8-bit only 255states, 12-bit variable doesn't exist.
uint16_t state_A1 = 1865;       //4095/2;
uint16_t state_A2 = 1865;       //4095/2;
uint16_t state_A3 = 1865;       //4095/2;

uint16_t state_A0_previous;
uint16_t state_A1_previous;
uint16_t state_A2_previous;
uint16_t state_A3_previous;

int16_t state_A0_signedInt;
int16_t state_A1_signedInt;
int16_t state_A2_signedInt;
int16_t state_A3_signedInt;

int16_t state_A0_signedInt_previous;
int16_t state_A1_signedInt_previous;
int16_t state_A2_signedInt_previous;
int16_t state_A3_signedInt_previous;

//definitions for DPad (~dictionary)
#define dpad_right 3
#define dpad_left 7
#define dpad_down 5
#define dpad_up 1

#define dpad_right_up 2
#define dpad_left_up 8
#define dpad_right_down 4
#define dpad_left_down 6

#define dpad_center 0

//defining the n==x states for later easy use, and re-mapability of 'right', 'down', 'left', 'up'
  //Dpad button / which case for n
#define right 15   //11
#define left  16   //12
#define down  17   //13
#define up  18     //14

//counter
byte n;
[... 393 lines cut ...]

          
           //Serial.print("state_A2_signedInt: ");
           //Serial.println(state_A2_signedInt);                           
       }
       if(n==22){    //up and down movement of the hama right stick   [IT'S INVERTED BY HAMA] moving down caused(?) -Y
         state_A3_signedInt = map(state_A3, 4095, 0, -32767, 32767);
         //Joystick.setRzAxis(state_A3_signedInt);

           bleGamepad.setRZ(state_A3_signedInt);
           //bleGamepad.setRightThumb(state_A2_signedInt, state_A3_signedInt);
          
           //Serial.print("state_A3_signedInt: ");
           //Serial.println(state_A3_signedInt);                   
       }
    
    change_n__flag_array[19] = 0; //empty flags for n == 19
    change_n__flag_array[20] = 0; //empty flags for n == 20
    change_n__flag_array[21] = 0; //empty flags for n == 21
    change_n__flag_array[22] = 0; //empty flags for n == 22
    }  

}

//END OF CODE

impatience-cmd, 2021-03-04:
i tried posting with < >
but my code it didn't catch it correctly

so here's the file directly as .ino:
https://drive.google.com/file/d/1haqXyhtnJn5gfhDoP5ac7Rp4-5_Lci7p/view?usp=sharing

lemmingDev, 2021-03-09:
Wow

There is a lot going on in your code.

May I suggest that you start a fresh sketch with just the button code only (no analogs, hats, sliders etc) and confirm that there are no extra events.

If that works, then slowly add one feature at a time until you find the culprit.

FYI, I am working on a new version of the library that has a customisable HID where you only enable the features you want. 

It currently supports any number of buttons from 0 to 64, plus any combination of axes out of X, Y, Z rZ, rX and rY

When I get motivated, I'll finish adding the hats, sliders and even some racing simulator controls.

impatience-cmd, 2021-03-09:
thank you for verifying, and trying to recreate the issue i had.

on all my other projects in arduino i was able to find the cause of the issue;
this is the first arduino sketch, where i somewhat "blamed it on the library";

because in the example sketches, even with disabling the button.press() etc. commands, (and leaving .begin(false) on, the issue still appeared;

the code that i made works fine under a leonardo over usb. so i took from that build, adjusted it, towards your library commands/functions, and then troubleshooted it with the serial.monitor

i still think my code should be fine,
since the gamepad-monitoring apps do **not** receive a permament 'hold down some button'; and disabling all the button events etc. still caused the issue... that's what's mysterious about it to me.

but anyway thank you for your time, and thoughts; since android doesn't face the issue i can still find a good use for the amount of time&work put in.

it's a good library thank you :)
and i'm also hyped about the tailoring the elements (less buttons etc.)
if you like, you can close this issue

lemmingDev, 2021-03-09:
Have you tried it on another Windows machine, another Bluetooth dongle or another ESP32 board?
I can't seem to reproduce the issue at all.

Your code seemed a bit weird with pin 2 being used for an LED output, and then used again repeatedly for different button inputs.

But removing your code from the equation and just using the MultipleButtonsDebounce example for testing, I am able to open the two emulators and map the buttons without issue. There's no "phantom" presses or the like.

So, let me know if you have another Windows machine to test with. Fire up either the SingleButtonDebounce or MultipleButtonsDebounce sketch (make sure to edit the pin variables to suit) and test for me

lemmingDev, 2021-04-03:
Try out this new test version
https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/54#issue-849162930

It allows you to enable only the features your controller is using

closed 2021-04-10 by lemmingDev