lemmingDev/ESP32-BLE-Gamepad #50: clarification for iOS compatibility note

societyofrobots, 2021-03-14:
Your note says:

> Compatible with iOS (No - not even for accessibility switch - This is not a “Made for iPhone” (MFI) compatible device)

Do you mean "_**Not**_ compatible with iOS"?

Or do you mean "Somewhat compatible, but not MFi"?

And if not compatible, what would need to be changed or added to make it compatible? I'm thinking of taking on this project.

lemmingDev, 2021-03-15:
Hi

It's not at all compatible.

It's not approved by Apple...

To make it work, I think there is some type of developer program you Have to enrol in.

kylethorson, 2021-04-01:
Hello, I was under the impression that for BLE HID devices, you do not need MFi enrollment on iPhone 5s and later. I've seen a few other ESP32-based repos that emulate HID mice/keyboards/gamepads that claim they work on iPhone. Is there something specific with your implementation that would make it not work, other than it not being enrolled in MFi?

societyofrobots, 2021-04-01:
kylethorson, to my understanding, HID BLE devices do not need official MFi approval from Apple. However, MFi communication is incompatible with the HID standard, so some additional code mojo is needed.

lemmingDev, 2021-04-01:
If anyone woks out what needs to be done, I'll try to include it

closed 2021-04-03 by lemmingDev

societyofrobots, 2021-06-16:
I spoke with a head engineer on the iOS team, and he said you *MUST* have an auth 3.0 hardware chip on board for authentication. And, it must also be approved via their MFi program.

I might jump through all these hoops. Such an unnecessary pain!