lemmingDev/ESP32-BLE-Gamepad #52: Hi. Brother. Can you share your wiring diagram? Do you have contact information?

zhai1987, 2021-03-30:
Hi. Brother. Can you share your wiring diagram? Do you have contact information?

lemmingDev, 2021-04-01:
Hi

The wiring is highly dependant on your own board and what features you're trying to get working.

You just wire up the number of buttons and axis you want according to your needs and ignore the stuff you don't need.

If I ever get around to finishing it, I'm working on a version that has a customisable HID report (features) similar to this  one:
https://github.com/MHeironimus/ArduinoJoystickLibrary/tree/version-2.0

zhai1987, 2021-04-02:
Brother. Is your firmware suitable for Esp32?

发自我的iPhone

 

 
 
 
 https://github.com/MHeironimus/ArduinoJoystickLibrary/tree/version-2.0

lemmingDev, 2021-04-02:
Yes. It is only for ESP32

zhai1987, 2021-04-02:
Can I just swipe your firmware. Do I have the right wiring diagram?

发自我的iPhone

lemmingDev, 2021-04-02:
You have to do a lot of programming yourself to get it to work with your hardware.

How many buttons do you need?

Which axes do you need?
What hardware are you using for analog axes, if any?

How many hat switches do you need?

zhai1987, 2021-04-02:
Brother, I can't program. Do you have time to help me? Is it convenient for you to leave your contact information? I need: 12 keys, 2 dials, 2 ec11 encoders, 3 potentiometers to control Z, y, X axes

zhai1987, 2021-04-03:
Hi. Brother. Are you free to help me?

发自我的iPhone

lemmingDev, 2021-04-03:
Hi

If you create code that will get your hardware working to output serial messages when the dials and potentiometers are turned, then I will help you getting my library working with it as a gamepad.

zhai1987, 2021-04-03:
Brother. Right. I want to use your library to make a Bluetooth game box. Well, thank you very much for your help. Can I share my wiring diagram with. Help change the code?

发自我的iPhone

lemmingDev, 2021-04-03:
Sure

Do you have any working examples for the dials though, as I don't have any of the same for testing

zhai1987, 2021-04-04:
I only have the drawings of the pcb controller sep32 here. But I don't know how to share it with you. You played app. Can I send a picture?

发自我的iPhone

lemmingDev, 2021-04-04:
You should be able to paste it in the message window

If not, give me the link

Also, what pins are you using

D

zhai1987, 2021-04-04:
Brother. You have Facebook Instagram.

发自我的iPhone

------------------ Original ------------------
From: lemmingDev ***@***.***&gt;
Date: Sun,Apr 4,2021 10:05 AM
To: lemmingDev/ESP32-BLE-Gamepad ***@***.***&gt;
Cc: zhai1987 ***@***.***&gt;, Author ***@***.***&gt;
Subject: Re: [lemmingDev/ESP32-BLE-Gamepad] Hi. Brother. Can you share your wiring diagram? Do you have contact information? (#52)

lemmingDev, 2021-04-04:
Na - I can help you through this thread

zhai1987, 2021-04-04:
I share pictures. Can you see it?!

发自我的iPhone

lemmingDev, 2021-04-04:
No

lemmingDev, 2021-04-04:
Try in Github, not on you email

zhai1987, 2021-04-04:
Right. You can't see the picture I sent back. You can only communicate by mailbox.

发自我的iPhone

 
Reopened #52.

lemmingDev, 2021-04-04:
Please use a PC instead of phone, or post a link to the pictures

Thanks

zhai1987, 2021-04-04:
[image]

zhai1987, 2021-04-04:
[image]

lemmingDev, 2021-04-04:
I can see the photos now, but it looks pretty complicated.

So, you're using a 4 x 3 button matrix (JP1 to JP12)?
Are JP13 to JP16 also part of a button matrix?
What devices are at JP13 to JP18?

zhai1987, 2021-04-04:
brother. Can you see the circuit diagram in the picture... Can you help to change the firmware? Thank you very much for your help

lov7777, 2021-04-04:
I’m here in police station you are agents to use my google account and
email just weit

lemmingDev, 2021-04-04:
I can see the circuit diagram

Have you already created that hardware?

What are the exact components at the positions mentioned in my last reply?

zhai1987, 2021-04-04:
Using a 4 x 3 button matrix (JP1 to jp12)
Four encoders, others are switch keys

zhai1987, 2021-04-04:
Using the 4 x 3 button matrix (JP1 to jp12)
Four encoders jp13 to jp16 
P13 to jp18 are paddle switches

lemmingDev, 2021-04-04:
Do you have a link to the switch keys and encoders?

If you're trying to get everything working, it's best to get one thing working, and then add one extra functionality at a time until it's all working.

I suggest starting with the button matrix

zhai1987, 2021-04-04:
https://www.ebay.com/itm/Rotary-Encoder-EC11-360-6mm-Shaft-Panel-Mount-6mm-Shaft-Arduino-Pi-Flux-Workshop/124276981910?_trkparms=ispr%3D1&amp;hash=item1cef7c4096:g:ELwAAOSwNDVfo2iP&amp;amdata=enc%3AAQAFAAACcBaobrjLl8XobRIiIML1V4Imu%252Fn%252BzU5L90Z278x5ickkp2NA0WuhQytmt4dbStg00CogJUpazgamjuMMAB1z1K4PQFQobwRcpAC0pOYCIhB%252Fdwsoom02NMyd4EE72SKnU81tSLRQBms%252FbCiezGIF427ohIQxOR3RSqFbB6fu5iYhdjuKq6AV4UJvN31GG74Z%252Fb%252B0VXI4zaLwYrmQ%252FSnWqCQlD76109%252B6Wz8LCVu8pWseFfWccMxS1sEnLsQz13tZ02HVaRITC0mJrS8AfMi2UUNpV04RUtyFAXWZA3tW59%252FeEO11qH6Gkxjk3jJEP3fnv7sf9hfyizlYHUR02Gw7G%252F2EpxooxImrx6UqYcKPs3A0b%252BKcn2SRPy3bpUSm58BP1KcHTnJhmosDxYiYa7k58oZMY1SOTXuz%252Bc%252Fyh1gBwOOmlKIbDWO94yMmdoLMWFzjah03wvxkJbMDh77XCIqdVRoIVMya9ddsO3tmNQUIN2nvYh5RMvUiEBah6fpQVZR9%252Bpp8OAOOzXHxK7r5px4rzSqZovi%252BcFfZBJnVRD30v%252FXnmiphfNF0Z4LkdaXxkZUILS%252Bv047d8d%252Bvqb%252BfzXpIx4zvzsjJXwHFoLjIh%252FMX8NHfvGdIpAe2CbqciHxqCka4gV1fkthD6q6WTsoG%252BgR%252F94MoaBvuddRWtL7dWmXtslf%252Fhyk7IL47EM8g%252FA6uzxQBJXVDuyoRabat2yXnI9w8jS6TBO6G%252BTM1%252FXkgexVcJZRYW%252Bl9CwI4XAVP8mvBpdwFOjGwR9FGJAIg8LqftsJsKNPiYG5S9hZRTTlTAH6blo08PMJzEkegveW6ZX5XMVt%252FwA%253D%253D%7Ccksum%3A124276981910240468b1a5724f8d8f92428786651f0c%7Campid%3APL_CLK%7Cclp%3A2334524

发自我的iPhone

zhai1987, 2021-04-04:
https://www.ebay.com/itm/5-x-SmartSwitch-SPST-12mm-IP67-Round-Head-Momentary-Plastic-Buttons/142074255526?hash=item21144918a6:g:1-YAAOSwoLFenNev

发自我的iPhone

lemmingDev, 2021-04-04:
Is it possible to show me a photo of everything connected to the ESP32, just so I get a sense of how it all looks?

lemmingDev, 2021-04-04:
Oh - so the rotary encoders also have a button, so there will be 16 buttons total in the button matrix?

zhai1987, 2021-04-04:
Right. It's just that one encoder can give three signals.

发自我的iPhone

zhai1987, 2021-04-04:
[image]

zhai1987, 2021-04-04:
[image]

zhai1987, 2021-04-04:
[image]

zhai1987, 2021-04-04:
Brother, I look forward to your help! thank!

lemmingDev, 2021-04-04:
I suggest you use the latest test version of my library 
[ESP32-BLE-Gamepad.zip](https://github.com/lemmingDev/ESP32-BLE-Gamepad/files/6254082/ESP32-BLE-Gamepad.zip)

You'll also need these 2 libraries
https://github.com/Chris--A/Keypad
https://github.com/h2zero/NimBLE-Arduino

Ok - let's start by getting the button matrix to work

```
/*
 * Custom controller for zhai1987
*/

#include <Keypad.h>       // https://github.com/Chris--A/Keypad
#include <BleGamepad.h>   // https://github.com/lemmingDev/ESP32-BLE-Gamepad

BleGamepad bleGamepad("ESP32 Keypad", "lemmingDev", 100);	//Shows how you can customise the device name, manufacturer name and initial battery level

#define ROWS 4
#define COLS 4
uint8_t rowPins[ROWS] = {13, 14, 15, 16};    //ESP32 pins used for rows      --> adjust to suit --> Pinout on board: R1, R2, R3, R4
uint8_t colPins[COLS] = {17, 18, 19, 21};    //ESP32 pins used for columns   --> adjust to suit --> Pinout on board: Q1, Q2, Q3, Q4
uint8_t keymap[ROWS][COLS] = 
{
  {0,1,2,3},          //Buttons  1,  2,  3,  4      --> Used for calulating the bitmask for sending to the library 
  {4,5,6,7},          //Buttons  5,  6,  7,  8      --> Adjust to suit which buttons you want the library to send
  {8,9,10,11},        //Buttons  9, 10, 11, 12      --> Just take 1 off the button number as shown to the left
  {12,13,14,15}       //Buttons 13, 14, 15, 16      --> Eg. The value 12 in the array refers to button 13
};

Keypad customKeypad = Keypad( makeKeymap(keymap), rowPins, colPins, ROWS, COLS);

void KeypadUpdate()
{
  customKeypad.getKeys();
  
  for (int i=0; i<LIST_MAX; i++)   // Scan the whole key list.      //LIST_MAX is provided by the Keypad library and gives the number of buttons of the Keypad instance
  {
      if ( customKeypad.key[i].stateChanged )   //Only find keys that have changed state.
      {
        uint8_t keystate = customKeypad.key[i].kstate;

        if(bleGamepad.isConnected()) 
        {
          if (keystate==PRESSED) {  bleGamepad.press(pow(2,(int(customKeypad.key[i].kchar)))); }    //Press or release button based on the current state
          if (keystate==RELEASED) { bleGamepad.release(pow(2,(int(customKeypad.key[i].kchar)))); }
        
          bleGamepad.sendReport();    //Send the HID report after values for all button states are updated, and at least one button state had changed
        }
      }
  }
}

void setup() 
{
  bleGamepad.begin();				//Begin library with default buttons/hats/axes
  bleGamepad.setAutoReport(false);	//Disable auto reports --> You then need to force HID updates with bleGamepad.sendReport()
}

void loop() 
{
  KeypadUpdate();
  delay(10);
}
```

zhai1987, 2021-04-04:
Great. Now you can press the matrix.

发自我的iPhone

 

 
 https://github.com/Chris--A/Keypad
 https://github.com/h2zero/NimBLE-Arduino
 
 /*  * Custom controller for zhai1987 */ #include <Keypad.h&gt;       // https://github.com/Chris--A/Keypad #include <BleGamepad.h&gt;   // https://github.com/lemmingDev/ESP32-BLE-Gamepad BleGamepad bleGamepad("ESP32 Keypad", "lemmingDev", 100);//Shows how you can customise the device name, manufacturer name and initial battery level #define ROWS 4 #define COLS 4 uint8_t rowPins[ROWS] = {13, 14, 15, 16};    //ESP32 pins used for rows      --&gt; adjust to suit --&gt; Pinout on board: R1, R2, R3, R4 uint8_t colPins[COLS] = {17, 18, 19, 21};    //ESP32 pins used for columns   --&gt; adjust to suit --&gt; Pinout on board: Q1, Q2, Q3, Q4 uint8_t keymap[ROWS][COLS] =  {   {0,1,2,3},          //Buttons  1,  2,  3,  4      --&gt; Used for calulating the bitmask for sending to the library    {4,5,6,7},          //Buttons  5,  6,  7,  8      --&gt; Adjust to suit which buttons you want the library to send   {8,9,10,11},        //Buttons  9, 10, 11, 12      --&gt; Just take 1 off the button number as shown to the left   {12,13,14,15}       //Buttons 13, 14, 15, 16      --&gt; Eg. The value 12 in the array refers to button 13 }; Keypad customKeypad = Keypad( makeKeymap(keymap), rowPins, colPins, ROWS, COLS); void KeypadUpdate() {   customKeypad.getKeys();      for (int i=0; i<LIST_MAX; i++)   // Scan the whole key list.      //LIST_MAX is provided by the Keypad library and gives the number of buttons of the Keypad instance   {       if ( customKeypad.key[i].stateChanged )   //Only find keys that have changed state.       {         uint8_t keystate = customKeypad.key[i].kstate;         if(bleGamepad.isConnected())          {           if (keystate==PRESSED) {  bleGamepad.press(pow(2,(int(customKeypad.key[i].kchar)))); }    //Press or release button based on the current state           if (keystate==RELEASED) { bleGamepad.release(pow(2,(int(customKeypad.key[i].kchar)))); }                    bleGamepad.sendReport();    //Send the HID report after values for all button states are updated, and at least one button state had changed         }       }   } } void setup()  {   bleGamepad.begin();//Begin library with default buttons/hats/axes   bleGamepad.setAutoReport(false);//Disable auto reports --&gt; You then need to force HID updates with bleGamepad.sendReport() } void loop()  {   KeypadUpdate();   delay(10); }

lemmingDev, 2021-04-04:
So - The matrix is working perfectly?

In your diagram, what are these 2 things?
[image]

zhai1987, 2021-04-04:
In fact, these two are also button switches. It's just that he uses a pick to block it!

发自我的iPhone

lemmingDev, 2021-04-04:
Why are they both connected to pin 33? 
So it doesn't matter which one you press, but they both activate the same button?

Also, what axes do you want the 4 encoders mapped to? X, Y, Z and rZ ?

Originally you mentioned potentiometers, but you're not using them any more?

zhai1987, 2021-04-04:
33 is just a common point. This is a separate one. 17 and 18. There are two buttons! The encoder is also used for buttons. An encoder. There are three gears. That's three signals! At first, I thought of adding 2 potentiometers to press the Y.x axis!

发自我的iPhone

lemmingDev, 2021-04-04:
So

The button at position 17 is connected to pin 33

and

The button at position 18 is also connected to pin 33?

zhai1987, 2021-04-04:
Yeah. You're right. These two are separate signals! Add four encoders and you have 26 signals. 2 potentiometers. So the 16-bit signal is not enough! What do you think, man?

发自我的iPhone

lemmingDev, 2021-04-04:
Ok, so are those 2 buttons down the bottom also part of the button matrix, and pin 33 is the row pin?

If so, why only 2 buttons and not 4?

zhai1987, 2021-04-04:
17 and 33 are a signal. 18 and 33 are another signal.

发自我的iPhone

lemmingDev, 2021-04-04:
What functionality do you want the encoders to have with the axes?

Encoders are continuous 360 degrees and can keep rotating
Do you want the axis value to reach the max or wrap back around if you keep turning?

zhai1987, 2021-04-04:
Brother. The encoder is not on the shaft. Ec11 encoders function like buttons, except that an encoder has three signals.

发自我的iPhone

lemmingDev, 2021-04-04:
ok - I think I understand

it is late here - I will look at it again tomorrow

zhai1987, 2021-04-04:
Brother, thank you very much! Looking forward to your firmware!

zhai1987, 2021-04-04:
> 我建议您使用我的库[ESP32-BLE-Gamepad.zip](https://github.com/lemmingDev/ESP32-BLE-Gamepad/files/6254082/ESP32-BLE-Gamepad.zip)的最新测试版本
> [](https://github.com/lemmingDev/ESP32-BLE-Gamepad/files/6254082/ESP32-BLE-Gamepad.zip)
> 
> 您还将需要这两个库
> [https://github.com/Chris--A/Keypad ](https://github.com/Chris--A/Keypad)
> https://github.com/h2zero/NimBLE-Arduino
> 
> 好的-让我们开始使用按钮矩阵
> 
> ```
> /*
>  * Custom controller for zhai1987
> */
> 
> #include <Keypad.h>       // https://github.com/Chris--A/Keypad
> #include <BleGamepad.h>   // https://github.com/lemmingDev/ESP32-BLE-Gamepad
> 
> BleGamepad bleGamepad("ESP32 Keypad", "lemmingDev", 100);	//Shows how you can customise the device name, manufacturer name and initial battery level
> 
> #define ROWS 4
> #define COLS 4
> uint8_t rowPins[ROWS] = {13, 14, 15, 16};    //ESP32 pins used for rows      --> adjust to suit --> Pinout on board: R1, R2, R3, R4
> uint8_t colPins[COLS] = {17, 18, 19, 21};    //ESP32 pins used for columns   --> adjust to suit --> Pinout on board: Q1, Q2, Q3, Q4
> uint8_t keymap[ROWS][COLS] = 
> {
>   {0,1,2,3},          //Buttons  1,  2,  3,  4      --> Used for calulating the bitmask for sending to the library 
>   {4,5,6,7},          //Buttons  5,  6,  7,  8      --> Adjust to suit which buttons you want the library to send
>   {8,9,10,11},        //Buttons  9, 10, 11, 12      --> Just take 1 off the button number as shown to the left
>   {12,13,14,15}       //Buttons 13, 14, 15, 16      --> Eg. The value 12 in the array refers to button 13
> };
> 
> Keypad customKeypad = Keypad( makeKeymap(keymap), rowPins, colPins, ROWS, COLS);
> 
> void KeypadUpdate()
> {
>   customKeypad.getKeys();
>   
>   for (int i=0; i<LIST_MAX; i++)   // Scan the whole key list.      //LIST_MAX is provided by the Keypad library and gives the number of buttons of the Keypad instance
>   {
>       if ( customKeypad.key[i].stateChanged )   //Only find keys that have changed state.
>       {
>         uint8_t keystate = customKeypad.key[i].kstate;
> 
>         if(bleGamepad.isConnected()) 
>         {
>           if (keystate==PRESSED) {  bleGamepad.press(pow(2,(int(customKeypad.key[i].kchar)))); }    //Press or release button based on the current state
>           if (keystate==RELEASED) { bleGamepad.release(pow(2,(int(customKeypad.key[i].kchar)))); }
>         
>           bleGamepad.sendReport();    //Send the HID report after values for all button states are updated, and at least one button state had changed
>         }
>       }
>   }
> }
> 
> void setup() 
> {
>   bleGamepad.begin();				//Begin library with default buttons/hats/axes
>   bleGamepad.setAutoReport(false);	//Disable auto reports --> You then need to force HID updates with bleGamepad.sendReport()
> }
> 
> void loop() 
> {
>   KeypadUpdate();
>   delay(10);
> }
> ```

[image]
Brother, I tested... Pin 13 sharing and 17, 18, 19, 21 usable functions. All 14 pins... No response... Why?

lemmingDev, 2021-04-04:
I'm not sure
Is it just the pin 14 row that doesn't work?
Do rows connected to pins 15 and 16 work?

zhai1987, 2021-04-05:
Brother. Morning! You're right. Only 13 matches and 17.18.19.21 can be used. 14.15.16 won't work.

发自我的iPhone

lemmingDev, 2021-04-05:
Try to change:

uint8_t rowPins[ROWS] = {13, 14, 15, 16};

to 

uint8_t rowPins[ROWS] = {16, 15, 14, 13};

lemmingDev, 2021-04-05:
Actually, perhaps its

uint8_t rowPins[ROWS] = {13, 14, 15, 16};
uint8_t colPins[COLS] = {21, 19, 18, 17};

zhai1987, 2021-04-05:
uint8_t rowPins [ROWS] = {13、14、15、16}；

到

uint8_t rowPins [ROWS] = {16，15，14，13};
uint8_t rowPins[ROWS] = {13, 14, 15, 16};    //ESP32 pins used for rows      --> adjust to suit --> Pinout on board: R1, R2, R3, R4
uint8_t colPins[COLS] = {21, 19, 18, 17};    //ESP32 pins used for columns   --> adjust to suit --> Pinout on board: Q1, Q2, Q3, Q4

Brother, after the change, it's still the same. 14.15.16 can't be used

lemmingDev, 2021-04-05:
I'm not sure then

It's really hard to help remotely

zhai1987, 2021-04-05:
Thank you for your help. Do you have time to try the encoder?

发自我的iPhone

lemmingDev, 2021-04-05:
I think the original code I gave you is correct.

Have you double-checked your wiring?

You shouldn't really move on to the next feature until this one is working

[106 more replies]

closed 2021-04-10 by zhai1987