lemmingDev/ESP32-BLE-Gamepad #53: iCade support for iOS?

wolfshow, 2021-04-01:
Is it possbile to add iCade support for iOS

wolfshow, 2021-04-01:
Here is another teensy 2.0 based project for iCade.
https://github.com/allenhuffman/iCadeTeensy

lemmingDev, 2021-04-01:
Hi

The iCade emulates a Bluetooth keyboard, not a Bluetooth gamepad

It should be pretty easy to get what you want working using this library:
https://github.com/T-vK/ESP32-BLE-Keyboard

or this version which uses a more up-to-date Bluetooth implementation:
https://github.com/wakwak-koba/ESP32-NimBLE-Keyboard

lemmingDev, 2021-04-01:
Here is the keymap you'll need:
[ION Arcade Dev Resource v1.5.pdf](https://github.com/lemmingDev/ESP32-BLE-Gamepad/files/6246395/ION.Arcade.Dev.Resource.v1.5.pdf)

lemmingDev, 2021-04-01:
Reading into it, the iCade send one keypress for button down, and a different one for button up

I was apparently done back in the day to get this type of device working with iOS before they added official keyboard/controller support.

It's a pretty clunky system and not sure if it's worth the effort.

closed 2021-04-03 by lemmingDev