lemmingDev/ESP32-BLE-Gamepad #54: New version with configurable HID descriptor

lemmingDev, 2021-04-02:
I have created a version of the library with a configurable HID descriptor

You can choose to enable/disable:

- 0 ~ 128 buttons
- 0 ~ 4 hat switches
- x, y, z, rZ, rX, rY axes (any combination)
- 0 ~ 2 sliders
- simulation controls (rudder, throttle, accelerator, brake, steering) (any combination)

You can also set the controller type between joystick, gamepad and multi-axis

Depending on your OS, only certain combinations/amounts may be seen

This version also uses the new and improved NimBLE library for efficiency

Haven't got around to updating all of the examples yet...

Anyone keen to test it out?

Johnbo100, 2021-04-03:
I would be happy to test it out

lemmingDev, 2021-04-03:
OK - For anyone looking to try it

I have updated/added examples/readme to reflect new functionality

[ESP32-BLE-Gamepad.zip](https://github.com/lemmingDev/ESP32-BLE-Gamepad/files/6253300/ESP32-BLE-Gamepad.zip)

**Installation**
- Delete your old version
- Extract this new one to your libraries
- Make sure NimBLE is installed either through the libraries manager or from https://github.com/h2zero/NimBLE-Arduino
- Fire up/modify some example

Let me know how it goes

FANHXIN, 2021-04-04:
Very good function
I'm willing to help test

FANHXIN, 2021-04-04:
Enable RX true is actually displayed as RY

FANHXIN, 2021-04-04:
Enable RZ true is actually displayed as RX

FANHXIN, 2021-04-04:
Enable RY true is actually displayed as RZ

lemmingDev, 2021-04-04:
Thanks for testing!

I have updated the post above with that fixed

The HID report was sent in the order x, y, z, rZ, rX, rY 
however
The HID descriptor was defined in the order x, y, z, rX, rY, rZ

They are now both correctly set in the order x, y, z, rZ, rX, rY 

Did you pick up anything else?

FANHXIN, 2021-04-05:
At present, no other problems have been found, I will continue to test

FANHXIN, 2021-04-05:
NimBLE-Arduino，What's the function of this? I am not using the library

FANHXIN, 2021-04-05:
I would like to ask how to write the power monitoring code

I want to test it

FANHXIN, 2021-04-05:
New library, test can connect, but the key, axis no response

FANHXIN, 2021-04-05:
Is the value of rotation axis 32737 or 65474

map(potValue, 0, 4095, 65474, -65474);

FANHXIN, 2021-04-05:
Same sketch, new library, axis no response

FANHXIN, 2021-04-05:
The display is normal

FANHXIN, 2021-04-05:
Single RX display will not work normally, but when RX RZ ry is turned on, RX will work

FANHXIN, 2021-04-05:
I suggest checking the shaft. It seems that there is something wrong with the shaft
With the example z-axis, there is no reaction

FANHXIN, 2021-04-05:
Test to me almost crazy, finally find the problem, ry open, but show as Rx

Open the shaft, even after normal operation, the shaft is not in the zero position, and then in the middle position

Oh, My God

FANHXIN, 2021-04-05:
Even if the rotation axis works, it is not in the zero position

LeeNX, 2021-04-05:
This is awesome! Thanks @lemmingDev ... Bummer, my Arduino IDE will not start on my Mac ... Why, cruel world!  
  
Is it possible to make a git branch tracking this work? I am going to track down another PC, so I can try and play with this new toy ...

lemmingDev, 2021-04-05:
Hi

Some of your previous sketches will not work with this new version as ALL axes are mapped from -32767 to 32767, instead of previously where some mapped from 0 to 65k or so

lemmingDev, 2021-04-05:
> NimBLE-Arduino，What's the function of this? I am not using the library

NimBLE is a library that has a new Bluetooth implementation

lemmingDev, 2021-04-05:
> I would like to ask how to write the power monitoring code
> 
> I want to test it

You'll need to find out how to get the percentage from your battery using a voltage divider (some ESP32 have them on a certain pin, while others don't, and you need 2 resistors), and then you send that value to your OS using , for example:
````
bleGamepad.setBatteryLevel(50);
````
to set it to 50%

lemmingDev, 2021-04-05:
> This is awesome! Thanks @lemmingDev ... Bummer, my Arduino IDE will not start on my Mac ... Why, cruel world!
> 
> Is it possible to make a git branch tracking this work? I am going to track down another PC, so I can try and play with this new toy ...

You could use a virtual machine or Bootcamp if you're desperate to try (assuming you don't have one on the new ARM Macs)

lemmingDev, 2021-04-05:
> Single RX display will not work normally, but when RX RZ ry is turned on, RX will work

I will try to check this out when I get a chance, but in the meantime, are you trying to set this with a potentiometer, or just virtually in code?

If you're using a potentiometer, perhaps try another pin

lemmingDev, 2021-04-06:
> Is it possible to make a git branch tracking this work? I am going to track down another PC, so I can try and play with this new toy ...

https://github.com/lemmingDev/ESP32-BLE-Gamepad/tree/NimBLE

FANHXIN, 2021-04-06:
Rx is working，nice！

What's your new library

How to write a new button after pressing the button to realize the third button

FANHXIN, 2021-04-06:
There is also a problem about the shaft. The shaft shakes very badly！

numberOfPotSamples，Set it higher. I don't know if it can be solved

lemmingDev, 2021-04-06:
> Rx is working，nice！
> 
> What's your new library
> 
> How to write a new button after pressing the button to realize the third button

I don't understand the question

If you want to press a button, you simply need to use, for example:
```
bleGamepad.press(3);
```
or
```
bleGamepad.press(BUTTON_3);
```
Both do the same thing

Then to release the button:
```
bleGamepad.release(3);
```
or
```
bleGamepad.release(BUTTON_3);
```
If you disable auto reports, then you need to remember to call
```
bleGamepad.sendReport();
```
every time you want to update the gamepad

FANHXIN, 2021-04-07:
Can your library be used with Bluetooth serial.h

FANHXIN, 2021-04-07:
Or can you add Bluetooth serial port function

FANHXIN, 2021-04-07:
Can your library be used with Bluetooth serial.h

Or can you add Bluetooth serial port function

lemmingDev, 2021-04-07:
> Can your library be used with Bluetooth serial.h
> 
> Or can you add Bluetooth serial port function

I believe you can use BT Serial with it. Give it a shot

FANHXIN, 2021-04-07:
It can display Bluetooth serial port and run your library perfectly at the same time

thx

lemmingDev, 2021-04-07:
> There is also a problem about the shaft. The shaft shakes very badly！
> 
> numberOfPotSamples，Set it higher. I don't know if it can be solved

This is now to how noisy the analog reads are on the ESP32 and the (poor) quality of potentiometer hardware used.

Pretty sure it has nothing to do with the library as when you generate the values via code, it's very precise.

If you run the PotAsAxis sketch and view the serial output, you'll see what I mean.

Could the example sketch be improved to reduce the jitter --> Yes - certainly
Does the library need to be improved --> correct me if I'm wrong, but I don't think it does

lemmingDev, 2021-04-07:
Oh - just realised you're probably trying to read 5V values.

ESP32 reads 0~3.3v, not 0~5v

So, make sure you are providing 3.3v from the ESP32 to the potentiometer instead of 5v

I just tried providing 5v to the potentiometer and the values were all over the place.
When I provided  3.3v, things were **much** better

Please provide feedback as to whether this was the issue you were having and whether it is now fixed

FANHXIN, 2021-04-08:
Yes, I use 5V

I'm still testing

I need your help！
Help me add the sketch of Bluetooth serial port in your library

lemmingDev, 2021-04-08:
Please try 3.3V to fix the potentiometer problem first.

Once that is fixed, we'll move on to the next issue

FANHXIN, 2021-04-08:
In fact, a good solution, in the potentiometer signal esp32, peripheral low-pass filtering, should be able to solve

lemmingDev, 2021-04-08:
So, what information exactly are you trying to send over BT Serial?

Can you post the code you've been trying?

I have used BLE Mouse library and BT Serial at the same time before and it worked pretty well (a prototype IR lightgun I was working on)

FANHXIN, 2021-04-08:
#include "BluetoothSerial.h"

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

BluetoothSerial SerialBT;

void setup() {
  Serial.begin(115200);
  SerialBT.begin("ESP32test"); //Bluetooth device name
  Serial.println("The device started, now you can pair it with bluetooth!");
}

void loop() {
  if (Serial.available()) {
    SerialBT.write(Serial.read());
  }
  if (SerialBT.available()) {
    Serial.write(SerialBT.read());
  }
  delay(20);
}

FANHXIN, 2021-04-08:
This is the esp32 Bluetooth serial port, which becomes a serial port after adapting to the computer. You can communicate with the screen by using RX TX, but I don't know how to add it to your library and use it with buttons

lemmingDev, 2021-04-08:
So, that sketch just takes information from the BT Serial and retransmits it over USB Serial.

Is that what you actually want to do, or do you want to do something else?

FANHXIN, 2021-04-08:
Do you know simhub？

lemmingDev, 2021-04-08:
Nope, but I have Google

FANHXIN, 2021-04-08:
I want nexion to receive information through Bluetooth serial port, but I want to use your library at the same time

lemmingDev, 2021-04-08:
What is nexion?

lemmingDev, 2021-04-08:
So, you want the ESP32 to have a 9600 baud rate Classic Serial BT connection that just receives (or do you need it to send too?) information from some type of device?

And then what do you want to do with that information once received?

FANHXIN, 2021-04-08:
Serial port screen

FANHXIN, 2021-04-08:
I just need to receive, and the serial port screen will work after receiving

FANHXIN, 2021-04-08:
Baud rate 115200

lemmingDev, 2021-04-08:
I think Classic BT only receives at 9600.

Of course you can output that info to the USB Serial at 115200

FANHXIN, 2021-04-08:
So I want you to help me finish it

lemmingDev, 2021-04-08:
So, is nexion the name of the device you are creating with my library, or some other device?

Also, what device will be sending the ESP32 the BT Serial information? A PC?

FANHXIN, 2021-04-08:
Just use the name of your library. It's a computer

FANHXIN, 2021-04-08:
In simple terms, just like using 32u4's ArduinoJoystickLibrary, its serial port still receives and sends messages.But it's wired

lemmingDev, 2021-04-08:
So, you just need to receive Serial BT information on the ESP32 and then spit the same info out over USB serial?

FANHXIN, 2021-04-08:
Esp32 only needs to receive the USB serial port information, and then through the TX pin to the screen RX work is completed

lemmingDev, 2021-04-08:
Sorry. 

It need to receive information on the USB Serial port?

FANHXIN, 2021-04-08:
yes

lemmingDev, 2021-04-08:
And then it transmits that same information over BT Serial?

[38 more replies]

closed 2021-04-10 by lemmingDev