lemmingDev/ESP32-BLE-Gamepad #55: Licence missing

Moje609, 2021-04-07:
Could you add a file that says what licence this library is published under? 
I'm planning on using this library for my graduation thesis and it's a bit unclear whether we can use libraries where the licence is not explicitly stated.

lemmingDev, 2021-04-07:
What licence do you suggest?

Moje609, 2021-04-07:
MIT is pretty common for this kind of stuff. It lets people do basically whatever they want with the code.

lemmingDev, 2021-04-07:
How about BSD simplified 2 clause, as that's what Adafruit libraries use:

https://choosealicense.com/licenses/bsd-2-clause/

BSD 2-Clause License

Copyright (c) [year], [fullname]
All rights reserved.

Redistribution and use in source and binary forms, with or without
modification, are permitted provided that the following conditions are met:

1. Redistributions of source code must retain the above copyright notice, this
   list of conditions and the following disclaimer.

2. Redistributions in binary form must reproduce the above copyright notice,
   this list of conditions and the following disclaimer in the documentation
   and/or other materials provided with the distribution.

THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE
FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL
DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR
SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER
CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY,
OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.

lemmingDev, 2021-04-07:
I think they're both completely compatible, so it wouldn't really matter which one

lemmingDev, 2021-04-07:
I guess MIT is fine also

Moje609, 2021-04-07:
Both look good to me. As far as I can tell the only difference is that BSD requires published binaries to print the licence somehow.

lemmingDev, 2021-04-07:
Will use MIT then

Moje609, 2021-04-07:
Cool, thanks. You can close this after you've pushed the licence.

Moje609, 2021-04-07:
Looks like the license is there.

closed 2021-04-07 by Moje609

lov7777, 2021-04-07:
I know who you are but you have nothing just a little girl bich

lemmingDev, 2021-04-07:
Yup