lemmingDev/ESP32-BLE-Gamepad #56: off-the-shelf ESP32 gamepad?

societyofrobots, 2021-04-09:
I'm looking for an off-the-shelf gamepad with the ESP32, but I've only been able to find custom one of a kind controllers.

Anyone know where I could buy one? Or want to sell me one? It's ok if it has some issues/quirks.

It's for a time sensitive project, otherwise I'd design my own.

zhai1987, 2021-04-09:
Brother, can you write.

lemmingDev, 2021-04-09:
Hi

I have just just re-written the last code I sent to make it better for everyone else that may use the code.
You can configure the amount of encoders, pins used, buttons used, button press duration etc

Please test it and tell me if it works ok (there is no keyboard matrix in this one)

If it works ok - then I can then try to put all of the code together

BTW - Do you need to be able to use all of the encoders at the same time, or would you only turn one at a time?
Also, will you be pressing buttons at the same time as rotating an encoder?

```
/*
 * Custom controller for zhai1987
 * 
 * Encoder portion --> arrays version
 */

#include <ESP32Encoder.h> // https://github.com/madhephaestus/ESP32Encoder
#include <BleGamepad.h>   // https://github.com/lemmingDev/ESP32-BLE-Gamepad

// Number of encoders
#define noOfEncoders 4

// Durations
#define  buttonHoldDuration     200   // Length of time (in ms) the botton is held for
#define  buttonDisableDuration  200   // Length of time (in ms) before the button can be actuated again

// Create gamepad, encoder objects and variable to store current encoder counts
BleGamepad bleGamepad("BLE Racing Controls", "lemmingDev");  
ESP32Encoder encoders[noOfEncoders];
int32_t encodersCount[noOfEncoders];
int8_t encoderPins[noOfEncoders * 2] = {22, 23, 26, 25, 32, 27, 5, 4};
int8_t encoderButtonsToActuate[noOfEncoders * 2] = {1, 2, 3, 4, 5, 6, 7, 8};

void setup()
{
  // Enable the weak pull up resistors
  ESP32Encoder::useInternalWeakPullResistors=UP;

  // Attach pins to encoders
  for(int currentEncoderIndex = 0 ; currentEncoderIndex < noOfEncoders ; currentEncoderIndex++)
  {
    encodersCount[currentEncoderIndex] = 0;
    encoders[currentEncoderIndex].attachHalfQuad(encoderPins[currentEncoderIndex * 2], encoderPins[(currentEncoderIndex * 2) + 1]);
    encoders[currentEncoderIndex].clearCount();
    encodersCount[currentEncoderIndex] = encoders[currentEncoderIndex].getCount();
  }
  
  bleGamepad.begin(noOfEncoders * 2, 0, false, false, false, false, false, false, false, false );        //Begin library with default buttons/hats/axes
  bleGamepad.setAutoReport(false);  //Disable auto reports --> You then need to force HID updates with bleGamepad.sendReport()
}

void loop() 
{
  int32_t tempEncodersCount[noOfEncoders];

  bool sendReport = false;
  
  for(int currentEncoderIndex = 0 ; currentEncoderIndex < noOfEncoders ; currentEncoderIndex)
  {
    tempEncodersCount[currentEncoderIndex] = encoders[currentEncoderIndex].getCount();

    if(tempEncodersCount[currentEncoderIndex] != encodersCount[currentEncoderIndex]) 
    {
      sendReport = true;
      
      if(tempEncodersCount[currentEncoderIndex] > encodersCount[currentEncoderIndex])
      {
        bleGamepad.press(encoderButtonsToActuate[currentEncoderIndex * 2]);
      }
      else
      {
        bleGamepad.press(encoderButtonsToActuate[(currentEncoderIndex * 2) + 1]);
      }

      encodersCount[currentEncoderIndex] = tempEncodersCount[currentEncoderIndex];
    }
  }

  if (sendReport)
  {
    bleGamepad.sendReport();
    delay(buttonHoldDuration);

    for(int currentEncoderIndex = 0 ; currentEncoderIndex < noOfEncoders ; currentEncoderIndex)
    {
      bleGamepad.release(currentEncoderIndex*2);
      bleGamepad.release((currentEncoderIndex*2) +1 );
    }
    
    bleGamepad.sendReport();
    delay(buttonDisableDuration);
  } 
}
```

lemmingDev, 2021-04-09:
Ok - hopefully that code above worked, so here is the complete (hopefully working) version

I can't test as I don't have any encoders handy

The code could be made more efficient using millis to track delay times and thus enable all encoders to be used at the same time, however I haven't got around to it yet, and it sould like you don't need it anyway

Please test and let me know how it goes

```
/*
 * Custom controller for zhai1987
 */

#include <ESP32Encoder.h> // https://github.com/madhephaestus/ESP32Encoder
#include <Keypad.h>       // https://github.com/Chris--A/Keypad
#include <BleGamepad.h>   // https://github.com/lemmingDev/ESP32-BLE-Gamepad

BleGamepad bleGamepad("BLE Racing Controls", "lemmingDev");  //Shows how you can customise the device name and manufacturer name

//**************** Button Matrix Setup ****************//
#define ROWS 5
#define COLS 4
uint8_t rowPins[ROWS] = {13, 14, 15, 16, 33};    //ESP32 pins used for rows      --> adjust to suit --> Pinout on board: R1, R2, R3, R4
uint8_t colPins[COLS] = {17, 18, 19, 21};    //ESP32 pins used for columns   --> adjust to suit --> Pinout on board: Q1, Q2, Q3, Q4
uint8_t keymap[ROWS][COLS] = 
{
  {1, 2, 3, 4},      //Buttons  1,  2,  3,  4      --> Used for calulating the bitmask for sending to the library 
  {5, 6, 7, 8},      //Buttons  5,  6,  7,  8      --> Adjust to suit which buttons you want the library to send
  {9, 10,11,12},     //Buttons  9, 10, 11, 12      --> 
  {13,14,15,16},     //Buttons 13, 14, 15, 16      --> Eg. The value 12 in the array refers to button 12
  {18,17,18,17}
};

Keypad customKeypad = Keypad( makeKeymap(keymap), rowPins, colPins, ROWS, COLS);

//**************** Encoders Setup ****************//
// Number of encoders
#define noOfEncoders 4

// Durations
#define  buttonHoldDuration     200   // Length of time (in ms) the botton is held for
#define  buttonDisableDuration  200   // Length of time (in ms) before the button can be actuated again

// Create encoder objects and variable to store current encoder counts
ESP32Encoder encoders[noOfEncoders];
int32_t encodersCount[noOfEncoders];
int8_t encoderPins[noOfEncoders * 2] = {22, 23, 26, 25, 32, 27, 5, 4};
int8_t encoderButtonsToActuate[noOfEncoders * 2] = {19, 20, 21, 22, 23, 24, 25, 26};

void KeypadUpdate()
{
  customKeypad.getKeys();
  
  for (int i=0; i<LIST_MAX; i++)   // Scan the whole key list.      //LIST_MAX is provided by the Keypad library and gives the number of buttons of the Keypad instance
  {
    if ( customKeypad.key[i].stateChanged )   //Only find keys that have changed state.
    {
      uint8_t keystate = customKeypad.key[i].kstate;

      if(bleGamepad.isConnected()) 
      {
        if (keystate==PRESSED)  { bleGamepad.press(customKeypad.key[i].kchar); }    //Press or release button based on the current state
        if (keystate==RELEASED) { bleGamepad.release(customKeypad.key[i].kchar); }
        
        bleGamepad.sendReport();    //Send the HID report after values for all button states are updated, and at least one button state had changed
      }
    }
  }
}

void setup()
{
  bleGamepad.begin(26, 0, false, false, false, false, false, false, false, false );        //Begin library with default buttons/hats/axes
  bleGamepad.setAutoReport(false);  //Disable auto reports --> You then need to force HID updates with bleGamepad.sendReport()

  // Enable the weak pull up resistors
  ESP32Encoder::useInternalWeakPullResistors=UP;

  // Attach pins to encoders
  for(int currentEncoderIndex = 0 ; currentEncoderIndex < noOfEncoders ; currentEncoderIndex++)
  {
    encodersCount[currentEncoderIndex] = 0;
    encoders[currentEncoderIndex].attachHalfQuad(encoderPins[currentEncoderIndex * 2], encoderPins[(currentEncoderIndex * 2) + 1]);
    encoders[currentEncoderIndex].clearCount();
    encodersCount[currentEncoderIndex] = encoders[currentEncoderIndex].getCount();
  }
}

void EncodersUpdate()
{
  int32_t tempEncodersCount[noOfEncoders];

  bool sendReport = false;
  
  for(int currentEncoderIndex = 0 ; currentEncoderIndex < noOfEncoders ; currentEncoderIndex)
  {
    tempEncodersCount[currentEncoderIndex] = encoders[currentEncoderIndex].getCount();

    if(tempEncodersCount[currentEncoderIndex] != encodersCount[currentEncoderIndex]) 
    {
      sendReport = true;
      
      if(tempEncodersCount[currentEncoderIndex] > encodersCount[currentEncoderIndex])
      {
        bleGamepad.press(encoderButtonsToActuate[currentEncoderIndex * 2]);
      }
      else
      {
        bleGamepad.press(encoderButtonsToActuate[(currentEncoderIndex * 2) + 1]);
      }

      encodersCount[currentEncoderIndex] = tempEncodersCount[currentEncoderIndex];
    }
  }

  if (sendReport)
  {
    bleGamepad.sendReport();
    delay(buttonHoldDuration);

    for(int currentEncoderIndex = 0 ; currentEncoderIndex < noOfEncoders ; currentEncoderIndex)
    {
      bleGamepad.release(encoderButtonsToActuate[currentEncoderIndex * 2] );
      bleGamepad.release(encoderButtonsToActuate[(currentEncoderIndex * 2) +1] );
    }
    
    bleGamepad.sendReport();
    delay(buttonDisableDuration);
  }
}

void loop() 
{
  KeypadUpdate();
  EncodersUpdate();
  delay(10);
}
```

lemmingDev, 2021-04-09:
https://www.electromaker.io/project/view/esp32-ble-gamepad
https://www.hackster.io/news/wifiboy32-an-esp32-based-iot-dev-kit-and-portable-game-console-5ca08f805052

I'm sure there are others you could find

societyofrobots, 2021-04-09:
First one isn't off-the-shelf, but that's ideally what I'm looking for.

How did you buy the WifiBoy? I can't find any ordering info on [their website](https://www.wifiboy.org/tw/). You just email them for it?

lemmingDev, 2021-04-09:
Yeah - I just emailed them

closed 2021-04-10 by lemmingDev