lemmingDev/ESP32-BLE-Gamepad #57: feature request: Add support for "START" and "SELECT" buttons

aguaviva, 2021-04-11:
I've been looking into doing this myself but I couldnt find any good examples, maybe this is low hanging fruit for you or you can point me to a good resource :)

This is how my descriptor looks so far but I am sure I am missing out many things

```cpp
	//       start and select
	{
		// USAGE_PAGE (Generic Desktop)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x05;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x01;

		// USAGE_MAXIMUM (Up to 128 buttons possible)            
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x29;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 2;

		// LOGICAL_MINIMUM (0)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x15;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x00;

		// LOGICAL_MAXIMUM (1)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x25;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x01;

		// REPORT_SIZE (1)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x75;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x01;

		// REPORT_COUNT (# of buttons)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x95;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 2;

		// UNIT (None)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x65;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x00;

		// USAGE (START)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x09;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x3D;

		// USAGE (SELECT)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x09;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x3E;
	}
```

aguaviva, 2021-04-11:
Alright I got this working :)

```cpp
	//       start and select
	{
		// USAGE_PAGE (Generic Desktop)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x05;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x01;

		// LOGICAL_MINIMUM (0)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x15;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x00;

		// LOGICAL_MAXIMUM (1)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x25;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x01;

		// REPORT_SIZE (1)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x75;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x01;

		// REPORT_COUNT (# of buttons)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x95;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 2;

		// UNIT (None)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x65;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x00;

		// USAGE (START)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x09;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x3D;

		// USAGE (SELECT)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x09;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x3E;
		
		// INPUT (Data,Var,Abs)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x81;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x02;		

		//padding
		
		// REPORT_SIZE (1)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x75;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x01;

		// REPORT_COUNT (# of padding bits)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x95;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 8-2;
				
		// INPUT (Const,Var,Abs)
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x81;
		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x03;		
	}
```

now time to add the "menu" button too!

aguaviva, 2021-04-11:
OK I made it! And using this feature I developed the following project: https://github.com/aguaviva/PS3ToBle

Could you check out my code and incorporate the changes I made however you see fit? You comments are more than welcome!

I hope this will help a lot of people use their ps3 controllers with any system :)

lemmingDev, 2021-04-11:
Nice work

Could you give any insight into what needs to be done to get BT Classic Serial and BLE working together?

I can get the ESP32 to use both when BLE is connected to a PC and BT Classic Serial is connected to a phone, but there seems to be an issue getting them both connected to the PC at the same time.

https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/54#issuecomment-816556504

aguaviva, 2021-04-11:
I had to change the way BT gets initialized, the commented out part is the old code, see:
https://github.com/aguaviva/PS3ToBle/blob/fe1561dd027bfb616992645527eb412b9da89e3a/libraries/PS3_Controller_Host/src/ps3_spp.c#L41

lemmingDev, 2021-04-11:
Any idea what would need to be changed either in my library or the official BT Serial library to get them to co-exist?

aguaviva, 2021-04-11:
I didn't have to change anything in your library. The only think is that I had to init your lib after the PS3 controller had connected via BT. 

In case you need more sampels, here you have 2 sample apps that show what is needed:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/coex

Any chance you could add the start and select buttons in your lib? I'd like to use the official libraries so it is easy to compile the app :)

lemmingDev, 2021-04-11:
I can add them:
0x3D = Start
0x3E = Select

What did you want to do with the PS button?

According to HID Usages document:
0x84 = System Context Menu
0x85 = System Main Menu
0x86 = System App Menu

aguaviva, 2021-04-11:
I wanted to map the PS button to the KEYCODE_BUTTON_MODE 

https://www.android-doc.com/reference/android/view/KeyEvent.html#KEYCODE_BUTTON_MODE

This key code is what brings up the emulator menu in Lemuroid, but I couldn't figure out its HID code in the documentation.

lemmingDev, 2021-04-11:
Let me know if you nut it out

lemmingDev, 2021-04-11:
Are you sure the OS / Application doesn't automatically assign those buttons based on the button numbering

Eg - button 1 = A / Cross
Eg - button 2 = B / Circle
Eg - button 3 = X / Square
Eg - button 4 = Y / Triangle
button 4 = 
button 4 = 
button 4 = 
button 4 =

lemmingDev, 2021-04-11:
Looks like the xbox controllers might map the menu button

https://github.com/ViGEm/ViGEmBus/issues/40

FANHXIN, 2021-04-12:
Bleh and BT classic serial coexist, I also want to know

FANHXIN, 2021-04-12:
Hi ！
lemmingDev

https://github.com/espressif/arduino-esp32/issues/1150

Looks like it's solved. Check it out

FANHXIN, 2021-04-12:
[image]

aguaviva, 2021-04-13:
> 
> Are you sure the OS / Application doesn't automatically assign those buttons based on the button numbering
> 
No it doesn't, you need to annotate buttons in the descriptor

Anyway, I am following this thread and I am excited to see you are making progress. 

BTW I am these days playing Zelda on my cellphone with the PS3 controller and I am having a blast, your lib rocks! :)

FANHXIN, 2021-04-14:
@aguaviva It seems that u are  a game love

u can search for gimx, which might help u

@lemmingDev  If u add the functionality of gimx to your library, it's great

Dreams r beautiful, reality is always cruel

lemmingDev, 2022-05-05:
Version 4.0 has start and select buttons as requested

closed 2022-05-08 by aguaviva