lemmingDev/ESP32-BLE-Gamepad #58: The operation in the code needs ble and Bluetooth serial port

FANHXIN, 2021-04-12:
https://desire.giesecke.tk/index.php/2018/02/27/esp32-bluetooth-serial-and-ble-server-running-together/

https://github.com/beegee-tokyo/ESP32-Weatherstation/blob/master/src/ble_bt.cpp

FANHXIN, 2021-04-12:
I'm also trying to modify 
hope to help you

pipech90, 2021-12-13:
Hi!

I have a problem.  I am trying to make a gamepad for pc.  But the PC does not see it in a bluetooth environment.  My phone sees and connects.  I tried to find it using Windows 8.1 and 10. What could be the problem?

lemmingDev, 2021-12-13:
> Hi!
> 
> I have a problem. I am trying to make a gamepad for pc. But the PC does not see it in a bluetooth environment. My phone sees and connects. I tried to find it using Windows 8.1 and 10. What could be the problem?

You need at least Bluetooth 4.0...

lemmingDev, 2022-05-26:
This library currently uses the NimBLE Bluetooth library for communication
Unfortunately, that library is not compatible with using classic Bluetooth UART (creates a standard COM port) at the same time
I may create another version that uses the default version of ESP32 BLE Arduino at a later stage.

Apparently, NimBLE can be used to create BLE UART (standard COM port not created), and there is an example in the NimBLE directory, however this library would need modification for it to work. May have a look at it at a later stage, however if someone wants to submit a pull request, then be my guest

closed 2022-05-26 by lemmingDev