lemmingDev/ESP32-BLE-Gamepad #59: Not sure it's a bug.

holography-be, 2021-04-13:
Hello,
I am using the latest version of ESP32-BLE-Gamepad but I have a problem in Windows 10 when I reset my ESP32.  
Win10 detects the device, I can connect it and test it with luser.github.io/gamepadtest/  (except that Win10 don't detect 128 buttons of the Gamepad but only 32, works well on Mac OS X).
Problem occurs when I reset or turn power off the ESP32.
Win10 switch the state of the device in loop : paired, connected, paired, connected, paired, connected and I got errors from esp32 here :
Starting BLE work!
Press all buttons one by one
Move x axis from center to max
Move x axis from min to max
Move x axis from max to center
Move y axis from center to max
Move y axis from min to max
Move y axis from max to center
Move rudder from min to max
Move throttle from min to max
Press all buttons one by one
Move x axis from center to max   ----> all work fine
␙␑��+� ��#␆\ǌ�␅��Starting BLE work!    <----- reset ESP32
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
(␑␅Aꡄ␜Ø��␌�␐�@Starting BLE work!          <----- reset again
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)

If I remove the device from win10, redetect it and reconnect it, it's working until the next reset of ESP32.
All is fine on my MBpro (Mac OS X).

Do you have any idea how I can solve the problem ?
Last updates have been deployed on Win10.

Thank you.

lemmingDev, 2021-04-17:
Can you post the exact code you used and I'll test it out?

Also, Windows supports 128 buttons, but will only visualise the first 32 in the default Game Controllers app.

Use this one http://www.planetpointy.co.uk/joystick-test-application/

holography-be, 2021-04-23:
Hi,
Found the problem.  My bluetooth card (Realtek) is too old.  I replace it with a Bluetooth dongle and now all work perfectly.
Thanks for your job.

holography-be, 2021-04-23:
issue closed

closed 2021-04-23 by holography-be