lemmingDev/ESP32-BLE-Gamepad #60: Feature request: Rotary encoders

deBoulangerie, 2021-05-08:
Iam using a fork of the old version:  https://github.com/MagnusThome/ESP32-BLE-Gamepad which supports 5 Rotary encoders. Is this possible to add to the new code?

lemmingDev, 2021-05-08:
Hi

The fork you refer to has an example sketch that maps 5 rotary encoders to buttons. 
The example sketch maps the first 16 buttons of the gamepad to a 4x4 button matrix, and then the next 15 buttons to rotateLeft/rotateRight/push of 5 encoders.
The fork doesn't have any explicit rotary encoder support, just the example code. Therefore my library will support encoders as well.

The way that buttons are referenced has changed in new version of my library so that example code will need some modifications

Here is the code with the modifications made for you. (I don't have any rotary encoders to test with)
```
#include <ESP32Encoder.h>     // https://github.com/madhephaestus/ESP32Encoder/
#include <Keypad.h>           // https://github.com/Chris--A/Keypad
#include <BleGamepad.h>       // https://github.com/MagnusThome/ESP32-BLE-Gamepad

BleGamepad bleGamepad("RaceKeys", "Arduino", 100);

////////////////////// BUTTON MATRIX //////////////////////
#define ROWS 4
#define COLS 4
uint8_t rowPins[ROWS] = {27, 25, 12, 32};
uint8_t colPins[COLS] = {17, 16, 00, 02};
uint8_t keymap[ROWS][COLS] = {
  {1,2,3,4},
  {5,6,7,8},
  {9,10,11,12},
  {13,14,15,16}
};
Keypad customKeypad = Keypad( makeKeymap(keymap), rowPins, colPins, ROWS, COLS); 

//////////// ROTARY ENCODERS (WITH PUSH SWITCHES) ////////////
#define MAXENC 5
uint8_t uppPin[MAXENC] = {36, 22, 39, 34, 19};
uint8_t dwnPin[MAXENC] = {26, 21, 35, 14, 23};
uint8_t prsPin[MAXENC] = {18, 15, 33, 13, 05};
uint8_t encoderUpp[MAXENC] = {17,20,23,26,29};
uint8_t encoderDwn[MAXENC] = {18,21,24,27,30};
uint8_t encoderPrs[MAXENC] = {19,22,25,28,31};
ESP32Encoder encoder[MAXENC];
unsigned long holdoff[MAXENC] = {0,0,0,0,0};
int32_t prevenccntr[MAXENC] = {0,0,0,0,0};
bool prevprs[MAXENC] = {0,0,0,0,0};
#define HOLDOFFTIME 150   // TO PREVENT MULTIPLE ROTATE "CLICKS" WITH CHEAP ENCODERS WHEN ONLY ONE CLICK IS INTENDED

////////////////////////////////////////////////////////////////////////////////////////

void setup() {
  Serial.begin(115200);

  for (uint8_t i=0; i<MAXENC; i++) {
    encoder[i].clearCount();
    encoder[i].attachSingleEdge(dwnPin[i], uppPin[i]);
    pinMode(prsPin[i], INPUT_PULLUP);
  }
  customKeypad.addEventListener(keypadEvent);
  //customKeypad.setHoldTime(1);
  bleGamepad.begin();
  Serial.println("Booted!");
}

////////////////////////////////////////////////////////////////////////////////////////

void loop() {

  unsigned long now = millis();

  // -- ROTARY ENCODERS : ROTATION -- //

  for (uint8_t i=0; i<MAXENC; i++) {
    int32_t cntr = encoder[i].getCount();
    if (cntr!=prevenccntr[i]) {
      if (!holdoff[i]) {
        if (cntr>prevenccntr[i]) { sendKey(encoderUpp[i]); }
        if (cntr<prevenccntr[i]) { sendKey(encoderDwn[i]); }
        holdoff[i] = now;
        if (holdoff[i]==0) holdoff[i] = 1;  // SAFEGUARD WRAP AROUND OF millis() (WHICH IS TO 0) SINCE holdoff[i]==0 HAS A SPECIAL MEANING ABOVE
      }
      else if (now-holdoff[i] > HOLDOFFTIME) {
        prevenccntr[i] = encoder[i].getCount();
        holdoff[i] = 0;
      }
    }
    
  // -- ROTARY ENCODERS : PUSH SWITCH -- //

    bool pressed = !digitalRead(prsPin[i]);
    if (pressed != prevprs[i]) {
      if (pressed) {  // PRESSED
        pressKey(encoderPrs[i]);
      }
      else {          // RELEASED
        releaseKey(encoderPrs[i]);
      }
      prevprs[i] = !prevprs[i];
    }
  }

  customKeypad.getKey();    // READ BUTTON MATRIX (EVENT CALLBACK SETUP)

  delay(10);
 
}

////////////////////////////////////////////////////////////////////////////////////////

void keypadEvent(KeypadEvent key){
  uint8_t keystate = customKeypad.getState();
  if (keystate==PRESSED)  { pressKey(key); }
  if (keystate==RELEASED) { releaseKey(key); }
}

////////////////////////////////////////////////////////////////////////////////////////

void sendKey(uint8_t key) {
    Serial.print("pulse\t");
    Serial.println(key);
    if(bleGamepad.isConnected()) {
      bleGamepad.press(key);
      delay(150);
      bleGamepad.release(key);
    }
}

void pressKey(uint8_t key) {
    Serial.print("hold\t");
    Serial.println(key);
    if(bleGamepad.isConnected()) {
      bleGamepad.press(key);
    }
}

void releaseKey(uint8_t key) {
    Serial.print("release\t");
    Serial.println(key);
    if(bleGamepad.isConnected()) {
      bleGamepad.release(key);
    }
}
```

lemmingDev, 2021-05-12:
Ok, so I will close the issue. Feel free to re-open it if you need a hand

closed 2021-05-12 by lemmingDev

wilker332, 2023-03-13:
hello everything is fine? I have this error. 'KeypadEvent' was not declared in this scope. I installed the libraries but it didn't work, the error continues. does anyone have an idea what it could be?