lemmingDev/ESP32-BLE-Gamepad #61: Use gamepad Bluetooth if not connected by serial to pc

AlEnterni, 2021-05-14:
Hello, is there any way to detect if it is connected to the computer and then send the data by serial? And if it is not connected to the pc it does it by Bluetooth?
Is another library needed to make it work as a serial gamepad?
Thank you very much

lemmingDev, 2021-05-15:
ESP32 does not support USB HID
You would need to send data over serial, and then write PC software to listen over serial for that data, and then translate that data to a virtual gamepad, such as ViGEm

I actually did it a couple of years ago for fun, and got it working, but have long since deleted the code.

I used C# (Visual Studio) to write the serial code and interact with the ViGEm framework (virtual Xbox 360 controller)

lemmingDev, 2021-05-15:
https://forums.vigem.org/topic/350/creating-a-vigem-feeder-application-c-step-by-step

closed 2022-05-26 by lemmingDev