lemmingDev/ESP32-BLE-Gamepad #65: Can't connect after controller has been connected to

KateDavidsonx, 2021-06-18:
After pairing and connecting to the controller, reconnecting doesn't work and the device is no longer visible and so cannot be paired to without the microcontroller being reset, is there any way around this? Thank you

lemmingDev, 2021-06-19:
Hi

It should be noted that if you change the number of buttons/axis etc then you'll need to unpair and repair the controller to get it working properly.

Anyway; what OS are you trying on?

And what exact scenario are you trying?

Scenario A:
1. Windows 10 machine on
2. Pair and connect ESP32 gamepad
3. Works fine
4. Leave Windows and ESP32 on, but turn off Bluetooth
5. Turn Bluetooth on
6. Device won't reconnect unless you power cycle the ESP32?

Scenario B:
1. Windows 10 machine on
2. Pair and connect ESP32 gamepad
3. Works fine
4. Leave ESP32 on
5. Restart Windows 10
6. Device won't reconnect unless you power cycle the ESP32

Scenario C;
Let me know exactly what you're trying and I'll try to reproduce...

KateDavidsonx, 2021-06-19:
Scenario A and B actually and any scenario whereby the Bluetooth connection
is dropped, its not able to reconnect without power cycling the esp32 nor
is it discoverable after a connection has been made and dropped.

I've tried this on Windows and android, same thing happens

deBoulangerie, 2021-06-19:
I have the same problem with my esp32, I bought an other bleutooth dongle and changed to another esp32 still having occasionally the connection problems. Even after power cycle.

I also found out that when switched off the esp32 the game controller is still visible in the set up USB gaming devices from Windows 10.

lemmingDev, 2021-06-19:
Hi

I don't have any of the same issues. Might be your specific board, power delivery or BT adapter.

You could try the NimBLE version of my library.
Delete the current one, install this one, and you'll also need to install NimBLE-Arduino
[ESP32-BLE-Gamepad-NimBLE.zip](https://github.com/lemmingDev/ESP32-BLE-Gamepad/files/6681354/ESP32-BLE-Gamepad-NimBLE.zip)

Details here: https://github.com/lemmingDev/ESP32-BLE-Gamepad/tree/NimBLE

KateDavidsonx, 2021-06-20:
Using NimBLE resolved the issue thanks!! Just a quick question, is there a way to disconnect the controller from the device it's connected to from the controller side? So I could press a button and it would technically act as a pairing button

lemmingDev, 2021-06-20:
Hi

Glad it's working for you.

Not sure about the pairing button question. Haven't really looked in to it.
I didn't write the BT side of the code
This library is a modification of the ESP32 BLE Mouse library, and I've only looked into the HID gamepad side of things.

lemmingDev, 2021-06-27:
Hi

I ended up realising that I was having similar Bluetooth connection issues as you described above, so I've now made the NimBLE code the default and released a new version

closed 2021-06-27 by lemmingDev

lov7777, 2021-11-16:
Zzzzzzzzzzzzz