lemmingDev/ESP32-BLE-Gamepad #66: Pull connected device name

mustang967, 2021-06-26:
I am trying to print the device name on an lcd when I am connected but I cannot find a function that works. Is this something that is currently possible and I'm just not seeing it?
Thanks!

lemmingDev, 2021-06-27:
Here you go. Obviously you'll have to integrate your own LCD library. Serial printing used for demonstration purposes.
NimBLE branch seems to give a better Bluetooth experience, so I'll do a new release in a few mins and make that the default version

```
#include <BleGamepad.h> 

char *deviceName = "My Gamepad";
bool padConnected = false;

BleGamepad bleGamepad(deviceName, "lemmingDev", 100);

void setup() 
{
  Serial.begin(115200);
  bleGamepad.begin();
}

void loop() 
{
  if(bleGamepad.isConnected()) 
  {
    if(padConnected == false)
    {
      padConnected = true;

      // Code here to deal with providing information on the device now being connected
      // Use the 'deviceName' variable declared at the top of the sketch
      Serial.print(deviceName);
      Serial.println(" connected");
    }
    
    // All of your regular gamepad code goes here
    // Ie. Check for inputs and then press buttons/move axes based on the inputs

    bleGamepad.press(1);
    delay(500);
    bleGamepad.release(1);
    delay(500);
  }
  else
  {
    if(padConnected == true)
    {
      padConnected = false;

      // Code here to deal with providing information on the device now being disconnected
      // Use the 'deviceName' variable declared at the top of the sketch
      Serial.print(deviceName);
      Serial.println(" disconnected");
    }
  }
}
```

lemmingDev, 2021-06-27:
https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/v0.3.3

Ver 3.3 now released, and works very well with the above code

mustang967, 2021-06-28:
Thank you so much for the help! Any chance we can also pull the name of the device that is connected to the esp32 via Bluetooth?
Thanks

lemmingDev, 2021-06-28:
Hi
Not sure how to do that
If you work it out, make a pull request ;)

mustang967, 2021-06-28:
Sounds good. I'll keep digging to see if I can figure it out. 
Thanks!

closed 2021-06-28 by mustang967