lemmingDev/ESP32-BLE-Gamepad #69: No guarantee of continuous and reliable BLE connection

Danjuanlab, 2021-07-20:
Computer system: windows 10 (19042.1110)
Bluetooth receiver: intel ax200 build in bluetooth
Development board: ESP32 DEV

I used the latest version 3.3 program to write a template code to the development board. When connecting to the computer, the connection cannot be successful once, and it needs to be repeated many times. 

When the connection is successful, the connection will be maintained for about 7/8 seconds, and then the development board  will be disconnected.

In the Bluetooth panel of the computer, I found that the connection status is repeatedly switched between "connected" and "paired". I have tried two different types of ESP32 development boards, and they all have this situation.
[image]

`/*
 * A simple sketch that maps a single pin on the ESP32 to a single button on the controller
 */

#include <BleGamepad.h>		// https://github.com/lemmingDev/ESP32-BLE-Gamepad

#define BUTTONPIN	21		// Pin button is attached to

BleGamepad bleGamepad;

int previousButton1State = HIGH;

void setup() 
{
  pinMode(BUTTONPIN, INPUT_PULLUP);
  pinMode(2,OUTPUT);
  pinMode(4,OUTPUT);
  Serial.begin(115200);
  bleGamepad.begin();
}

void loop() 
{
  if(bleGamepad.isConnected()) 
  {
  digitalWrite(4,LOW);
    int currentButton1State = digitalRead(BUTTONPIN);

    if (currentButton1State != previousButton1State)
    {
      if(currentButton1State == LOW)
      {
        bleGamepad.press(BUTTON_1);
        digitalWrite(2,HIGH);
        Serial.println("button1 push");
      }
      else
      {
        bleGamepad.release(BUTTON_1);
        digitalWrite(2,LOW);
        Serial.println("button1 res");
      }
    }
    previousButton1State = currentButton1State;
  }
  else{
  digitalWrite(4,HIGH);
  }
}`

lemmingDev, 2021-07-21:
Hi

The code you posted above works perfectly fine with my ESP32

The connection is stable, and does not disconnect.

Please try on a different computer/phone, different Bluetooth adapter, or a different ESP32 / power method

Danjuanlab, 2021-07-21:
@lemmingDev 
Hi，I currently cannot test it on other computers or Android phones, but the V3.3 version of the code cannot make my esp32 stable. I used the LOLIN D32 and the standard ESP32 dev board for testing, and the result is that the connection is not stable. When I use the old 3.1 version code, both development boards can be connected to my computer stably with BLE connection.

In addition, I was also testing the new ESP32-C3 development board recently. Unfortunately, the NimBLE library did not adapt to this new development board, and the code could not even be compiled correctly using the NimBLE library.

lemmingDev, 2021-07-24:
Thanks for the heads up.
I guess just use the 3.1 version for now

closed 2021-07-24 by lemmingDev