lemmingDev/ESP32-BLE-Gamepad #70: rotary encoders

osiriskidd, 2021-07-22:
i was also working off this fork's 5 rotary example  https://github.com/MagnusThome/ESP32-BLE-Gamepad and noticed you posted an updated code for someone a month ago in Feature request: Rotary encoders #60.
i'm trying to set up a 5x3 button matrix with additional 4 encoders at the bottom and running into issues getting the dials to work correctly.
ENC 1 - 18 on press and 16 on both up/down.
ENC 2 - 21 on press and 21 up/down.
ENC 3 - 24 on press and 24 up/down.
ENC 4 works as intended - 27 on press and 25 up 26 down
>
#include <ESP32Encoder.h>
#include <Keypad.h>
#include <BleGamepad.h>

BleGamepad bleGamepad("RaceKeys", "Arduino", 100);

////////////////////// BUTTON MATRIX //////////////////////
#define ROWS 3
#define COLS 5
uint8_t rowPins[ROWS] = {25, 26, 32};
uint8_t colPins[COLS] = {17, 18, 19, 23, 22};
uint8_t keymap[ROWS][COLS] = {
  {1,2,3,4,5},
  {6,7,8,9,10},
  {11,12,13,14,15}
  
};
Keypad customKeypad = Keypad( makeKeymap(keymap), rowPins, colPins, ROWS, COLS); 

//////////// ROTARY ENCODERS (WITH PUSH SWITCHES) ////////////
#define MAXENC 4
uint8_t uppPin[MAXENC] = {02, 04, 12, 14};
uint8_t dwnPin[MAXENC] = {00, 05, 13, 15};
uint8_t prsPin[MAXENC] = {16, 27, 34, 35}; //encoder press signal pins
uint8_t encoderUpp[MAXENC] = {16,19,22,25};
uint8_t encoderDwn[MAXENC] = {17,20,23,26};
uint8_t encoderPrs[MAXENC] = {18,21,24,27};
ESP32Encoder encoder[MAXENC];
unsigned long holdoff[MAXENC] = {0,0,0,0};
int32_t prevenccntr[MAXENC] = {0,0,0,0};
bool prevprs[MAXENC] = {0,0,0,0};
#define HOLDOFFTIME 150   // TO PREVENT MULTIPLE ROTATE "CLICKS" WITH CHEAP ENCODERS WHEN ONLY ONE CLICK IS INTENDED

////////////////////////////////////////////////////////////////////////////////////////

void setup() {
  Serial.begin(115200);

  for (uint8_t i=0; i<MAXENC; i++) {
    encoder[i].clearCount();
    encoder[i].attachSingleEdge(dwnPin[i], uppPin[i]);
    pinMode(prsPin[i], INPUT_PULLUP);
  }
  customKeypad.addEventListener(keypadEvent);
  customKeypad.setHoldTime(1);
  bleGamepad.begin(27);
  Serial.println("Booted!");
}

////////////////////////////////////////////////////////////////////////////////////////

void loop() {

  unsigned long now = millis();

  // -- ROTARY ENCODERS : ROTATION -- //

  for (uint8_t i=0; i<MAXENC; i++) {
    int32_t cntr = encoder[i].getCount();
    if (cntr!=prevenccntr[i]) {
      if (!holdoff[i]) {
        if (cntr>prevenccntr[i]) { sendKey(encoderUpp[i]); }
        if (cntr<prevenccntr[i]) { sendKey(encoderDwn[i]); }
        holdoff[i] = now;
        if (holdoff[i]==0) holdoff[i] = 1;  // SAFEGUARD WRAP AROUND OF millis() (WHICH IS TO 0) SINCE holdoff[i]==0 HAS A SPECIAL MEANING ABOVE
      }
      else if (now-holdoff[i] > HOLDOFFTIME) {
        prevenccntr[i] = encoder[i].getCount();
        holdoff[i] = 0;
      }
    }
    
  // -- ROTARY ENCODERS : PUSH SWITCH -- //

    bool pressed = !digitalRead(prsPin[i]);
    if (pressed != prevprs[i]) {
      if (pressed) {  // PRESSED
        pressKey(encoderPrs[i]);
      }
      else {          // RELEASED
        releaseKey(encoderPrs[i]);
      }
      prevprs[i] = !prevprs[i];
    }
  }

  customKeypad.getKey();    // READ BUTTON MATRIX (EVENT CALLBACK SETUP)

  delay(10);
 
}

////////////////////////////////////////////////////////////////////////////////////////

void keypadEvent(KeypadEvent key){
  uint8_t keystate = customKeypad.getState();
  if (keystate==PRESSED)  { pressKey(key); }
  if (keystate==RELEASED) { releaseKey(key); }
}

////////////////////////////////////////////////////////////////////////////////////////

void sendKey(uint8_t key) {
    Serial.print("pulse\t");
    Serial.println(key);
    if(bleGamepad.isConnected()) {
      bleGamepad.press(key);
      delay(150);
      bleGamepad.release(key);
    }
}

void pressKey(uint8_t key) {
    Serial.print("hold\t");
    Serial.println(key);
    if(bleGamepad.isConnected()) {
      bleGamepad.press(key);
    }
}

void releaseKey(uint8_t key) {
    Serial.print("release\t");
    Serial.println(key);
    if(bleGamepad.isConnected()) {
      bleGamepad.release(key);
    }
}

lemmingDev, 2021-07-24:
Hi

Head to near the bottom of this issue https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/52
and you might find what you need...

osiriskidd, 2021-07-25:
> 
> 
> Hi
> 
> Head to near the bottom of this issue #52
> and you might find what you need...

i adjusted my wiring and the code to match.
i've changed the pins around as well for testing.
its a lot better but:
enc1 press - 16/21
up - 16-20
down - 16-21

enc2 press - 17
up - 16-20 and 23
down - 16-20 and 22

enc3 press - 18
up - 24 (sometimes triggers 16-20)
down - 25 (sometimes triggers 16-20)

end4 press - 20
up - 26
down - 27

`
#define ROWS 4
#define COLS 5
uint8_t rowPins[ROWS] = {25, 26, 32, 33};    //ESP32 pins used for rows      --> adjust to suit --> Pinout on board: R1, R2, R3, R4
uint8_t colPins[COLS] = {17, 18, 19, 23, 22};    //ESP32 pins used for columns   --> adjust to suit --> Pinout on board: Q1, Q2, Q3, Q4
uint8_t keymap[ROWS][COLS] = 
{
  {1, 2, 3, 4, 5},      //Buttons  1,  2,  3,  4      --> Used for calulating the bitmask for sending to the library 
  {6, 7, 8, 9, 10},      //Buttons  5,  6,  7,  8      --> Adjust to suit which buttons you want the library to send
  {11, 12, 13, 14, 15},     //Buttons  9, 10, 11, 12      --> 
  {16, 17, 18, 19, 20}                          //Buttons 13, 14, 15, 16      --> Eg. The value 12 in the array refers to button 12
};

Keypad customKeypad = Keypad( makeKeymap(keymap), rowPins, colPins, ROWS, COLS);

//Potentiometers
const int pot1Pin = 34;                // Potentiometer is connected to GPIO 34 (Analog ADC1_CH6) 
const int pot2Pin = 35;                // Potentiometer is connected to GPIO 34 (Analog ADC1_CH7) 
const int numberOfPotSamples = 5;     // Number of pot samples to take (to smooth the values)
const int delayBetweenSamples = 4;    // Delay in milliseconds between pot samples

void KeypadUpdate()
{
  customKeypad.getKeys();
  
  for (int i=0; i<LIST_MAX; i++)   // Scan the whole key list.      //LIST_MAX is provided by the Keypad library and gives the number of buttons of the Keypad instance
  {
    if ( customKeypad.key[i].stateChanged )   //Only find keys that have changed state.
    {
      uint8_t keystate = customKeypad.key[i].kstate;

      if (keystate==PRESSED)  { bleGamepad.press(customKeypad.key[i].kchar); }    //Press or release button based on the current state
      if (keystate==RELEASED) { bleGamepad.release(customKeypad.key[i].kchar); }
      
      bleGamepad.sendReport();    //Send the HID report after values for all button states are updated, and at least one button state had changed
      
    }
  }
}

void setup()
{
  analogReadResolution(9);
  
  // Enable the weak pull up resistors
  ESP32Encoder::useInternalWeakPullResistors=UP;

  // Attach pins to encoders
  encoder1.attachHalfQuad(27, 16);
  encoder2.attachHalfQuad(04, 05);
  encoder3.attachHalfQuad(12, 13);
  encoder4.attachHalfQuad(14, 02);`

lemmingDev, 2021-11-11:
Unfortunately I don't actually own any rotary encoders, so can't really help troubleshoot...

closed 2021-11-11 by lemmingDev

osiriskidd, 2021-11-12:
> 
> 
> Unfortunately I don't actually own any rotary encoders, so can't really help troubleshoot...

no worries this already helped a lot