lemmingDev/ESP32-BLE-Gamepad #72: Possibility to receive data

AlEnterni, 2021-08-04:
Suppose I have connected the esp32 as a controller to the pc.
Can I make the pc send a series of data to the controller (esp32) by bluetooth?

lemmingDev, 2021-08-05:
The current version of my library relies on NimBLE, which is BLE only, no classic BT serial

If you go back a few versions to one that uses the standard library, you might be able to get it to work,

AlEnterni, 2021-08-29:
Thank you very much for replying, and if I use NimBLE with python I can send data to the gamepad ?

lemmingDev, 2021-08-29:
Hi

Not too sure. Another person patched in NimBLE to the library and I'm not too sure what can be achieved with it.

The NimBLE GitHub is here: https://github.com/h2zero/NimBLE-Arduino

lemmingDev, 2022-05-26:
This library currently uses the NimBLE Bluetooth library for communication
Unfortunately, that library is not compatible with using classic Bluetooth UART (creates a standard COM port) at the same time
I may create another version that uses the default version of ESP32 BLE Arduino at a later stage.

Apparently, NimBLE can be used to create BLE UART (standard COM port not created), and there is an example in the NimBLE directory, however this library would need modification for it to work. May have a look at it at a later stage, however if someone wants to submit a pull request, then be my guest

closed 2022-05-26 by lemmingDev

lemmingDev, 2022-09-22:
Works with the version I posted above here https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/111#issuecomment-1254595287