lemmingDev/ESP32-BLE-Gamepad #76: Ubuntu not receiving data

Kylemark64, 2021-09-17:
I'm currently trying to get a custom 2-Axis Bluetooth joystick working in Ubuntu.  It works flawlessly in Windows 10, but when moving over to my Ubuntu VM the axis' values don't change when moving the joystick around.  I am currently using jstest to verify functionality and I've had no issues with it when using other controllers/joysticks.

I know this isn't much to go off, but is there anything I need to change specifically when using this library for linux?

lemmingDev, 2021-09-17:
Hi Kyle

The library supports 6 different axes and 2 sliders.

What you could do is try the TestAll sketch and see if any of the axes/sliders work. 

If they do, try assigning your axes to ones other than the 2 you've tried.

Let me know how you go

lemmingDev, 2021-09-19:
Did you end up getting it to work?

Kylemark64, 2021-10-01:
Hi lemmingDev,

Sorry for the super late reply, I've been very busy with my University work.  I tried all 6 axis provided: X, Y, Z, RZ, RX, & RY.  Unfortunately, none of them work with my Ubuntu VM even though the joystick Bluetooth device is detected.  It is being seen as a 2-Axis device at least.

Do you know of any extra drivers I may have to install to get this working?

Here's a screenshot of said VM with the device connected:
[image]

lemmingDev, 2022-05-26:
Have you tried Ubuntu 21.10, with a kernel 5.13
Perhaps an updated distro etc will get it working

closed 2022-05-26 by lemmingDev