lemmingDev/ESP32-BLE-Gamepad #77: Game Controller naming

TheRussian007, 2021-09-20:
Hi,
great library.
I just have a small niggle.
My device is recognised as I named it "Load Cell pedals" as Bluetooth name, but I cannot figure out how to change the Game Controller name in Windows 10 under "Game Controllers". 
In my case it always shows up as "8 axis 16 button device with hat switch" and I have uploaded multiple different sketches.

I found that entry in the Registry Editor that is saved in the "OEMName" and deleted it. Once connected I get "3 axis 0 button device" per latest configuration.

Anyone got an idea how to control that "OEMName" name within the sketch, so my BT device can have a nice name in Windows and all games automatically?

Thank you

[image]

lemmingDev, 2021-09-20:
Hi

It's been asked a few times, but I unfortunately don't know how to do it.

TheRussian007, 2021-09-21:
Hi
thank you for the response.
I happily live with it and change it manually in the registry.
The library works great.
Thank you

lemmingDev, 2021-09-21:
For the benefit of others, where is it in the registry?

Glad you're enjoying the library

TheRussian007, 2021-09-21:
My reg location looks a bit different with some long numbers in it.
I just used Ctrl+F search and looked for the exact current name of the game controller, "8 axis 16 button device with hat switch", had only one entry in my case.

deBoulangerie, 2021-09-22:
https://youtu.be/UiZ77CzEkb8

I have found this, changing the register would only affect the pc you change it on?

Haven't tried if it works because it requires coding via simhub.

lemmingDev, 2021-09-22:
Yes - Changing the registry only affects the PC you change it on.

If anyone works out how to do it through the library - let me know...

davidgrohl, 2022-03-21:
Maybe I misunderstood, did you try to change name in this function?
`BleGamepad::BleGamepad(std::string deviceName, std::string deviceManufacturer, uint8_t batteryLevel)`

TheRussian007, 2022-03-21:
yes, but that only comtrols the Bluetooth device name, not the gamepad name. Have a look at the screenshot on the first post.
Insted of "Load Cell pedals" is shows "8 axis 16 button device with hat switch".

closed 2022-05-05 by lemmingDev