lemmingDev/ESP32-BLE-Gamepad #79: Doesn't work after reconnection

AlexM600, 2021-10-12:
1. I uploaded example (any of them), connect to Windows 10 computer over bluetooth, checked it works.
2. Reset or repower ESP32
3. It reconnected to Windows automatically but doesn't work. 
The only way to make it work again: Remove paired device from Windows and connect as new one. And it works again until reset ESP32

deBoulangerie, 2021-10-12:
Are you using the latest nimble version? That fixed my connection problems

lemmingDev, 2022-05-26:
Please ensure that you have updated ESP32 version in the boards manager, and are running the latest versions of the libraries etc.

It is working fine on reconnect for hundreds of users so I will put it down to something specific on your system and close the issue

closed 2022-05-26 by lemmingDev

RaazP, 2025-01-04:
Hi, a bit late to the party, but I'm having the same issue with two different boards.
First, thanks a lot for sharing this project, I'm building a DIY wireless buttonplate for simracing, grabbing power from the pins for the original Moza wheels. Wouldn't be possible without this!

Anyway:
One board is the 38 Pin AZDelivery DevKit V2 and the other is some 30 Pin DOIT DevKit V1 Clone from Aliexpress.
When compiling, I'm getting a warning about a deprecated adc driver, but that probably doesn't matter for just using buttons.

I'm using the latest NimBLE 2.1.2 by h2zero and your BLE Gamepad 0.6.1.
Windows 11 23H2 and Win 10 22H2. Same issue with my Surface Pro 8 on 24H2.

My workaround is to "forget" the device in the Windows BT menu, pressing the reset button (connected the EN pin with a microswitch), searching for devices and everything works until the ESP32 shuts down, resets or the PC.

Any ideas?

Thanks in advance!

LeeNX, 2025-01-04:
Hey @RaazP, this issue is closed and might not get much in the way of feedback.  
  
Just commented on a similar, thou not the exact say issue on Windows - https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/251. 

Could try some of the suggestions I have made there.
  
I am guessing that this might be Windows been more strict than other host OSs or devices ... or Microsoft sabotaging none licensed Microsoft XBox gamepads </rant> ;-)