lemmingDev/ESP32-BLE-Gamepad #82: Error compiling for every esp32 board (versions 3.0 and higher)

MIKOLAJBF, 2021-10-30:
I get compilation errors only witch the 3.x versions, I tried with 2.0 and everything works fine. Error occurs even when uploading empty code with included <BleGamead.h>. Any idea what this error message means?
```
C:\Users\Miko�aj\Documents\Arduino\libraries\ESP32-BLE-Gamepad\BleGamepad.cpp: In member function 'void BleGamepad::press(uint8_t)':

C:\Users\Miko�aj\Documents\Arduino\libraries\ESP32-BLE-Gamepad\BleGamepad.cpp:546:34: error: array subscript has type 'char' [-Werror=char-subscripts]

   uint8_t result = _buttons[index] | bitmask;

                                  ^

C:\Users\Miko�aj\Documents\Arduino\libraries\ESP32-BLE-Gamepad\BleGamepad.cpp:548:31: error: array subscript has type 'char' [-Werror=char-subscripts]

   if (result != _buttons[index]) 

                               ^

C:\Users\Miko�aj\Documents\Arduino\libraries\ESP32-BLE-Gamepad\BleGamepad.cpp:550:19: error: array subscript has type 'char' [-Werror=char-subscripts]

     _buttons[index] = result;

                   ^

C:\Users\Miko�aj\Documents\Arduino\libraries\ESP32-BLE-Gamepad\BleGamepad.cpp: In member function 'void BleGamepad::release(uint8_t)':

C:\Users\Miko�aj\Documents\Arduino\libraries\ESP32-BLE-Gamepad\BleGamepad.cpp:562:35: error: array subscript has type 'char' [-Werror=char-subscripts]

   uint64_t result = _buttons[index] & ~bitmask;

                                   ^

C:\Users\Miko�aj\Documents\Arduino\libraries\ESP32-BLE-Gamepad\BleGamepad.cpp:564:31: error: array subscript has type 'char' [-Werror=char-subscripts]

   if (result != _buttons[index]) 

                               ^

C:\Users\Miko�aj\Documents\Arduino\libraries\ESP32-BLE-Gamepad\BleGamepad.cpp:566:19: error: array subscript has type 'char' [-Werror=char-subscripts]

     _buttons[index] = result;

                   ^

C:\Users\Miko�aj\Documents\Arduino\libraries\ESP32-BLE-Gamepad\BleGamepad.cpp: In member function 'bool BleGamepad::isPressed(uint8_t)':

C:\Users\Miko�aj\Documents\Arduino\libraries\ESP32-BLE-Gamepad\BleGamepad.cpp:795:32: error: array subscript has type 'char' [-Werror=char-subscripts]

   if ((bitmask & _buttons[index]) > 0)

                                ^

cc1plus.exe: some warnings being treated as errors

exit status 1
Error compiling for board Node32s.
```

MIKOLAJBF, 2021-10-30:
I just found out, that this only show up when library is added via library manager. When I added it's files to my project folder and included them, everything is compiling and working without problems.

lemmingDev, 2021-10-30:
Hi

Release https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/v0.3.4 should sort the issue
Arduino library manager should sync in the next few days

The code on GitHub has had the fix for a couple of months but I hadn't published a new new release yet

closed 2021-11-11 by lemmingDev