lemmingDev/ESP32-BLE-Gamepad #83: BT Serialport with Gamepad

sefatunckanat, 2021-11-06:
Hello, i want to use Bluetooth Serial port with library. How can i do that?

I get an error when i add "#include "BluetoothSerial.h" to any ble game pad example and it not work gamepad and serialport. 

i want to do https://dev.to/nmwilk/simhub-without-the-wires-18ij with game pad

lemmingDev, 2021-11-07:
Hi

The current version of my library uses NimBLE - which is unfortunately incompatible with Bluetooth Serial

sefatunckanat, 2021-11-07:
> Hi
> 
> The current version of my library uses NimBLE - which is unfortunately incompatible with Bluetooth Serial

Doesn't NimBLE support a similar system?

lemmingDev, 2022-05-26:
This library currently uses the NimBLE Bluetooth library for communication
Unfortunately, that library is not compatible with using classic Bluetooth UART (creates a standard COM port) at the same time
I may create another version that uses the default version of ESP32 BLE Arduino at a later stage.

Apparently, NimBLE can be used to create BLE UART (standard COM port not created), and there is an example in the NimBLE directory, however this library would need modification for it to work. May have a look at it at a later stage, however if someone wants to submit a pull request, then be my guest

closed 2022-05-26 by lemmingDev

lemmingDev, 2022-09-22:
Works with the version I posted above here https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/111#issuecomment-1254595287